FedeDP/Clight #157: [FEATURE REQ]

linusrg1, 2020-10-20:
**Is your feature request related to a problem? Please describe.**
I would like it if clight changed brightness more quickly, and if it made the camera just as active as an ambient light sensor would be. I also have an issue where it sometimes lowers the brightness to 0 rendering the backlight off. It would be nice if it had a cap where the brightness would stop at one instead of 0.

**Describe the solution you'd like**
If it is possible to change in the config please tell me how.

**Describe alternatives you've considered**
My solution for the black screen issue is to turn my brightness up by one manually.

FedeDP, 2020-10-20:
Hi! Thanks for opening this feature request!
Actually, both requests are already satisfied :) 
Have a look at BACKLIGHT configuration in [clight conf](https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L30); you should look for `shutter_threshold` property, and `trans_step` + `trans_timeout` properties.

linusrg1, 2020-10-22:
is the higher the number the longer it takes for the camera to respond to
the ambient light?

linusrg1, 2020-10-22:
sry Actually i am trying to understand what all this means. would you mind
explaining in detail what each of those settings do?

FedeDP, 2020-10-26:
Hi! Sorry for the delay!
Well i am pretty sure i'd fail to explain them better than clight conf file...
```
## Set a threshold: if detected ambient brightness is below this threshold,
## capture will be discarded and no backlight change will be made.
## Very useful to discard captures with covered webcam.
# shutter_threshold = 0.10;
```
Instead, transition step and timeout are useful for smooth backlight changes, if this is what you were looking.
If instead you were looking to lengthen time between clight's captures, you should change {ac,batt}_timeouts options: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L49

closed 2020-11-01 by FedeDP

linusrg1, 2020-11-12:
After uncommenting shutter threshold what would you suggest changing its value to to have it constantly constantly keep the camera on.

And on a slightly different note will clight interfere with video chats?

FedeDP, 2020-11-13:
> After uncommenting shutter threshold what would you suggest changing its value to to have it constantly constantly keep the camera on.

Mmh shutter threshold prevents this issue:
> I also have an issue where it sometimes lowers the brightness to 0 rendering the backlight off

There is no way to continuously capture from camera to adjust in real time; but changing [ac,batt]_timeouts under BACKLIGHT should will allow clight to capture N frames from camera (num_captures option) every M seconds (as set in [ac,batt]_timeouts). 

> And on a slightly different note will clight interfere with video chats?

It does not as it is not continuously capturing. Simply, clight will fail to capture if any other application is using webcam, and nothing happens.