FedeDP/Clight #158: [FEATURE REQ] Possibility to calibrate the sensor

coldspark29, 2020-10-20:
**Is your feature request related to a problem? Please describe.**
I don't know if this is possible. Maybe the sensor of my Thinkpad ist just bad. I always get the lowest sensor value unless direct bright light is shining onto the sensor. It is now evening and the light in my room is on, but still I get 0.044 of sensed brightness. It seems to me that it is calibrated very badly. Is there a possibility to change this?

FedeDP, 2020-10-20:
Hi, thanks for opening this feature request!
Unfortunately calibrating the sensor is nothing clight can do...
Are you using an ALS sensor? Or webcam?

coldspark29, 2020-10-20:
It is an ALS sensor. The webcam acutally has better results, but then you see the light flashing, which annoys me.

closed 2020-10-20 by coldspark29