FedeDP/Clight #160: [FEATURE REQ] Capture on wake

NICHOLAS85, 2020-10-31:
**Is your feature request related to a problem? Please describe.**
I'm not sure if this is a bug report or simply a lost feature. I recall earlier versions of Clight in which whenever I woke my computer from suspend it would automatically take a capture and adjust my backlight accordingly. It seems this is no longer the case.

**Describe the solution you'd like**
Clarification on if this is desired behavior and if so, work together on creating a custom module that reproduces this behavior to be included as a skeleton.

**Describe alternatives you've considered**
Simply running a capture manually or creating a module or systemd unit to do it for me.

FedeDP, 2020-11-01:
Hi!
That should indeed be the case right now.
Note that it will only trigger a new capture on resume if capture timeout has elapsed; was that the case?
eg: with 10mins timeout, suspend at 7:40PM, resume at 8:00PM should trigger a capture.

FedeDP, 2020-11-01:
You can see here that it works like expected:
```
(I)[18:01:11]{backlight.c:325}  Ambient brightness: 0.746 -> Backlight pct: 0.732.
(I)[18:09:53]{upower.c:63}      Laptop lid closed.
(I)[22:58:07]{upower.c:63}      Laptop lid opened.
(I)[22:58:08]{backlight.c:325}  Ambient brightness: 0.389 -> Backlight pct: 0.477.
```

NICHOLAS85, 2020-11-02:
Ah yes, it's working as described. I think I just didn't notice this new behavior. I might create a module which performs a scan when I open my lid as I usually change locations when closing and opening my laptop which may have not allowed enough time to elapse since the last scan, especially with the long timeout I use when on battery.

closed 2020-11-02 by NICHOLAS85

FedeDP, 2020-11-02:
As this was an open question in my mind (whether to trigger a new capture leaving suspended state), in 3f2eac6925e58c32af742db8f56f3f7db7638d8b i added a configuration option to trigger a new capture on lid opened. Hope you like that! :)

NICHOLAS85, 2020-11-02:
Thank you very much! The new config option does exactly what I was expecting and is quite nice to have.