FedeDP/Clight #161: [BUG] Failed to validate inhibit request

NICHOLAS85, 2020-11-05:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 3f2eac6925e58c32af742db8f56f3f7db7638d8b

**Used distribution:**
> Bedrock Linux 0.7.18 Poki

**Describe the bug**
Sometimes upon wake from suspension or hibernation, Clight loses the ability to validate inhibit requests. This breaks any behaviors that are supposed to be paused when ScreenSaver inhibition is being held as well as any modules that rely on inhibition requests.

**Expected behavior**
For this issue not to occur.

**To Reproduce**
I unfortunately haven't been able to reproduce this reliably, I only notice it occasionally towards the end of the day when I expect my INHIBIT_BL custom module to remove and set night gamma.

Please attach a Clight verbose log too!
[clight.log](https://github.com/FedeDP/Clight/files/5491065/clight.log)

FedeDP, 2020-11-06:
That is weird; the only thing it could happen is that the number of inhibitions requested is higher that number of inhibitions dropped; this clight is stuck in a "inhibited" state.
This is quite hard to debug as most probably many applications may be involved.

As a workaround, you should be able to unlock clight through its desktop file, or directly through dbus:
`busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Inhibit "b" false`

FedeDP, 2020-11-21:
Hi! Working on fixing #146  i have found the issue for this as well :) 
I will hopefully be able to commit in the next days! I just need to fix a small issue as soon as i got time.

FedeDP, 2020-11-28:
Hi! Now that #166  is merged, this bug is finally fixed!
Let me know if you ecounter any problem! I will close this for now; feel free to reopen if needed!

closed 2020-11-28 by FedeDP

NICHOLAS85, 2020-11-29:
Thanks! If I run into it again I'll be sure to reopen.