FedeDP/Clight #162: [BUG] Setting Pixel Format: Device or resource busy

Gert-dev, 2020-11-12:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
4.1

**Used distribution:**
Arch Linux x86-64

**Describe the bug**
I've been using clight for quite some time now and it works well (thanks for your efforts!). I recently switched to new hardware and it seems clightd is having some minor issues getting up and running after a cold boot:

```
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
     Loaded: loaded (/usr/lib/systemd/system/clightd.service; enabled; vendor preset: disabled)
     Active: active (running) since Thu 2020-11-12 16:52:03 CET; 4min 26s ago
   Main PID: 1241 (clightd)
      Tasks: 2 (limit: 19051)
     Memory: 792.0K
     CGroup: /system.slice/clightd.service
             └─1241 /usr/lib/clightd/clightd

nov 12 16:52:03 pc systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
nov 12 16:52:03 pc systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
nov 12 16:52:08 pc clightd[1241]: Setting Pixel Format: Device or resource busy
```

I first thought this originated from clightd, but if I restart `clightd`, the error disappears, though running `clight` from its `.desktop` file in GNOME still doesn't update the brightness. If I kill the existing `clight` instance, start a new one on the command line, only then everything works as expected.

Perhaps all that is required is some sort of timeout/sleep before clight does its initial measurements?

**Expected behavior**
The brightness adjusts itself after logging in or booting and continues to update itself automatically.

**To Reproduce**
Configure clightd as usual and to automatically start with the desktop.

FedeDP, 2020-11-12:
Hi! Thanks for opening this bug report!
Are you so brave to maybe update to clight-git and clightd-git? They have looots of improvements that i plan to release before end of year :)
Both masters should be quite stable indeed.
If the issue persists, we will proceed with some debugging!

> I first thought this originated from clightd, but if I restart clightd, the error disappears, though running clight from its .desktop file in GNOME still doesn't update the brightness. If I kill the [...]

First idea: clightd fails to use your camera and for some reason it hangs somewhere; clight that is waiting for a response from clightd, hangs too.
This is why killing both works.
What is weird is that clightd failed to use your camera but then after restart magically it can use it!

Gert-dev, 2020-11-14:
Thanks for your response!

I've updated to the Git versions of both, and tried rebooting, but the error still appears to be the same, unfortunately.

I've now also tried _not_ restarting clightd after this problem occurs, but just killing clight and restarting it, which also works. So it appears that the clightd error is just temporary, or maybe it's clight that is causing the problem.

I see this in the logs:

```
nov 14 13:35:52 pc systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
nov 14 13:35:52 pc systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
nov 14 13:35:57 pc clightd[1089]: Setting Pixel Format: Device or resource busy
nov 14 13:38:09 pc clightd[1089]: Registered 'Als' sensor.
nov 14 13:38:09 pc clightd[1089]: Registered 'Camera' sensor.
nov 14 13:38:09 pc clightd[1089]: Registered 'Custom' sensor.
nov 14 13:38:09 pc clightd[1089]: Registered 'YoctoLight' sensor.
nov 14 13:38:09 pc clightd[1089]: Registered 'Drm' gamma plugin.
nov 14 13:38:09 pc clightd[1089]: Registered 'Wl' gamma plugin.
nov 14 13:38:09 pc clightd[1089]: Registered 'Xorg' gamma plugin.
nov 14 13:38:09 pc clightd[1089]: Registered 'Drm' dpms plugin.
nov 14 13:38:09 pc clightd[1089]: Registered 'Wl' dpms plugin.
nov 14 13:38:09 pc clightd[1089]: Registered 'Xorg' dpms plugin.
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Creating client 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Adding inotify watch as first client was started.
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Starting Client 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Creating client 1
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Starting Client 1
nov 14 13:38:09 pc clightd[1089]: [default]|GAMMA|: Temperature target value set: 6500.
nov 14 13:38:09 pc clightd[1089]: [default]|GAMMA|: Reached target temp: 6500.
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Client 0 -> Idle: 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Client 0 -> Idle: 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Stopping Client 1
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Freeing client 1
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Removing inotify watch as only client using it was stopped.
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Stopping Client 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Freeing client 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Returning unused client 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Adding inotify watch as first client was started.
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Starting Client 0
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Returning unused client 1
nov 14 13:38:09 pc clightd[1089]: [default]|IDLE|: Starting Client 1
nov 14 13:38:09 pc clightd[1089]: [default]|GAMMA|: Temperature target value set: 6500.
nov 14 13:38:09 pc clightd[1089]: [default]|GAMMA|: Reached target temp: 6500.
```

You can see from the timestamps that at startup, it ended on `Setting Pixel Format: Device or resource busy`, and then a few minutes later I killed and restarted `clight` itself, which caused clightd to start operating as normal, strangely enough.

(Also, I was confused at first by the wording of "Registered 'Als' sensor", which appears to mean "Started 'Als' sensor backend" or something, since I don't appear to have any ALS or YoctoLight sensors, but I got slightly excited at first, thinking I _did_ in fact have an ALS I didn't know about :smile:.)

FedeDP, 2020-11-14:
Very weird! Maybe clightd is started before your webcam is fully loaded?
But i don't really understand why does clight freeze! Can you share a clight log?

> (Also, I was confused at first by the wording of "Registered 'Als' sensor", which appears to mean "Started 'Als' sensor backend" or something, since I don't appear to have any ALS or YoctoLight [...]

You are right! it is kind of misleading; i may change the strings :) thank you!

Gert-dev, 2020-11-15:
That is certainly possible.

Here is clight's log on the same moment as shown above, after I had killed and restarted it and everything started working:

```
(W)[13:38:04]{config.c:299}	Config file /home/me/.config/clight.conf not found.

Clight
* Software version:		4.2
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Sat Nov 14 13:38:04 2020

Starting options:

### GENERIC ###
* Verbose (debug):		Disabled

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth steps:		0.05
* Smooth timeout:		30
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Backlight path:		Unset
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled
* Capture on lid opened:		Disabled

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		Unset
* Settings:		Unset

### KEYBOARD ###
* Dim:		Disabled
* Threshold:		1.00

### GAMMA ###
* Smooth trans:		Enabled
* Smooth steps:		50
* Smooth timeout:		300
* Daily screen temp:		6500
* Nightly screen temp:		4000
* Long transition:		Disabled
* Ambient gamma:		Disabled
* Delay:		0

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800

### DIMMER ###
* Smooth trans:		ENTER: Enabled, EXIT: Enabled
* Smooth steps:		ENTER: 0.05, EXIT: 0.05
* Smooth timeout:		ENTER: 30, EXIT: 30
* Timeouts:		AC 45	BATT 20
* Backlight pct:		0.20

### DPMS ###
* Timeouts:		AC 900	BATT 300

### SCREEN ###
* Timeouts:		AC 30	BATT -1
* Contrib:		0.10
* Samples:		10
[... 116 lines cut ...]
### SCREEN ###
* Timeouts:		AC 30	BATT -1
* Contrib:		0.10
* Samples:		10

### INHIBIT ###
* Docked:		Disabled
* PowerManagement:		Disabled

(I)[19:54:33]{main.c:137}	Clightd found, version: 5.0.
(I)[19:54:33]{keyboard.c:71}	Keyboard backlight calibration unsupported.
(I)[19:54:33]{upower.c:52}	AC cable connected.
(I)[19:54:33]{upower.c:63}	Laptop lid opened.
(I)[19:54:33]{location.c:60}	New location received: 50.96, 5.50.
(I)[19:54:33]{daytime.c:190}	Next alarm due to: Mon Nov 16 07:27:00 2020
(I)[19:54:33]{gamma.c:139}	Normal transition to 4000 gamma temp.
```

I'm still running clight-git and clightd-git at this time.

I've also did some inspecting of the processes: both clight and clightd are now running, but clight doesn't start updating my backlight, running the Clight desktop entry from GNOME shell also does not cause clight to update anything.

After restarting clight, clight updates the backlight after a few seconds and responds when starting the Clight desktop entry from GNOME shell, as intended.

This is probably expected, but on my previous hardware, where everything worked on boot, the webcam LED did activate briefly to update the brightness, whilst now it does not. This indeed makes me suspect that the webcam is not yet "initialized" or something at that point in time.

FedeDP, 2020-11-15:
Ok i think i spotted the issue!
When clight starts, asks clightd if any [Sensor](https://github.com/FedeDP/Clightd/wiki/Sensors) is available.
Clightd then looks for available devices; for each device it finds, it has a validation code; for Camera sensor, the validation code just checks that it is a real streaming device and it does support any of the clightd-supported image formats; you can see it here: https://github.com/FedeDP/Clightd/blob/master/src/modules/sensors/camera.c#L203.

Unfortunately, it also tries to set image formats; this is the action that is failing for you (here: https://github.com/FedeDP/Clightd/blob/master/src/modules/sensors/camera.c#L255).
Most probably, at that stage, your camera drivers fails to correctly set pixel fmt, maybe it is just a driver bug.

Consequently, clightd reports that no available sensors were found and clight does not calibrate your screen until it gets restarted or a sensor is attached, eg: an usb webcam.

I think that explanation is pretty technical, sorry for that; hope you followed me.

BUT, even it that seems like a webcam driver issue, i still think that clightd should not set anything on webcam during validation phase, because that is logically wrong!
I will move image-fmt setting code before actual capturing from device; hopefully this will fix your issue as well.
Anyway, you allowed me to spot a stupid issue in clightd code :) thank you very much!

FedeDP, 2020-11-15:
Latest clightd now has the aforementioned fix! Let me know if this fixes your bug!

Gert-dev, 2020-11-16:
This does seem to fix it for me, thanks! Great that you were able to spot the issue :smile:.

closed 2020-11-16 by Gert-dev

FedeDP, 2020-11-16:
I am super happy :) thank you to allow me to find this issue!