FedeDP/Clight #163: [FEATURE REQ]

linusrg1, 2020-11-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. EG: I'm always frustrated when [...]
As I and I know alot of other people use this method to simulate auto brightness adjust on laptops without light sensors, lots of us are not really comfortable changing things from a config file. 
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
It would be nice if there was a ui we could use to configure clight

linusrg1, 2020-11-12:
Config file /home/linus/.config/clight.conf not found.
Clightd found, version: 5.0.
Keyboard backlight calibration supported.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
AC cable disconnected.
Laptop lid opened.
Error loading from cache file.
I get that now so I dont think it is starting correctly. And after uncommenting shutter threshold what would you suggest changing its value to to have it constantly constantly keep the camera on.

And on a slightly different note will clight interfere with video chats?

FedeDP, 2020-11-13:
It would be really great to have some kind of gui around clight, and it should be quite feasable as clight exposes a [pretty dbus interface](https://github.com/FedeDP/Clight/wiki/Api).
Unfortunately, i have no such time to make that gui. It would be really great if anyone steps forward and makes it!

FedeDP, 2020-11-14:
I'll leave this one open and lingering, if anyone is interested in making a gui!

fuxx, 2021-01-06:
Do you have already any ideas what kind of feature set the gui should at least have?
Would it be a standalone application or would it be nicer to be integrated within Gnome?

FedeDP, 2021-01-06:
Hi! 
I was thinking more of an applet, something like a gnome extension or kde plasma applet.
I imagined a small applet to eg: do a quick calibration, change some settings (like screen backlight and temperature, or inhibit DIMMER), and with a small summary of clight state (eg: "DAY" "Next Event: Sunset @ 6.27pm", "Currently Inhibited").
Something very basic. I am not that good at GUIs :) 
Even if i know Qt well enough, i haven't much time to spend on my side projects lately.
If you could provide any help, it would be greatly appreciated! Thank you!

nullobsi, 2021-01-20:
I am currently working on a GUI for Clight at https://github.com/nullobsi/clight-gui
I'm not very good at c++ and this is my first Qt project but it currently does some basic work!

FedeDP, 2021-01-20:
Wow man this is great!
Let me know if you need some support for dbus api or need some new apis! 
This is a wonderful news, really really thanks!

FedeDP, 2021-01-20:
I built it! It's really great, i like it: simple yet super effective!
 I'd suggest to allow building with qt5 too maybe :) qt6 is not widely adopted for now.
Btw really thank you again!
Looking forward to its improvements! I am watching the repo ;)

nullobsi, 2021-01-20:
Thank you!
I am currently working on getting tabs to configure each aspect of the API, once that is done I will look into how to make it a tray applet. But you have done a very good job on the DBus APIs, everything is exposed!

I used Qt6 because it was the latest but I think you are right. It should be fairly easy to change to Qt5 so I will do that later :)

nullobsi, 2021-01-22:
I have implemented the important parts of the bus API. Up next is figuring out how to make it a tray applet.

FedeDP, 2021-01-22:
I have literally pulled each commit :) that's a high hype here!
It is really beautiful, what can i say! You have a really sweet design taste!
Considered i worked with qt for years and... well i don't really like dealing with user interfaces :)

nullobsi, 2021-01-22:
thank you! The tray applet works now, and I am quite proud :)
But if there is anything to be improved, please let me know!

FedeDP, 2021-01-22:
Well i'll leave some suggestions!
* I'd actually link clight in your readme (and i'll be very happy to mention clight-gui in clight's readme too!)
* May be add a Pause action to the tray icon?
* If you're looking for an app icon/ tray icon, you can have a look here: https://iconduck.com/search?query=webcam
* It would be great to remove "Open Window" from tray, and instead opening window on click, closing to tray on another click
* It would be great to be able to ctrl-q to completely quit the program 
* I'd add a cmdline switch to start clight-gui in background (to tray, ie: current behaviour) and by default i'd open the window too; it seems a common behaviour among other apps!
* In main window, i'd add a conf "Store" button, to actually store all changes made to the conf!

Those are just some hints :) Let me know what do you think!

nullobsi, 2021-01-22:
thanks for the suggestions! I'll be sure to add a link to clight
there is file>store button, but originally I added a store button at the bottom. i think both wouldn't hurt though :)

FedeDP, 2021-01-22:
A couple of ohter ideas:
* It seems like "delete" button on daytime conf for sunrise/sunset does not work!
* Adding a button to open the log file?

Thank you! It is shaping super good! :)

nullobsi, 2021-01-22:
You're welcome!

The delete button should write an empty string to the property. Testing with the dbusviewer and busctl (which doesn't work for some reason), it only changes when a valid time is written. Is there some special value that resets it?

FedeDP, 2021-01-23:
> The delete button should write an empty string to the property. Testing with the dbusviewer and busctl (which doesn't work for some reason), it only changes when a valid time is written. Is there some special value that resets it?

Ops! Clight validation code is too strict here :) Fixed it!

Btw i think you can lower cmake required version, as eg: on my work laptop (ubuntu 20.04) it fails:
```
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
CMake 3.17 or higher is required.  You are running version 3.16.3

-- Configuring incomplete, errors occurred!
```
All in all, right now it works pretty good! Once again, i really love the gui design, super clean and effective!
We need a nice tray icon!! :D
Hopefully i'll release this new clight version with the new fixes asap, i'd like to deal with #174 before!

FedeDP, 2021-01-23:
Btw i consider this issue closed; for any more suggestion / issues i'll be sure to open a request on your repo!

Thank you very much!

closed 2021-01-23 by FedeDP