FedeDP/Clight #164: [BUG] Error when starting cmake

linusrg1, 2020-11-12:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> ...5.0

**Used distribution:**
> ...Pop!_OS

**Describe the bug**
A clear and concise description of what the bug is.
When I try to start cmake this happens

Config file /home/linus/.config/clight.conf not found.
Clightd found, version: 5.0.
Keyboard backlight calibration supported.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
AC cable disconnected.
Laptop lid opened.
Error loading from cache file.
I get that now so I dont think it is starting correctly.
**Expected behavior**
A clear and concise description of what you expected to happen.
For cmake to start correctly and start changing my brightness.
**To Reproduce**
Steps to reproduce the behavior.
Start clight 5 on pop
Please attach a Clight verbose log too!
I would if I knew how, sry.

FedeDP, 2020-11-13:
I don't see any issue in there! What is the issue you are seeing?
Btw: i think you meant "clight" instead of "cmake" :) 
And...clight 5.0 does not exist! You are referring to clightd 5.0 (that is clightd devel version).

linusrg1, 2020-11-13:
ok that makes sense.

On a slightly different note what would i set shutter threshhold to to make the camera always change the brightness?

FedeDP, 2020-11-14:
> On a slightly different note what would i set shutter threshhold to to make the camera always change the brightness? 

I already responded on #157  :)

closed 2020-11-14 by FedeDP