FedeDP/Clight #165: [FEATURE REQ] Support for dimming screen based on location or time of day.

H-Hubert, 2020-11-21:
I really like clight, but one of my laptops does not have a built in camera. I think it would be useful to have the option to dim the display at night based on location or time of day when a camera is unavailabe. Clight already adjusts gamma based on time or location, so it seems like it might be fairly easy to implement this feature. I know I could plug in an external webcam, but that would be rather inconvienient for me.  I am not much of a coder, but I appreciate all the work that you have put into this great little application. Thank you :)

FedeDP, 2020-11-21:
Hi! Thanks for opening the issue!
Well, while i can see your feature request's purpose, i don't think clight should handle this itself. It may seem super-easy but it would require yet more config options (at least 2 i guess: one to enable this mode for BACKLIGHT, and another to specify day/night/on_ac/on_batt backlight levels) plus the logic to manage.

It is instead super- easy to add this as a [custom module](https://github.com/FedeDP/Clight/wiki/Custom-Modules): this should quite do the trick!

```
#include <clight/public.h>

CLIGHT_MODULE("INHIBIT_BL");

DECLARE_MSG(bl_req, BL_REQ);

static void init(void) {
    bl_req.bl.smooth = -1;  // use conf values
    
    /* Subscribe to inhibit state */
    M_SUB(DAYTIME_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case DAYTIME_UPD: {
        daytime_upd *up = (daytime_upd *)MSG_DATA();
        if (up->new == NIGHT) {
            bl_req.bl.new = 0.5; // 50% backlight level
            INFO("It is night: set a lower backlight level to save my eyes!\n");
        } else {
            bl_req.bl.new = 1.0; // 100% backlight level
            INFO("It is daytime: set highest backlight level!\n");
        }
        M_PUB(&bl_req);
        break;
    }
    default:
        break;
    }
}
```

H-Hubert, 2020-11-21:
It worked! Thank you so much!!! :)

FedeDP, 2020-11-21:
Great to hear!! Hopefully it was easy to understand how to build and use the custom module :)
You can even expand it much more than that, if you ever wish to do so!

H-Hubert, 2020-11-21:
Yes. I followed the directions in the top of the example module usr/share/clight/nightmode.skel  Thank you.

closed 2020-11-21 by H-Hubert