FedeDP/Clight #167: [FEATURE REQ] Better installation instructions, ALS attaching instructions

Code7R, 2020-11-25:
**Is your feature request related to a problem? Please describe.**
I tried to install and configure your package by following common sense and the Wiki contents.
On a Debian system, not Arch.

**Describe the solution you'd like**
A better HOWTO-like list of installation instructions.
Which tells the user where and how to start configuring it.
Also provide with hints how to run it without a DE (i.e. explain what to do and not hide that in some XDG startup .desktop file, because not everybody uses GNOME and friends).
Explain what "ALS" means and how I can connect my own ALS (assuming that is stands for Ambient Light Sensor).

Actually I have created my own cheap&simple sensor and I would like to connect it somehow to existing software (like yours); it's built on a Arduino Leonardo with a photo sensor attached and it sends current brightness value as ASCII numbers over usb-serial protocol, every second. I can share the Arduino sketches and build instructions in case anyone cares.

**Describe alternatives you've considered**
Originally, considered writing my own daemon which controls ddcutil (or maybe via https://gitlab.com/ddcci-driver-linux/ddcci-driver-linux , you could also have a look at this as the easier option compared to gddcutil).

FedeDP, 2020-11-25:
Hi! Thanks for opening this issue!
>  A better HOWTO-like list of installation instructions.

Mmmh mind to tell me which part should be expanded? https://github.com/FedeDP/Clight/wiki/Build
It seems pretty straightforward for me, but obviously i am biased :) 

> Also provide with hints how to run it without a DE

https://github.com/FedeDP/Clight/wiki/Start, first lines!

>  Explain what "ALS" means and how I can connect my own ALS (assuming that is stands for Ambient Light Sensor).

I think ALS is a pretty easily googlable, much established acronym, isn't it? 
Anyway i can probably expand it to make it more understandable!

> how I can connect my own ALS (assuming that is stands for Ambient Light Sensor).
Actually I have created my own cheap&simple sensor and I would like to connect it somehow to existing software (like yours)

This seems very nice! Clight backend daemon ([clightd](https://github.com/FedeDP/Clightd)) does indeed support ALS devices capture but your device should expose linux IIO interface.
Otherwise, you are also able to use a text file as fake als sensor; it was purposedly implemented targeting third part ALS devices (ie: your use case!); have a look here: https://github.com/FedeDP/Clightd/wiki/Sensors#custom and here: https://github.com/FedeDP/Clightd/wiki/Api#orgclightdclightdsensor .

>  (or maybe via https://gitlab.com/ddcci-driver-linux/ddcci-driver-linux , you could also have a look at this as the easier option compared to gddcutil).

I am following ddcci-driver-linux development but unfortunately it is not very actively developed. Moreover, it does not work in many cases. Again, i am tracking it in clightd: https://github.com/FedeDP/Clightd/blob/master/TODO.md#ideas

Code7R, 2020-11-25:
> Mmmh mind to tell me which part should be expanded? https://github.com/FedeDP/Clight/wiki/Build
> It seems pretty straightforward for me, but obviously i am biased :)

Yeah, building is just one part. How to run it? How to put into ~/.Xsession or similar (or when do you not to care about?). I would expect this in some classic "INSTALL" chapter but there is none. BUILD seems to be the one but those instructions cannot be found there.

> https://github.com/FedeDP/Clight/wiki/Start, first lines!

Actually this is the very last chapter but it should probably be right beneath "Config". Maybe also rename to "Startup" or "Desktop Integration". And the chapter you call "DE Integration", that one I would call "DE Automation".

And I am sorry, I have not read so far. I scanned the Wiki for "Startup" or "Desktop Integration" or "Integration", found "DE Integration" and stopped reading because it felt like the documentation is incomplete.

> I think ALS is a pretty easily googlable, much established acronym, isn't it?

a) maybe not so easy, because b) an established acronym it is but probably not for something you probably had in mind. It's a disease, a very bad one. Member "ALS Ice Bucket Challenge" a few years ago? IMHO googling stuff is not a good idea, Google filter-bubbles you. Even if I search with more keywords by stupid guessing, I find HID_SENSOR_ALS from the kernel config or the old ACPI driver in hid-sensor-als.c but it doesn't feel right. I would have expected some Device Class name but no luck.

> This seems very nice! Clight backend daemon (clightd) does indeed support ALS devices capture but your device should expose linux IIO interface.
> Otherwise, you are also able to use a text file as [...]

Okay, THAT is the Wiki I was looking for. I was stranded at https://github.com/FedeDP/Clight/wiki and I had NO CLUE about 
 https://github.com/FedeDP/Clightd/wiki . Maybe it should be cross-linked prominently, like, called "Developer Corner" in the user wiki? Do you want me to send some Wiki PR here?

Anyway, I would prefer a userspace implementation for my HW interaction for now. Not because I am afraid of adding a kernel module, but looking at with my Debian Maintainer hat on: integrating 3rd party kernel drivers is always messy work.

BTW, I just checked https://github.com/FedeDP/Clightd/wiki/Sensors#custom-1 and it looks intriguing. Just wondering, can this be extended to have a "lazy" option instead of a fixed polling interval, i.e. driven only by inotify notifications? If so, do you want a patch?

Also BTW, I am wondering why use $CMAKE_INSTALL_FULL_DATADIR/... for the module location and not $CMAKE_INSTALL_FULL_LIBDIR . I guess those modules are shared libs loaded at runtime, and not data.

FedeDP, 2020-11-25:
> Actually this is the very last chapter but it should probably be right beneath "Config".

It actually is the third, right below Build and before Config: https://github.com/FedeDP/Clight/wiki

>  Maybe also rename to "Startup" or "Desktop Integration". And the chapter you call "DE Integration", that one I would call "DE Automation".

That seems like great ideas! Thank you!

> a) maybe not so easy, because b) an established acronym it is but probably not for something you probably had in mind. It's a disease, a very bad one. Member "ALS Ice Bucket Challenge" a few years [...]

Wow you are right! Did not realize! I will surely explain it better!

> Maybe it should be cross-linked prominently, like, called "Developer Corner" in the user wiki? 

Yeah, may be i should add it.

> Anyway, I would prefer a userspace implementation for my HW interaction for now. Not because I am afraid of adding a kernel module, but looking at with my Debian Maintainer hat on: integrating 3rd party kernel drivers is always messy work.

I understand that, that's why i added a CUSTOM sensor in Clightd :) 

> BTW, I just checked https://github.com/FedeDP/Clightd/wiki/Sensors#custom-1 and it looks intriguing. Just wondering, can this be extended to have a "lazy" option instead of a fixed polling interval,

Unfortunately that is not easy;
Let me explain the full architecture: Clightd exposes some dbus API to access various display-related functions; it exposes a Sensor API too that allows to return ambient brightness through various sources;
Clight uses Clightd API to do its work; as Clight was born with webcam sensors in mind, Clightd api does only allow to pull data from sensors.
While i think it could be useful to add a "Changed" signal for Sensor API, it would be quite hard to implement it on any supported sensor.
It would break any clight usage (ie: in clight you can customize timeouts between captures) too; again, keep in mind that clight was born with laptops cameras in mind.

In the end, i may give it a try for next release cycle (5.0 is nearing release); the idea sounds quite intriguing to me :)
Can you open an issue on clightd for this?

> Also BTW, I am wondering why use $CMAKE_INSTALL_FULL_DATADIR/... for the module location and not $CMAKE_INSTALL_FULL_LIBDIR . I guess those modules are shared libs loaded at runtime, and not data.

Well, that is a nice question; most probably i just used the wrong directory when releasing clight 4.0!

FedeDP, 2020-11-25:
BTW i should have addressed your wiki concerns! Let me know if you spot anything else!

And if you are really interested, i can give you write access to wiki pages; it is super hard to actually maintain them; actually, i often just update API page before releasing a new clight version.

FedeDP, 2020-11-28:
Btw note that under "Sensor -> devname" settings in clight you can specify sensor that will be used with clightd: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L91
This is the exact param that will be forwarded in Sensor API requests to clightd; this means Clight can support your CUSTOM sensor out of the box.

FedeDP, 2020-12-22:
Hopefully all the questions were answered and the wiki pages updated for better.
Thank you very much, closing this one!

closed 2020-12-22 by FedeDP