FedeDP/Clight #170: [CUSTOM] Custom module help

NICHOLAS85, 2020-12-04:
**Custom Module's behaviour**
Upon entering an inhibited state:
- Change night gamma to 6500
- Set keyboard brightness to 50% if currently above 50%
- Set a keyboard backlight timeout of 2 seconds
- Pause autocalibration
Upon exiting an inhibited state:
- Change night gamma to 4200
- Set keyboard backlight timeout to 10 seconds
- Perform a quick calibration, also sets keyboard brightness
- Start autocalibration

**Custom Module's code**

<details>
<summary>inhibit_bl3.c</summary>

```C
#include <clight/public.h>
#include <stdlib.h>

/**
 * Small example custom module for Clight.
 *
 * It just hooks on INHIBIT state updates and:
 * -> when entering INHIBIT state (eg: start watching a movie), disables automatic calibration,
 *    set low kbd brightness and timeout, and pauses any gamma adjustment
 * -> when leaving INHIBIT state, triggers a new backlight calibration and re-enables automatic calibration,
 *    unpauses gamma adjustment, and restores kbd timeout
 **/

/*
 * Rename to: inhibit_bl3.c
 *
 * Build with: gcc -shared -fPIC inhibit_bl3.c -o inhibit_bl3 -Wno-unused
 *
 * Place inhibit_bl_mod in: $HOME/.local/share/clight/modules.d/ OR, globally, in /usr/share/clight/modules.d/
 */

CLIGHT_MODULE("INHIBIT_BL");

DECLARE_MSG(capture_req, CAPTURE_REQ);
DECLARE_MSG(calib_req, NO_AUTOCALIB_REQ);
DECLARE_MSG(kbd_bl_req, KBD_BL_REQ);
DECLARE_MSG(temp_req, TEMP_REQ);
double cont = 1.0;

static void init(void) {
    capture_req.capture.reset_timer = false; // avoid resetting clight internal BACKLIGHT timer
    temp_req.temp.smooth = -1;               // use current smooth mode
    temp_req.temp.daytime = NIGHT;           // only affect night gamma

    /* Subscribe to inhibit and kbd_bl state */
    M_SUB(INHIBIT_UPD);
    M_SUB(KBD_BL_UPD);
}

static void kbd_timeout(char* timeout) {
    char cmd[128];
    snprintf(cmd, sizeof(cmd), "echo %s | sudo /usr/bin/tee /sys/devices/platform/dell-laptop/leds/dell::kbd_backlight/stop_timeout", timeout);
    system(cmd);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case KBD_BL_UPD: {
        bl_upd *up = (bl_upd *)MSG_DATA();
        cont = up->new;
        break;
    }
    case INHIBIT_UPD: {
        inhibit_upd *up = (inhibit_upd *)MSG_DATA();
        calib_req.nocalib.new = up->new;
        if (up->new) {
            INFO("Pausing autocalibration and night color.\n");
            temp_req.temp.new = 6500;                 // day gamma value
            kbd_bl_req.bl.new = 0.5;                  // 50% kbd brightness
            if (cont >= 0.75) { M_PUB(&kbd_bl_req) }; // set 50% kbd brightness if kbd above 50%
            kbd_timeout("2s");                        // set 2 second kbd backlight timeout
        } else {
            INFO("Doing a quick backlight calibration and unpausing night color.\n");
            temp_req.temp.new = 4200; // night gamma value
            kbd_timeout("10s");       // set 10 second timeout, causes keyboard to wake
            M_PUB(&capture_req);      // ask for a quick calibration, sets kbd to proper brightness
        }
        M_PUB(&temp_req);  // set gamma value
        M_PUB(&calib_req); // stop or start backlight autocalibration
        break;
    }
    default:
        break;
    }
}

```
</details>

Current issues:
The gamma adjustment works great except when clight is in an Event and gammas long_transition is enabled. If inhibition starts during an event, the gamma is set to 6500, and upon exiting inhibition set to 4200. I'd like to set the gamma back to its long transition value and allow the transition to continue as normal.

Currently, Clight doesn't manage keyboard backlight timeout. Doing some research it doesn't seem there's a standard dbus interface to adjust this either. Do you happen to know a better way of modifying this value compared to the current methods employed in my module? Due to using sudo, it needs a specific entry in sudoers to run without a password.

If you have any suggestions I'd love to hear them! Thanks.

FedeDP, 2020-12-05:
> If inhibition starts during an event, the gamma is set to 6500, and upon exiting inhibition set to 4200. I'd like to set the gamma back to its long transition value and allow the transition to continue as normal.

That's the correct behaviour as far as i can tell! As you are forcing different temperatures from extern, clight does stop long transitioning.
You'd like instead to go on the long transition but with different transition starting and target temp?

> Currently, Clight doesn't manage keyboard backlight timeout.

You mean keyboard dimming period? If that's the case, there is currently no way to reach that in clight, as UPower does not offer it either; see here: https://upower.freedesktop.org/docs/KbdBacklight.html
Clight could just offer such a feature by setting keyboard backlight to 0 after a timeout.

NICHOLAS85, 2020-12-05:
>  You'd like instead to go on the long transition but with different transition starting and target temp?

I basically want to stop any gamma correction (The setting to 6500) but resume it as normal after inhibition is released. So if a transition is occurring and the gamma is 5500, I'd want it to change to 6500 during inhibition and then back to 5xxx afterward and let Clight continue its long transition, instead of how I currently immediately set it to 4200. If this isn't currently possible that's alright, was just curious if you would know how to achieve that logic.

> You mean keyboard dimming period? If that's the case, there is currently no way to reach that in clight, as UPower does not offer it either; see here: upower.freedesktop.org/docs/KbdBacklight.html
Clight could offer such a feature by setting keyboard backlight to 0 after a timeout.
I don't really like this solution though as if keyboard internal 'stop_timeout' is lower, it would surely cause lots of headaches to users 

This is true, fighting with the keyboards internal 'stop_timeout' would probably cause a lot of issues. Hopefully one day in the future UPower could control this and offer a standard way of adjusting it. Thanks for the response!

FedeDP, 2020-12-06:
> I basically want to stop any gamma correction (The setting to 6500) but resume it as normal after inhibition is released. So if a transition is occurring and the gamma is 5500, I'd want it to change [...]

Oh i now understand!
Mmmh it seems pretty hard to internally recognize that when entering inhibited mode (note that gamma know nothing, it does only know that you ask it to set 6500 temp) gamma should set the asked temp without further checks, but when leaving inhibited mode it should instead resume long transition (even if the request is the same!).
Or may be a "bool force_now" could be added to temp_req, and if that is set, it will set temp immediately, otherwise it will follow current state (ie: long_transitioning).
I will check if this is feasible without too much hassle :)

FedeDP, 2020-12-11:
After a second thought, i think you can reach desired behaviour by setting temp 6500 immediately, and then when leaving inhibited mode, you can set 4200 temp with a very long transition, eg: { smooth = 1, step = 50, timeout = 500ms }

NICHOLAS85, 2020-12-12:
I'll see if that works and report back but here are some initial concerns.
A) This doesn't respect what the current gamma value should be after exiting inhibition. If I'm understanding correctly it'd start from 6500 and transition to 4200 even if clight would've normally been transitioning from 5400 to 4200 at that point in time. (5400 being just an example value below 6500)
B) I don't need the long transition when clight is no longer in an 'event'

B I could solve by just checking if clight is in an event and adjusting the behavior accordingly, A I'll have to think about further when I have some time.
Thanks for helping me out of course!

FedeDP, 2020-12-12:
Well you can subscribe to TEMP_UPD and IN_EVENT_UPD and you should have all the infomations you need!
For TEMP_UPD, note that clight-git does now send a TEMP_UPD message with target temp (eg: 4200), and then a TEMP_UPD for each step; you might then expect something like:
* TEMP_UPD 4200
* TEMP_UPD 6450
* TEMP_UPD 6400
* TEMP_UPD 6350
...

This allows you to restart a smooth transition from last temp!

FedeDP, 2020-12-12:
Indeed it was not the case until a couple of months ago :) let me know if you need some more indications!

NICHOLAS85, 2020-12-16:
> For TEMP_UPD, note that clight-git now sends a TEMP_UPD message with target temp (eg: 4200), and then a TEMP_UPD for each step

This doesn't seem to be working on my machine, I'm only seeing the initial TEMP_UPD message with the target temp but no subsequent TEMP_UPD's

FedeDP, 2020-12-16:
Uh you are right, it is not properly emitted! I will fix it asap!

FedeDP, 2020-12-18:
Issue should now be fixed in master!

NICHOLAS85, 2020-12-19:
Awesome! looks like it's working now. Some things I've noticed: temp_upd.timeout is only valued in the first notification for a regular transition and has a value of 0 for subsequent updates. temp_upd.timeout is only valued 0 for long transitions, including the first notification.

That first thing I'm just double checking if it's expected behavior, the second thing I believe may be a bug.

On another note, is there a way to queue `M_PUB()`'s? At the moment I am attempting to restore pre-inhibit gamma value, and then start the long transition from that point. Here's an example snippet code:
```C
// set gamma to pre-inhibit value
temp_req.temp.new = real_temp; // This is the gamma value before inhibition
M_PUB(&temp_req);
// long transition to night value from this point
temp_req.temp.new = 4200;      // night gamma value
time(&timer);
auto int remaining_time = (end_time - timer );
temp_req.temp.step = (remaining_time/1000) ;
temp_req.temp.timeout = 10000;
M_PUB(&temp_req);              // set gamma value
```
It seems only the last `M_PUB()` request is actually running.

FedeDP, 2020-12-19:
> temp_upd.timeout is only valued in the first notification for a regular transition and has a value of 0 for subsequent updates. temp_upd.timeout is only valued 0 for long transitions, including the first notification.

The first one is wanted: you should receive a "smooth transition started" temp_upd with all details for the smooth transition too, then each step is not "smooth", it is just a step.
The second one may be a bug; i don't understand though, care to explain it further? Thanks!

> On another note, is there a way to queue M_PUB()'s? At the moment I am attempting to restore pre-inhibit gamma value, and then start the long transition from that point. Here's an example snippet code:

Yup, as DECLARE_MSG() is declaring a single msg on stack, your code is indeed publishing the same message twice with 2 different values; it is sent as a reference (thus no copy happens) and receiver will then receive 2 messages that points to the same memory address, whose value was updated before sending the second request.
Ie: you send twice the same request.

Clight-git public.h does now offer a DECLARE_HEAP_MSG() macro, that allows declaring a message on heap; the message is autofreed after all recipients received it.
Thus, you should use it differently from DECLARE_MSG, something like:

```
// set gamma to pre-inhibit value
DECLARE_HEAP_MSG(first_req, TEMP_UPD);
first_req.temp.new = real_temp; // This is the gamma value before inhibition
M_PUB(first_req);

// long transition to night value from this point
DECLARE_HEAP_MSG(second_req, TEMP_UPD);
second_req.temp.new = 4200;      // night gamma value
time(&timer);
auto int remaining_time = (end_time - timer );
second_req.temp.step = (remaining_time/1000) ;
second_req.temp.timeout = 10000;
M_PUB(second_req);              // set gamma value
```
I should add this to clight wiki page indeed!

NICHOLAS85, 2020-12-19:
> The first one is wanted: you should receive a "smooth transition started" temp_upd with all details for the smooth transition too, then each step is not "smooth", it is just a step.

Understood!

>The second one may be a bug; i don't understand though, care to explain it further? Thanks!
Sounds 

I'll see if I can make an understandable bug report for this and open a separate issue for it once I find the time.

> Clight-git public.h does now offer a DECLARE_HEAP_MSG() macro ...

Ahh okay cool! So I'll just share what I've got so far:

<details>
<summary>inhibit_mod.c</summary>

```c
#include <clight/public.h>
#include <stdlib.h>
#include <time.h>

/**
 * Small example custom module for Clight.
 *
 * It just hooks on INHIBIT state updates and:
 * -> when entering INHIBIT state (eg: start watching a movie), disables automatic calibration,
 *    set low kbd brightness and timeout, and pauses any gamma adjustment
 * -> when leaving INHIBIT state, triggers a new backlight calibration and re-enables automatic calibration,
 *    unpauses gamma adjustment, and restores kbd timeout
 **/

/*
 * Rename to: inhibit_mod.c
 *
 * Build with: gcc -shared -fPIC inhibit_mod.c -o inhibit_mod -Wno-unused
 *
 * Place inhibit_bl_mod in: $HOME/.local/share/clight/modules.d/ OR, globally, in /usr/share/clight/modules.d/
 */

CLIGHT_MODULE("INHIBIT_MOD");

DECLARE_MSG(capture_req, CAPTURE_REQ);
DECLARE_MSG(calib_req, NO_AUTOCALIB_REQ);
DECLARE_MSG(kbd_bl_req, KBD_BL_REQ);

bool in_event = false;
bool first_run_temp = true;
bool first_run_event = true;
int default_step = 50;
int default_timeout = 300;
double kbd_lev = 1.0;
int saved_temp = 6500;

time_t timer;
int end_time;

static void init(void) {
    capture_req.capture.reset_timer = false; // avoid resetting clight internal BACKLIGHT timer

    M_SUB(INHIBIT_UPD);  // inhibit state
    M_SUB(KBD_BL_UPD);   // keyboard backlight levels
    M_SUB(IN_EVENT_UPD); // Clight event state
    M_SUB(TEMP_UPD);     // temperature levels
}

static void kbd_timeout(char* timeout) {
    char cmd[128];
    // Add following line to sudoers, replace USERNAME
    // USERNAME ALL=(root) NOPASSWD: /usr/bin/tee /sys/devices/platform/dell-laptop/leds/dell\:\:kbd_backlight/stop_timeout
    snprintf(cmd, sizeof(cmd), "echo %s | sudo /usr/bin/tee /sys/devices/platform/dell-laptop/leds/dell::kbd_backlight/stop_timeout", timeout);
[... 83 lines cut ...]
    }
    case IN_EVENT_UPD: {
        in_event = ! in_event;
        if (in_event) {
            time(&timer);
            end_time = ( timer + 3600); // determine when event will end for transition calculations
        } else {
            first_run_event = true; // reset first_run for next round
        }
        break;
    }
    case KBD_BL_UPD: {
        bl_upd *up = (bl_upd *)MSG_DATA();
        kbd_lev = up->new;
        break;
    }
    default:
        break;
    }
}
```
</details>
One thing to note, I used a `req->temp.new` syntax when assigning the temp variables instead of `req.temp.new` because I was getting `'...' is a ponter; did you mean to use '->'?` errors.

It seems I'm doing something wrong. Lets ignore all the if(in_event) stuff under the INHIBIT_UPD for now. I'd expect my gamma value to toggle between `6500` and `4200` when inhibited and not inhibited. However, this is not occurring. When the message is published, my `TEMP_UPD()` INFO message has the following message: `New temp: 6500, Saved temp: 4200, Step: 4200, Timeout: 0` instead of what I would expect: `New temp: 6500, Saved temp: 4200, Step: 50, Timeout: 300`. It's as if the variables I'm assigning aren't being passed correctly. If you could help me out here that'd be great!

FedeDP, 2020-12-20:
> I'll see if I can make an understandable bug report for this and open a separate issue for it once I find the time.

I fixed the issue! Guess what? i was using DECLARE_MSG() too instead of DECLARE_HEAP_MSG() but updates from clightd arrived too quick thus i was sending messages with wrong values all over :D 
This is the same source of bug of INHIBIT counter going crazy, if you remember the issue where clight always remained inhibited.

> One thing to note, I used a req->temp.new syntax when assigning the temp variables instead of req.temp.new because I was getting '...' is a pointer; did you mean to use '->'? error

Yup that's right!

For other issue: can you retry after the new fixes? You were probably receiving wrong values!

NICHOLAS85, 2020-12-20:
Awesome!
> I'll see if I can make an understandable bug report for this and open a separate issue for it once I find the time.

So I'll describe that issue a little more in depth as I believe it's still present:
I've added the following snippet for `TEMP_UPD` updates (saved_temp being just a variable for use in other parts of the module):
```C 
INFO("New temp: %d, Saved temp: %d, Step: %d, Timeout: %d\n", up->new, saved_temp, up->step, up->timeout)
```
This outputs all the expected information, including timeout during first message and subsequent updates having just temp and step info. However when a long transition begins (only during an event with `long_transition = true;`), the first message is missing any timeout info, it has a value of 0. This isn't important to my module, but is a small discrepancy in expected behavior.

NICHOLAS85, 2020-12-20:
> It seems I'm doing something wrong. Lets ignore all the if(in_event) stuff under the INHIBIT_UPD for now.

This also still seems to be occurring. Some details:
```c
switch (MSG_TYPE()) {
    case INHIBIT_UPD: {
        inhibit_upd *up = (inhibit_upd *)MSG_DATA();
if(up->new){
	DECLARE_HEAP_MSG(pause_req, TEMP_UPD);
    pause_req->temp.daytime = NIGHT;           // only affect night gamma
    pause_req->temp.new = 6500;                // day gamma value
    pause_req->temp.smooth = -1;               // use current smooth mode
    pause_req->temp.step = default_step;       // default step
    pause_req->temp.timeout = default_timeout; // default timeout
    M_PUB(pause_req);
} else {
	DECLARE_HEAP_MSG(fast_resume_req, TEMP_UPD);
    fast_resume_req->temp.daytime = NIGHT;           // only affect night gamma
    fast_resume_req->temp.new = 4200;                // night gamma value
    fast_resume_req->temp.smooth = -1;               // use current smooth mode
    fast_resume_req->temp.step = default_step;       // default step
    fast_resume_req->temp.timeout = default_timeout; // default timeout
    M_PUB(fast_resume_req);
}
	break;
	}
	default:
    break;
    }
```
From the above, I expect to transition to 6500 and 4200 during inhibition and non-inhibition. This worked well using `DECLARE_MSG` however it seems using `DECLARE_HEAP_MSG` like above nothing is occurring. I see a single `TEMP_UPD` with all the correct information however no further updates are iterated and the gamma stays the same.

FedeDP, 2020-12-21:
Hi! A couple of quick notes:
> int default_step = 50;
int default_timeout = 300;

You can avoid setting this: when passing smooth = -1 in request, it will load smooth params from config!

> using DECLARE_HEAP_MSG like above nothing is occurring

Obviously, you know i am an idiot! :D 
> DECLARE_HEAP_MSG(pause_req, TEMP_UPD);

I pasted the wrong message type: you are publishing an UPDate and not a REQuest!!
Switch to TEMP_REQ and it should be fixed; sorry!

> The first message is missing any timeout info, it has a value of 0

This is very weird, as you can see here: https://github.com/FedeDP/Clight/blob/master/src/modules/gamma.c#L172, the published TEMP_UPDate should have all the fields set.
Timeout should be fixed to 10000ms when long transitioning.

NICHOLAS85, 2020-12-21:
>  Switch to TEMP_REQ and it should be fixed; sorry!

Huzzah! I've finally got a working module! Thanks so much for all the help since I began working on this

> You can avoid setting these ones: when passing smooth = -1 in request, it will load smooth params from config!

Noted!

>This is very weird, as you can see here: https://github.com/FedeDP/Clight/blob/master/src/modules/gamma.c#L172, the published TEMP_UPDate should have all the fields set.
Timeout should be fixed to 10000ms when long transitioning.

I'll see if I can find any helpful info for this and open a new issue.

I'm going to clean up some of this module's code and test it in all the cases I'd expect it to operate, for now here's the gist: https://gist.github.com/NICHOLAS85/c68e5f47d5b6bf1f6b061d30f705ebc2
A quick thing to note, I wasn't able to use two subsequent `MPUB()`s for the temp req as the second one would interrupt the first before it finished which I didn't want. So I moved the second one into `TEMPUPD` and made it execute once the temp has reached a certain target value. Looks like it is working just like I wanted it to.

FedeDP, 2020-12-21:
> A quick thing to note, I wasn't able to use two subsequent MPUB() s for the temp req as the second one would interrupt the first before it finished which I didn't want. So I moved the second one [...]

Yup, the fact is that it is up to clightd to actually manage the smooth transition, and it is smart enough to correctly manage a temp request on same $DISPLAY, ie: continuing the smooth transition but with different params and to different target temp :)

> 
I'll see if I can find any helpful info for this and open a new issue

That would be great, thanks! I plan to release clight before Xmas so we are nearing the release date!

NICHOLAS85, 2020-12-21:
Interesting, I can't reproduce that timeout issue anymore. Maybe something I did in prior iterations of the module was returning the data wrong. For now, it seems to be running correctly. Looking forward to the release!

closed 2020-12-21 by NICHOLAS85

FedeDP, 2020-12-22:
Btw, as a sidenote, i opened a merge request on UPower to add support for "stop_timeout" keyboard properties: https://gitlab.freedesktop.org/upower/upower/-/merge_requests/58.
Let's hope it gets accepted and merged!

NICHOLAS85, 2020-12-23:
> Btw, as a sidenote, i opened a merge request on UPower to add support for "stop_timeout" keyboard properties: https://gitlab.freedesktop.org/upower/upower/-/merge_requests/58.
> 
> Let's hope it gets accepted and merged!

Awesome, thanks for looking into that! Hopefully it gets merged indeed

FedeDP, 2021-02-05:
Unfortunately the upower PR is not going to be merged because UPower Keyboard module needs some internal rework prior to adding any new feature. 
But the ongoing work is somewhat stopped since last year.

>  Bastien Nocera @hadess · 1 day ago Maintainer
We can't really add more functionality to the keyboard backlight until the API is fixed (see !25) or just completely replaced by a separate helper daemon, just as systemd offers for display backlights.

Consequently, i decided to ditch UPower KbdBacklight API and instead implementing a new API in Clightd.
I still need to add documentation but it is pretty straightforward, you will have /org/clightd/clightd/KbdBacklight/$any_found_kbd; on each kbd you have some methods { Get, Set, GetTimeout, SetTimeout } and a constant property {MaxBrightness}.

Latest clight commit: d8478996e4928a57c3724fec2d0826fbe507b54f implemented keyboard timeout on top of Clightd new API.

FedeDP, 2021-02-05:
Added to clightd wiki: https://github.com/FedeDP/Clightd/wiki/Api#orgclightdclightdkbdbacklight

hadess, 2021-02-05:
>  But the ongoing work is somewhat stopped since last year.

I mean, patches welcome. But I guess it was easier to reimplement things directly into your daemon.

FedeDP, 2021-02-05:
Yup, sorry didn't mean to bother you with the mention!

NICHOLAS85, 2021-02-06:
Thanks for the update! I was able to update my module to take advantage of the new api

FedeDP, 2021-02-06:
I forgot to mention i also added a clight option for that, under keyboard settings!