FedeDP/Clight #172: [FEATURE REQ] Add a post-capture Sensor settings value

NICHOLAS85, 2020-12-11:
**Is your feature request related to a problem? Please describe.**
Currently, we can add Sensor settings for our captures using the `{sensor}/settings` config option. I find this useful to turn off auto exposure and set a constant exposure value for consistent readings. However afterward these settings persist and I must turn auto exposure back on manually for use in other applications.

**Describe the solution you'd like**
I'd like to add the option of setting sensor options after a capture completes so I can turn auto exposure back on automatically. 

**Describe alternatives you've considered**
I could try creating a custom module but it seems this behavior might be useful to have in Clight itself.

FedeDP, 2020-12-11:
Hi! Thanks for opening this!
>  I'd like to add the option of setting sensor options after a capture completes so I can turn auto exposure back on automatically.

May be the most simple thing is to just let clightd automatically revert settings to default at end of captures. 
Indeed, now that i think about this, it was pretty stupid not to do this in the first place.
What do you think?

NICHOLAS85, 2020-12-12:
This seems like a possible solution. However, I can see users not liking their camera setting modifications being overwritten each time Clight finishes a capture. Maybe saving current values before capture and then restoring the saved values afterward could work. I am unsure how much complexity that would add of course and if it would be a viable solution.

FedeDP, 2020-12-12:
Yeah that's the best idea!
Look forward for next clightd commit (hopefully today!).
Thanks!

FedeDP, 2020-12-12:
Ok, latest clightd master has the new fix :)

NICHOLAS85, 2020-12-12:
Just rebuilt and it looks like it's working perfectly! Thanks for the quick fix

closed 2020-12-12 by NICHOLAS85