FedeDP/Clight #173: Can it work with colorimeters?

sotiris-bos, 2021-01-05:
Hello! I own an X-rite Colormunki Display colorimeter for calibrating displays.

The X-rite software under Windows provides automatic brightness control based on measurements of ambient light taken from the colorimeter at regular intervals.

I have not found any similar functionality on Linux.

Is this something that is/could be supported by clight?

FedeDP, 2021-01-05:
Hi! Thanks for opening the issue!
It seems like linux kernel does indeed support those devices: https://wiki.analog.com/software/linux/docs/iio/iio#iio_overview.

I am not sure this can be easily added to clightd; i need some informations; can you paste here output of:
* ls /sys/bus/iio/devices
* udevadm info -a /sys/bus/iio/devices/iio:deviceX

Where latter "X" can be found from ls command above.  
If this is successful, it should be fairly trivial to add support for it in clightd. 
Btw I am not sure what i am supposed to do with sensor readings; i think that won't be so simple to implement.
Thanks!

sotiris-bos, 2021-01-05:
Thanks for the reply! I am afraid there isn't a /sys/bus/iio directory on my system.
These devices are supported by the Linux kernel but they are USB devices. They can be used with DisplayCAL.
If you need them to be accessible through iio I am afraid that they can't be used for Clight.

Here is some info:

```
$ lsusb -v
Bus 005 Device 006: ID 0765:5020 X-Rite, Inc. i1 Display Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0765 X-Rite, Inc.
  idProduct          0x5020 i1 Display Pro
  bcdDevice            0.01
  iManufacturer           1 X-Rite, Inc.
  iProduct                2 ColorMunki Display
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

```
udevadm info -a /sys/bus/usb/devices/5-3

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/pci0000:40/0000:40:07.1/0000:42:00.3/usb5/5-3':
    KERNEL=="5-3"
    SUBSYSTEM=="usb"
    DRIVER=="usb"
    ATTR{authorized}=="1"
    ATTR{avoid_reset_quirk}=="0"
    ATTR{bConfigurationValue}=="1"
    ATTR{bDeviceClass}=="00"
    ATTR{bDeviceProtocol}=="00"
    ATTR{bDeviceSubClass}=="00"
    ATTR{bMaxPacketSize0}=="64"
    ATTR{bMaxPower}=="100mA"
    ATTR{bNumConfigurations}=="1"
    ATTR{bNumInterfaces}==" 1"
    ATTR{bcdDevice}=="0001"
    ATTR{bmAttributes}=="c0"
    ATTR{busnum}=="5"
    ATTR{configuration}==""
    ATTR{devnum}=="6"
    ATTR{devpath}=="3"
    ATTR{devspec}=="(null)"
    ATTR{idProduct}=="5020"
    ATTR{idVendor}=="0765"
    ATTR{ltm_capable}=="no"
    ATTR{manufacturer}=="X-Rite, Inc."
    ATTR{maxchild}=="0"
    ATTR{power/active_duration}=="816577"
    ATTR{power/async}=="enabled"
    ATTR{power/autosuspend}=="2"
    ATTR{power/autosuspend_delay_ms}=="2000"
    ATTR{power/connected_duration}=="816577"
    ATTR{power/control}=="on"
    ATTR{power/level}=="on"
    ATTR{power/persist}=="1"
    ATTR{power/runtime_active_kids}=="0"
    ATTR{power/runtime_active_time}=="816310"
    ATTR{power/runtime_enabled}=="forbidden"
    ATTR{power/runtime_status}=="active"
    ATTR{power/runtime_suspended_time}=="0"
    ATTR{power/runtime_usage}=="1"
    ATTR{product}=="ColorMunki Display"
    ATTR{quirks}=="0x0"
    ATTR{removable}=="unknown"
    ATTR{rx_lanes}=="1"
    ATTR{speed}=="12"
    ATTR{tx_lanes}=="1"
    ATTR{urbnum}=="16"
    ATTR{version}==" 2.00"
```

I think this is a bit of a stretch to implement support but let me know if you need any more info.
Thanks a lot!

FedeDP, 2021-01-06:
Uh! Thanks for the outputs!
Well, as far as i can tell, if it falls under usb subsystem, i guess it must have some sort of proprietary usb protocol to fetch info from device; looking for more informations i found: https://github.com/beku/Argyll-Releases/blob/master/spectro/i1d3.c that should be right code for your device (found looking at "idVendor" and "idProduct" in https://github.com/beku/Argyll-Releases/blob/master/usb/55-Argyll.rules#L31.
It seems like a ton of code; but it does not seem to use libusb, thus i am not sure that is the correct one.
If you can find some sources to support it, i will gladly look at them!
I will also look around again when i find some time!

FedeDP, 2021-02-25:
I cannot find the time to look for more info, could you?
I fear that this is too time consuming to implement and mantain as it is though.

closed 2021-04-25 by FedeDP

volfyd, 2022-03-02:
I have an X-rite i1display pro and I wrote up this script to drive 'spotread' to save the ambient light value to a file every ten seconds. This will probably work with other X-rite products or anything spotread understands.

```
#!/bin/env tclsh

package require Expect

set timeout 10

spawn spotread -a
expect {
    -re {Result is XYZ: \d+\.\d+ (\d+)\.} {
        set Y $expect_out(1,string)

        set fd [open "/tmp/clight-custom-sensor" "w"]
        puts $fd "$Y\n"
        close $fd

        exp_continue
    }
    timeout {
        exp_send "\n"
        exp_continue
    }
}
```
And then you can use this by running something like 
```
clight --verbose --device=/tmp/clight-custom-sensor -f1
```

FedeDP, 2022-03-02:
Wow man! This is super cool!!🚀
I will add it to the wiki for sure! 
Thanks for sharing this solution :)
Finally Clightd CUSTOM Sensor has a valid usage!

FedeDP, 2022-03-02:
Updated wiki: https://github.com/FedeDP/Clight/wiki/Colorimeters

Thank you!!