FedeDP/Clight #174: [FEATURE REQ] Please add a man page or --help entry

brent-holland, 2021-01-16:
Maybe I've missed something, but I've read the README.md, I've read your wiki.  I've tried both man clight and clight --help.  NONE of these tell me how to use your program.  The most basic example:  How do I change the brightness of my display from the command line?  Please add a man page or a --help flag and outline basic usage.

FedeDP, 2021-01-16:
Hi! Thanks for opening the issue! 
`--help` cmdline switch does indeed explain all clight parameters. 
Anyway, while clight has a ton of options, running it is just as easy as run "clight" in your terminal.
> How do I change the brightness of my display from the command line?

Clight does its work in an autonomous manner; but you can talk with him through its [dbus](https://github.com/FedeDP/Clight/wiki/Api) interface.

Normally, you should not care about it anyway, and just let clight do its magic in the background.

brent-holland, 2021-01-16:
Ah I see, you check for an already running instance of clight _before_ checking your command line options.  The help wasn't displaying because clight started automatically.  When I killed it I did get the --help message.

Maybe you can read your command line options before checking for a running instance?

FedeDP, 2021-01-17:
> Maybe you can read your command line options before checking for a running instance?

Unfortunately it is not so simple: at startup clight does:
* open log file 
* log file is also the clight lock file, ie: if another instance is locking it, clight will fail and leave with aforementioned message
* load conf from conf file
* load cmdline params

As you can see, cmdline params are parsed quite late in the startup process. 
Moreover, i need log to be opened before parsing conf, as any configuration error must be logged.

All in all, may be i can just improve clight "another instance is running" message?

brent-holland, 2021-01-17:
It's a bit odd to me that the priority is loading the log and config files first, and the command line options (clo) last.  In most programs the clo override all other config and are processed first.  Having said that, this is your baby and you raise it as you see fit.  At minimum I think you could throw the --help output into a man page for a first iteration.  I'm not sure if there's a standard for man pages or if that would rile the spirit of Richard Stallman, but it'd be better than nothing :^)

You can close this request if you like.

FedeDP, 2021-01-17:
>  In most programs the clo override all other config and are processed first.

Yep, as you said, they *override* the config, thus in my mind, they're loaded later in the process to override the loaded config. 
This allows me to skip lots of checks while loading config like `if conf.X != X_default_value // not overridden by clo`.

It could be great to add a man page anyway, it seems like a great idea!
Any pr is welcomed btw :) 
Thank you, i'll leave this open for now!

commit 595e2a01 (2021-02-06): Added a man page. Fixes #174.
closed 2021-02-06 by FedeDP