FedeDP/Clight #175: [BUG] attempting to get Conf/Daytime/Sunrise or Sunset after setting causes a SIGSEGV

nullobsi, 2021-01-22:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.2

**Used distribution:**
> Arch Linux

**Describe the bug**
When using the DBus interface to set the Daytime module override for the sunrise or the sunset, attempting to retrieve the value set causes the program to SIGSEGV. The Sunrise/Sunset on the org.clight.clight interface works fine.

**Expected behavior**
When overriding the Daytime module via DBus, I expect the program to return the newly overwritten value attempting to retrieve it again.

**To Reproduce**
Use qdbusviewer or an equivalent to set and then retrieve Conf/Daytime/Sunrise or Sunset values.

```
Wrong AC frames value. Resetting default value.                                                                                                                                            [34/530]
Wrong BATT frames value. Resetting default value.
Clightd found, version: 5.0.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **
|     *
|    *
|   *
|  *
|
| *
|*
|
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|
|
|       ***
|     **
|    *
|   *
|  *
|
|**
|
*---------->BR
init_kbd_backlight(): パス /org/freedesktop/UPower/KbdBacklight のオブジェクトにインターフェース“org.freedesktop.UPower.KbdBacklight”がありません
Keyboard backlight calibration unsupported.
Disarmed timerfd on fd 38.
AC cable disconnected.
Laptop lid opened.
Emitting 'AcState' property
Disarmed timerfd on fd 41.
Emitting 'LidState' property
#### loaded from cache file.
Loc distance: #### km.
New location received: ####.
Set timeout of 0s 1ns on fd 42.
Emitting 'Location' property
Next alarm due to: Fri Jan 22 06:19:00 2021
Set timeout of 40617s 0ns on fd 42.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
Normal transition to 4000 gamma temp.
Set timeout of 0s 1ns on fd 43.
Sensor '/dev/iio:device2' is now available.
Resumed as a sensor is now available.
Emitting 'NextEvent' property
Emitting 'Temp' property
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/iio:device2'. Ambient brightness: 0.808524.
Ambient brightness: 0.809 -> Backlight pct: 0.757.
Set timeout of 5400s 0ns on fd 43.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
#### received from Geoclue2.
Backlight level updated: 0.76.
Loc distance: #### km.
Failed to validate location request.
Set timeout of 0s 1ns on fd 42.
Next alarm due to: Fri Jan 22 06:19:00 2021
Set timeout of 40561s 0ns on fd 42.
Emitting 'Sunrise' property
Normal transition to 4000 gamma temp.
Emitting 'Sunset' property
Emitting 'Temp' property
Received sigsegv signal. Aborting.
fish: 'clight --verbose' terminated by signal SIGSEGV (Address boundary error)
```

FedeDP, 2021-01-22:
Thanks for opening this issue!
It should now be solved in latest master! 
And obviously, thanks for creating clight-gui helping me test clight dbus api!

closed 2021-01-22 by FedeDP

nullobsi, 2021-01-22:
thank you! i'll test it out :)

nullobsi, 2021-01-22:
yup, it appears to be fixed; thank you!

commit ba3832fa (2021-01-22): Fix #175.