FedeDP/Clight #176: [BUG] calling Pause method causes stack smashing error and SIGABRT

nullobsi, 2021-01-22:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.2

**Used distribution:**
> Arch Linux

**Describe the bug**
Calling the Pause method on the DBus interface causes the program to crash with a stack smashing error.

**Expected behavior**
The Pause method should inhibit all features of Clight.

**To Reproduce**
Use qtdbusviewer or equivalent to call the Pause method.

```
Wrong AC frames value. Resetting default value.                                                                                                                                            [27/666]
Wrong BATT frames value. Resetting default value.
Clightd found, version: 5.0.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **
|     *
|    *
|   *
|  *
|
| *
|*
|
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|
|
|       ***
|     **
|    *
|   *
|  *
|
|**
|
*---------->BR
init_kbd_backlight(): パス /org/freedesktop/UPower/KbdBacklight のオブジェクトにインターフェース“org.freedesktop.UPower.KbdBacklight”がありません
Keyboard backlight calibration unsupported.
Disarmed timerfd on fd 38.
AC cable disconnected.
Laptop lid opened.
Emitting 'AcState' property
Disarmed timerfd on fd 41.
Emitting 'LidState' property
#### loaded from cache file.
Loc distance: #### km.
New location received: ####.
Set timeout of 0s 1ns on fd 42.
Emitting 'Location' property
Next alarm due to: Fri Jan 22 06:19:00 2021
Set timeout of 39963s 0ns on fd 42.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
Normal transition to 4000 gamma temp.
Set timeout of 0s 1ns on fd 43.
Sensor '/dev/iio:device2' is now available.
Resumed as a sensor is now available.
Emitting 'NextEvent' property
Emitting 'Temp' property
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/iio:device2'. Ambient brightness: 0.795394.
Ambient brightness: 0.795 -> Backlight pct: 0.752.
Set timeout of 5400s 0ns on fd 43.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
#### received from Geoclue2.
Backlight level updated: 0.75.
Loc distance: #### km.
Failed to validate location request.
*** stack smashing detected ***: terminated
fish: 'clight --verbose' terminated by signal SIGABRT (Abort)
```

FedeDP, 2021-01-22:
Hi!
Yep, this hit me a couple of times but i could never find the root cause; in my case, it happens very rarely; is this happening always for you?
Btw i now have a clight instance with -DCMAKE_BUILD_TYPE=asan running, and the following script continuously triggering Pause calls:
```
#!/bin/bash

while true;
do
    pidof clight
    if [ "$?" = "1" ]
    then
          break
    fi
    busctl --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" true
    sleep 5
    busctl --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" false
    sleep 5
done
```

nullobsi, 2021-01-22:
It happens every time for me, trying your script too it breaks immediately
I also tried building with `asan` option and it also crashes, so it seems to be an easily reproducible issue

strangely, using cmake debug and calling pause has no such issue; i'm able to pause and unpause without any issues? but the release builds crash

FedeDP, 2021-01-22:
> It happens every time for me, trying your script too it breaks immediately

That's great! May be it is simpler to debug on your machine indeed!
>  I also tried building with asan option and it also crashes, so it seems to be an easily reproducible issue

`asan` options should provide a stack trace like report after the crash, is this the case? If not, can you test with:
`cmake .. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_FLAGS="-fsanitize=address"` ? Thanks!

> strangely, using cmake debug and calling pause has no such issue; i'm able to pause and unpause without any issues? but the release builds crash

yup it seems like it is related to how the code is built, ie: how are variables aligned or whatsoever... static code analyzers (cppcheck) do not provide any help unfortunately.

FedeDP, 2021-01-22:
Can you try again with current master?

nullobsi, 2021-01-22:
yup, let me pull

nullobsi, 2021-01-22:
it appears that it doesn't crash on release now! :)

closed 2021-01-22 by nullobsi