FedeDP/Clight #179: a way to inhibit clight when using a photo manager ?

promeneur, 2021-02-04:
Is there a cli command to inhibit clight when using a photo manager ?

Thanks

FedeDP, 2021-02-04:
Hi! What do you intend to inhibit? Clight autocalibration? Dimming?
Btw right click on desktop file and clight provides some actions ready for you!

promeneur, 2021-02-05:
I did not get any desktop file on my desktop. Where is it supposed to be ?

I want just the same thing which happens when I display a video with vlc or firefox.

FedeDP, 2021-02-05:
> I want just the same thing which happens when I display a video with vlc or firefox.

Ok! if you look for clight in your DE menu you should find it! Then right click on it and select "Pause Screen Dimmer"!

promeneur, 2021-02-05:
It's not exactly what I need.

I want to write a script like this.

pause clight (no brigthness,no gamma management)
launch digikam
unpause clight

promeneur, 2021-02-05:
With redshift I had this script

pkill -usr1 redshift (trun off)
digikam
pkill -usr1 redshift (turn on)

FedeDP, 2021-02-05:
Ok! In clight you can do:
```
busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" true
digikam
busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" false
```

promeneur, 2021-02-05:
Wow, very techie ! Thanks.

closed 2021-02-05 by FedeDP