FedeDP/Clight #180: ambient gamma ?

promeneur, 2021-02-05:
A question about "ambient gamma".

I heard that some tv adapt brightness and color, using icc profile and the ambient brightness, to get the best color.

Is it the case of "ambient gamma"

FedeDP, 2021-02-05:
Hi! Somewhat: ambient_gamma enables clight to match screen temperature to ambient brightness, ie: the lower the ambient brightness, the lower the screen temperature.
This feature helps to reduce eyestrain in any occasion, not only during the night.

promeneur, 2021-02-05:
Eyestrain occurs also viewing a video. Is "ambient gamma" turned off viewing a video ?

FedeDP, 2021-02-05:
>  Is "ambient gamma" turned off viewing a video ?

No it is not; it follows backlight changes, thus when backlight calibration is enabled, ambient gamma does its job.

closed 2021-02-05 by FedeDP