FedeDP/Clight #181: ambient gamma always minimum temperature

promeneur, 2021-02-05:
"Ambient gamma" feature does not adjust gamma to ambient brightness.

see the log

(I)[12:56:41]{keyboard.c:71}	Keyboard backlight calibration unsupported.
(W)[12:56:41]{interface.c:396}	Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[12:56:41]{interface.c:397}	Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
(I)[12:56:41]{upower.c:95}	Not a laptop device. Killing UPower module.
(W)[12:56:41]{upower.c:45}	Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
(I)[12:56:41]{location.c:62}	New location received: 44.86, -0.58.
(I)[12:56:41]{daytime.c:175}	Next alarm due to: Fri Feb  5 17:42:00 2021
(I)[12:56:44]{backlight.c:322}	Ambient brightness: 0.565 -> Backlight pct: 0.773.
(I)[12:56:44]{gamma.c:180}	Normal transition to 3500 gamma temp.
(I)[13:06:47]{backlight.c:320}	Ambient brightness: 0.848 (-0.017 screen compensation) -> Backlight pct: 0.951.
(I)[13:06:47]{gamma.c:180}	Normal transition to 3500 gamma temp.

FedeDP, 2021-02-05:
It's working fine here:
```
Ambient brightness: 0.720 -> Backlight pct: 0.892.
Normal transition to 4000 gamma temp.
Normal transition to 6368 gamma temp.
Normal transition to 6243 gamma temp.
Normal transition to 6229 gamma temp.
Ambient brightness: 0.711 -> Backlight pct: 0.886.
Normal transition to 6229 gamma temp.
Normal transition to 6216 gamma temp.
```
Can you share your clight log? ($HOME/.local/share/clight/clight.log)

promeneur, 2021-02-05:
[clight.log](https://github.com/FedeDP/Clight/files/5933192/clight.log)

after testing i disabled "ambient light"

FedeDP, 2021-02-05:
Ok, i need a clight.log with ambient gamma enabled :) and enable verbose mode too (you can quickly test by killing clight and running `clight --verbose --ambient-gamma` from cmd line!)

promeneur, 2021-02-05:
[clight.log](https://github.com/FedeDP/Clight/files/5933495/clight.log)

FedeDP, 2021-02-05:
Ok! Can you try again with new master? Thank you!

promeneur, 2021-02-05:
Problem solved. Thanks.

In fact "ambient gamma" just shifts color to red ?

FedeDP, 2021-02-05:
> In fact "ambient gamma" just shifts color to red ?

Yes but instead of being sunrise/sunset based, it is ambient brightness based.

closed 2021-02-05 by FedeDP