FedeDP/Clight #182: clight still running for user whose kde session is closed

promeneur, 2021-02-05:
For user1 
open a kde session
close kde session

for user2 open a kde session

You can check that user1 clight is still running !

FedeDP, 2021-02-05:
Hi! Not clight' s fault: https://github.com/FedeDP/Clight/wiki/Startup#systemd-logind-killuserprocesses :)

FedeDP, 2021-02-05:
Btw note that Clightd requires an active polkit session to call setter methods; this means that a non-active clight won't be able to change backlight, gamma, dpms etc etc.

closed 2021-02-05 by FedeDP

promeneur, 2021-02-08:
There is actually a problem.

When kde session is closed for user2 then clicht continue to adjust brightness and fails and switches to some king of pause and do nothing if we again open a kde session for user2.

See the log

[clight_user2.log](https://github.com/FedeDP/Clight/files/5942232/clight_user2.log)