FedeDP/Clight #183: clight report brightness 0 % but brightness is 24 %

promeneur, 2021-02-06:
clight 4.3, clightd 5.2

clight-gui reports "brightness 0 %" in its contextual menu.

i open a bug report at clight-gui 

see the report

https://github.com/nullobsi/clight-gui/issues/8#

FedeDP, 2021-02-06:
Hi! It is working fine for me.
Can you share clight.log?

promeneur, 2021-02-06:
[clight.log](https://github.com/FedeDP/Clight/files/5936602/clight.log)

promeneur, 2021-02-06:
remember i use master branch.

FedeDP, 2021-02-06:
Btw you are setting some wrong config values:

> (W)[08:35:14]{opts.c:284}	Wrong AC frames value. Resetting default value.
(W)[08:35:14]{opts.c:288}	Wrong BATT frames value. Resetting default value.

I need a verbose log, enable verbose in config or through --verbose cmd line switch.

promeneur, 2021-02-06:
[clight.log](https://github.com/FedeDP/Clight/files/5936683/clight.log)

FedeDP, 2021-02-06:
I think i have found the root cause: it is a bug in Clightd: to emit Changed signals on each backlight step it monitors udev; but for external monitors only setup, udev won't report anything (it only sees internal backlight on laptops).
Thus your Clight was not receiving any backlight update signal.
I will fix this asap.

FedeDP, 2021-02-06:
I have no such environment (ie: a desktop pc) but hopefully latest Clightd should fix the issue. Care to test?

promeneur, 2021-02-06:
Problem solved. What about

(W)[13:50:55]{opts.c:284}	Wrong AC frames value. Resetting default value.
(W)[13:50:55]{opts.c:288}	Wrong BATT frames value. Resetting default value.

FedeDP, 2021-02-06:
Number of captures must be between 1 and 20! Check your config.
I added a comment in clight conf file to highlight the limit, thanks!

closed 2021-02-06 by FedeDP

promeneur, 2021-02-06:
i use 5 captures on AC and 5 on Battery.

FedeDP, 2021-02-06:
Can you share your clight.conf?

promeneur, 2021-02-06:
[clight.conf.log](https://github.com/FedeDP/Clight/files/5936935/clight.conf.log)

FedeDP, 2021-02-06:
WHoopsie that was another stupid bug! Fixed in latest commit!

promeneur, 2021-02-07:
> WHoopsie that was another stupid bug! Fixed in latest commit!

Problem solved with last master branch