FedeDP/Clight #184: [FEATURE REQ] support older versions of systemd

lgbaldoni, 2021-02-06:
Would it be very problematic to backport Clight for lder systemd releases?
Many stable (as in non rolling) distros are stuck at 234.

FedeDP, 2021-02-06:
Hi! Clight master already depends on systemd >= 234 (i recently added back support for opensuse 15.2 that ships it, fixing another feature req :)

FedeDP, 2021-02-06:
(I can't go lower than 234 because in 234 a PR of mine (https://github.com/systemd/systemd/pull/5753) was merged and it is needed by Clight.
Anyway i think that supporting a 4yrs old systemd is more than enough :)

closed 2021-02-06 by FedeDP

lgbaldoni, 2021-02-06:
Sorry, I had missed the commit that lowered it from 237: excellent news!

FedeDP, 2021-02-06:
Btw i plan a clight and clightd release soon!