FedeDP/Clight #186: viewing a video, adjust brigthness ?

promeneur, 2021-02-07:
I installed openSuse packages from alois.

clight 4.2
clightd 5.1

If I view a video then brightness is not restored to 100 % and still adjusted.

I understand this. We may still want to adjust brightness also if viewing a video (this is my case) but perhaps someone doesn't. Perhaps it would be a good thing to have the choice by setting a parameter like this

in_case_of_viewing_video_do_not_adjust_brightness_restore_it_to_100_% = yes

NICHOLAS85, 2021-02-07:
Checkout custom modules and inhibit-backlight: https://github.com/FedeDP/Clight/wiki/Custom-Modules#inhibit-backlight
I think it does what you're wanting

promeneur, 2021-02-07:
Thanks

Yes but this is for techie persons. I would want something for Mr Jo.

FedeDP, 2021-02-07:
Hi! Unfortunately I added a custom module for that because I did not want to mantain yet one use case more.
Any added code increments maintenance cost and time; moreover I think Clight already covers loooots of different use cases.
Custom module is the way to go :) see them as plugins; i could may be ship default custom modules already compiled (ready to be copied in custom modules dir)?

promeneur, 2021-02-07:
I understand. I suggest an easier way.
- a cli command to pause brightness adjusting and restore brightness to 100 %
- a cli command to resume brightness adjusting and to adjust immediately brightness

 for including in a script launching an app.

Thanks

promeneur, 2021-02-07:
When i say a cli command i think about the ones you supplied to me

busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" true

busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" false

FedeDP, 2021-02-07:
Ok! Then you can do:
`busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Backlight org.clight.clight.Conf.Backlight NoAutoCalib "b" true`
`busctl --user call org.clight.clight /org/clight/clight org.clight.clight IncBl "d" 1.0`

Then, to reset, you only need to enable back autocalib:
`busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Backlight org.clight.clight.Conf.Backlight NoAutoCalib "b" false`

promeneur, 2021-02-07:
Thanks. Very kind.

It would be a good idea to add this to your wiki. No ?

FedeDP, 2021-02-08:
https://github.com/FedeDP/Clight/wiki/DE-Automation#100-backlight-while-watching-movies :)

closed 2021-02-08 by FedeDP