FedeDP/Clight #187: viewing a video, adjust gamma ?

promeneur, 2021-02-07:
clight 4.2
clightd 5.1

ambient gamma enabled

If I view a video then gamma is not restored to 6500 K and still adjusted.
Ambient gamma or not who wants to see a video with red filter and not real colors ?

Please give the choice with some setting as these

ambient_gamma_in_case_of_viewing_video_do_not_adjust_gamma_restore_it_to_6500_K = yes
no_ambient_gamma_in_case_of_viewing_video_do_not_adjust_gamma_restore_it_to_6500_K = yes

Thanks

FedeDP, 2021-02-07:
Hi! Dbus API comes in help once more!
`busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 6500`

Then, to reset previous value:
`busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 4000`

In last command, just remember to change "4000" to whatever value you set in config file!

promeneur, 2021-02-07:
Thanks. Very kind.

It would be a good idea to add this to your wiki. No ?

promeneur, 2021-02-07:
This is to set Night temp. But we need to pause gamma adjusting and resume it.

FedeDP, 2021-02-07:
> This is to set Night temp. But we need to pause gamma adjusting and resume it.

Well once day and night gamma have same value, there won't be anything that changes gamma :)

FedeDP, 2021-02-07:
> It would be a good idea to add this to your wiki. No ?

I am planning to ask community for help; i don't need much help for code, but all auxiliar things like improve wiki, keep API doc up to date, update various completions scripts... 
I would love to let more people join me and help me deliver the best software i can!

promeneur, 2021-02-07:
I think about ambient gamma. If ambient gamma is not paused then 10 mn later it will adjust gamma. No ?

FedeDP, 2021-02-07:
You are right!
Ambient gamma works together with backlight calibration; thus if you mix this call with #186 method to disable autocalibration, nobody will touch your gamma anymore :)

closed 2021-02-08 by FedeDP

promeneur, 2021-02-09:
>> You are right!
>> Ambient gamma works together with backlight calibration; thus if you mix this call with #186 method to disable >> autocalibration, >> nobody will touch your gamma anymore :)

What i want is

case "ambient gamma"

- keep brightness adjusting
- pause color temperature adjusting
- launch an app
- quit app
- resume color temperature adjusting

 I fear that disabling "autocalibrating" disables brightness adjusting. No ?