FedeDP/Clight #189: [Wiki] DE Automation

yochananmarqos, 2021-02-08:
We completely forgot about Xfce! This should work:

```
#!/usr/bin/env python

import dbus
import subprocess
from gi.repository import GLib
from dbus.mainloop.glib import DBusGMainLoop

old_time = -1; # 0 -> day, 1 -> night

def handle_time_change(*args):
    global old_time
    if 'Time' in args[1] and old_time != args[1]['Time']:
        if args[1]['Time'] == 1:
            # GTK style
            subprocess.Popen(["xfconf-query -c xsettings -p /Net/ThemeName -s", "Matcha-dark-sea"])
            # Xfwm4 style
            subprocess.Popen(["xfconf-query -c xfwm4 -p /general/theme -s", "Matcha-dark-sea"])
            # Cursor
            subprocess.Popen(["xfconf-query -c xsettings -p /Gtk/CursorThemeName -s", "xcursor-breeze-snow"])
            # Icons
            subprocess.Popen(["xfconf-query -c xsettings -p /Net/IconThemeName -s", "Papirus-Dark-Maia"])
        elif args[1]['Time'] == 0:
            # GTK style
            subprocess.Popen(["xfconf-query -c xsettings -p /Net/ThemeName -s", "Matcha-sea"])
            # Xfwm4 style
            subprocess.Popen(["xfconf-query -c xfwm4 -p /general/theme -s", "Matcha-sea"])
            # Cursor
            subprocess.Popen(["xfconf-query -c xsettings -p /Gtk/CursorThemeName -s", "xcursor-breeze"])
            # Icons
            subprocess.Popen(["xfconf-query -c xsettings -p /Net/IconThemeName -s", "Papirus-Maia"])
        old_time = args[1]['Time']

DBusGMainLoop(set_as_default=True)
bus = dbus.SessionBus()
bus.add_signal_receiver(
    handle_time_change,
    'PropertiesChanged',
    'org.freedesktop.DBus.Properties',
    path='/org/clight/clight'
)

loop = GLib.MainLoop().run()
```

FedeDP, 2021-02-08:
> We completely forgot about Xfce! This should work:

Thank you! :D

FedeDP, 2021-02-08:
Added to the wiki :D

closed 2021-02-08 by FedeDP