FedeDP/Clight #190: dimmer, no smoothing

promeneur, 2021-02-08:
clight 4.2, clightd 5.1

smoothing is enabled for dimmer

There is no smoothing. The screen is instantly black.

FedeDP, 2021-02-08:
May be your external monitor is buffering requests and putting them together? I don't know if that's possible.
I can say that it works fine for me.
Maybe you can try to extend dimmer trans timeout (https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L223) to eg: 1s and see if that makes any difference?

promeneur, 2021-02-09:
i tried 5000 ms. Problem still there.

My monitor is an IIyama Prolite EE24B3HS.

The GPU is a CPU/GPU Intel HD Graphics 630 driven by i915.

FedeDP, 2021-02-10:
That is really weird, can you share a verbose log?

promeneur, 2021-02-11:
I  get a black screen at 10:32

[clight.log](https://github.com/FedeDP/Clight/files/5964172/clight.log)

FedeDP, 2021-02-11:
> (D)[10:27:51]{display.c:34}	Entering dimmed state...
(D)[10:27:51]{display.c:40}	A lower than dimmer_pct backlight level is already set. Avoid changing it.

It seems like you still have got the BlPct issue (it stays at 0.0); 
can  you check output of: `busctl --user introspect org.clight.clight /org/clight/clight`?
I need 
```
.AmbientBr                          property  d         0.992352     emits-change
.BlPct                              property  d         0.983373     emits-change
```
lines.
If BlPct is 0.0, please upgrade clight and clightd to their master. 
I am pretty sure it is 0.0 though because there is no line like (in my case):
> Backlight 'intel_backlight' level updated: 0.85.
Backlight 'G2LMTJ006007' level updated: 0.85.
Emitting 'BlPct' property

This means that Clight received `Backlight.Changed` signal from Clightd (you can see that in my case it is received for both internal laptop monitor (intel_backlight) and external monitor).

> I get a black screen at 10:32

The screen gets black because DPMS kicks in (you set a 5minutes timeout for DPMS); that is correct behavior.

promeneur, 2021-02-11:
.AmbientBr                          property  d         0.84872         emits-change
.BlPct                              property  d         0               emits-change

FedeDP, 2021-02-11:
Can you try to upgrade both Clight and Clightd?

promeneur, 2021-02-12:
I uopdated to master branch.

 I tried with 30 ms then no smoothing.

I will try with 5000 ms.

Note: just before getting a black screen clight suddenly consumes 100 % of one core. I had to terminate clight to stop this phenomenon.

promeneur, 2021-02-12:
dim duration 5000 ms

no smoothing

FedeDP, 2021-02-12:
Can you check again BlPct? Ps: did you reboot/restart clightd after upgrade?

promeneur, 2021-02-12:
.AmbientBr                          property  d         0.802544        emits-change
.BlPct                              property  d         0.925115        emits-change

yes
i stopped clicht adn clightd
i uninstalled opensuse package 
I installed branch master 
I started clightd and clight

FedeDP, 2021-02-12:
> .AmbientBr property d 0.802544 emits-change
.BlPct property d 0.925115 emits-change

Ok, now BlPct is correct! Can you share a new verbose log?

promeneur, 2021-02-13:
i use clight.conf to set trans to 5000 ms.
a verbose log

[clight.conf.txt](https://github.com/FedeDP/Clight/files/5975493/clight.conf.txt)
[clight.log](https://github.com/FedeDP/Clight/files/5975496/clight.log)

FedeDP, 2021-02-13:
Well... the log is fine now:

> (D)[11:19:48]{display.c:34}	Entering dimmed state...
(D)[11:19:49]{utils.c:56}	Pausing SCREEN
(D)[11:19:49]{interface.c:448}	Emitting 'DisplayState' property
(D)[11:19:49]{utils.c:56}	Pausing BACKLIGHT
(D)[11:19:49]{interface.c:448}	Emitting 'BlPct' property
(D)[11:19:49]{backlight.c:511}	Backlight '1127851325201' level updated: 0.91.
(D)[11:19:49]{interface.c:448}	Emitting 'BlPct' property
(D)[11:19:54]{backlight.c:511}	Backlight '1127851325201' level updated: 0.86.
(D)[11:19:54]{interface.c:448}	Emitting 'BlPct' property
(D)[11:19:59]{backlight.c:511}	Backlight '1127851325201' level updated: 0.81.
(D)[11:19:59]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:04]{backlight.c:511}	Backlight '1127851325201' level updated: 0.76.
(D)[11:20:04]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:10]{backlight.c:511}	Backlight '1127851325201' level updated: 0.71.
(D)[11:20:10]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:15]{backlight.c:511}	Backlight '1127851325201' level updated: 0.66.
(D)[11:20:15]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:20]{backlight.c:511}	Backlight '1127851325201' level updated: 0.61.
(D)[11:20:20]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:25]{backlight.c:511}	Backlight '1127851325201' level updated: 0.56.
(D)[11:20:25]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:30]{backlight.c:511}	Backlight '1127851325201' level updated: 0.51.
(D)[11:20:30]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:36]{backlight.c:511}	Backlight '1127851325201' level updated: 0.46.
(D)[11:20:36]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:41]{backlight.c:511}	Backlight '1127851325201' level updated: 0.41.
(D)[11:20:41]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:46]{backlight.c:511}	Backlight '1127851325201' level updated: 0.36.
(D)[11:20:46]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:51]{backlight.c:511}	Backlight '1127851325201' level updated: 0.31.
(D)[11:20:51]{interface.c:448}	Emitting 'BlPct' property
(D)[11:20:57]{backlight.c:511}	Backlight '1127851325201' level updated: 0.26.
(D)[11:20:57]{interface.c:448}	Emitting 'BlPct' property
(D)[11:21:02]{backlight.c:511}	Backlight '1127851325201' level updated: 0.21.
(D)[11:21:02]{interface.c:448}	Emitting 'BlPct' property
(D)[11:21:07]{backlight.c:511}	Backlight '1127851325201' level updated: 0.20.

As you can see Clightd is setting every step until 0.20 (dimmed backlight pct).
can you set back something like 50ms for the entering transition?

promeneur, 2021-02-13:
yes but there is no smoothing.

FedeDP, 2021-02-13:
That is very weird: if clightd emits the signal it means it actually tried to set correct backlight level.
I pushed a new commit that adds a log in clightd to check whether backlight setting works during smoothing; can you upgrade? 
Then, stop old service, and restart new one:
```
sudo systemctl stop clightd
sudo systemctl daemon-reload
sudo systemctl start clightd
```
To check clightd log:
`journalctl -u clightd.service -b -f`

You should see eg:
> feb 13 12:13:56 DPFXPS clightd[1375147]: [default]|BACKLIGHT|: G2LMTJ006007 reached target backlight: 0.99.
feb 13 12:13:56 DPFXPS clightd[1375147]: [default]|BACKLIGHT|: intel_backlight reached target backlight: 0.99.
feb 13 12:13:56 DPFXPS clightd[1375147]: [default]|BACKLIGHT|: Target pct: 0.99

In case of issues, you should see:
> "failed to set backlight for ..."
message.

Thank you!

promeneur, 2021-02-13:
i see only this

sudo journalctl -u clightd.service -b -f
[sudo] Mot de passe de root : 
-- Logs begin at Sat 2021-02-13 09:22:22 CET. --
févr. 13 12:31:41 grognon clightd[2937]: [default]|IDLE|: Leaving idle state.
févr. 13 12:31:41 grognon clightd[2937]: [default]|BACKLIGHT|: Target pct: 0.95
févr. 13 12:31:41 grognon clightd[2937]: [default]|GAMMA|: Temperature target value set: 6358.
févr. 13 12:31:41 grognon clightd[2937]: [default]|BACKLIGHT|: 1127851325201 reached target backlight: 0.95.
févr. 13 12:31:41 grognon clightd[2937]: [default]|GAMMA|: Reached target temp: 6358.
févr. 13 12:37:53 grognon systemd[1]: Stopping Bus service to manage various screen related properties (gamma, dpms, backlight)...
févr. 13 12:37:53 grognon clightd[2937]: [default]|IDLE|: Client 0 ->
févr. 13 12:37:53 grognon systemd[1]: Stopped Bus service to manage various screen related properties (gamma, dpms, backlight).
févr. 13 12:42:59 grognon systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
févr. 13 12:42:59 grognon systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).

promeneur, 2021-02-13:
[clight.log](https://github.com/FedeDP/Clight/files/5975587/clight.log)

promeneur, 2021-02-13:
No smoothing

FedeDP, 2021-02-13:
I need Clightd log; you can try the following:

```
killall clight
sudo systemctl stop clightd.service
// go into your clightd repository build folder
sudo ./clightd
(from another terminal) clight
```
You will now be able to provide me with full clightd output!
THank you very much!

FedeDP, 2021-02-13:
Uh i should have found the issue! CAn you upgrade Clightd and give this another try?

promeneur, 2021-02-13:
No smoothing.
[clight.log](https://github.com/FedeDP/Clight/files/5975801/clight.log)

sudo journalctl -u clightd.service -b -f
[sudo] Mot de passe de root : 
-- Logs begin at Sat 2021-02-13 09:22:22 CET. --
févr. 13 13:52:39 grognon clightd[29068]: [default]|BACKLIGHT|: 1127851325201 reached target backlight: 0.96.
févr. 13 13:52:39 grognon clightd[29068]: [default]|GAMMA|: Temperature target value set: 6370.
févr. 13 13:52:39 grognon clightd[29068]: [default]|GAMMA|: Reached target temp: 6370.
févr. 13 13:52:39 grognon clightd[29068]: [default]|IDLE|: Client 0 -> Idle: 0
févr. 13 13:52:39 grognon clightd[29068]: [default]|IDLE|: Client 1 -> Idle: 0
févr. 13 13:52:39 grognon clightd[29068]: [default]|IDLE|: Client 0 -> Idle: 0
févr. 13 13:52:39 grognon clightd[29068]: [default]|IDLE|: Client 1 -> Idle: 0
févr. 13 14:35:44 grognon systemd[1]: Stopping Bus service to manage various screen related properties (gamma, dpms, backlight)...
févr. 13 14:35:44 grognon clightd[29068]: [default]|IDLE|: Client
févr. 13 14:35:44 grognon systemd[1]: Stopped Bus service to manage various screen related properties (gamma, dpms, backlight).

FedeDP, 2021-02-13:
Does dimming work? Ie: does backlight level go to 0.20 after N seconds of inactivity?
Moreover, does normal backlight smooth change work? (Ie: if you cover the webcam with a hand to lower ambient brightness, does backlight smooths until new level?)

promeneur, 2021-02-13:
does backlight level go to 0.20 after N seconds of inactivity?

No i just perceive a red shift after 30 s of inactivity.

promeneur, 2021-02-13:
Moreover, does normal backlight smooth change work? (Ie: if you cover the webcam with a hand to lower ambient brightness, does backlight smooths until new level?)

It is difficult to perceive any smoothing. We see that it is darkier.

FedeDP, 2021-02-13:
Can you paste here output of:
`sudo ddcutil capabilities`
Thanks!

promeneur, 2021-02-13:
sudo ddcutil capabilities
[sudo] Mot de passe de root : 
Model: W2413
MCCS version: 2.0
Commands:
   Command: 01 (VCP Request)
   Command: 02 (VCP Response)
   Command: 03 (VCP Set)
   Command: 07 (Timing Request)
   Command: 0c (Save Settings)
   Command: 4e (unrecognized command)
   Command: f3 (Capabilities Request)
   Command: e3 (Capabilities Reply)
VCP Features:
   Feature: 02 (New control value)
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 08 (Restore color defaults)
   Feature: 0B (Color temperature increment)
   Feature: 0C (Color temperature request)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values:
         05: 6500 K
         06: 7500 K
         08: 9300 K
         0b: User 1
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 52 (Active control)
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B2 (Flat panel sub-pixel layout)
   Feature: B6 (Display technology type)
   Feature: C0 (Display usage time)
   Feature: C6 (Application enable key)
   Feature: C8 (Display controller type)
   Feature: C9 (Display firmware level)
   Feature: CC (OSD Language)
      Values:
         02: English
         03: French
         04: German
         05: Italian
         06: Japanese
         09: Russian
         0a: Spanish
         0d: Chinese (simplified / Kantai)
         12: Czech
         14: Dutch
         1e: Polish
   Feature: D6 (Power mode)
      Values:
         01: DPM: On,  DPMS: Off
         05: Write only value to turn off display
   Feature: DF (VCP Version)
   Feature: 60 (Input Source)
      Values:
         01: VGA-1
         03: DVI-1
         11: HDMI-1
   Feature: 62 (Audio speaker volume)
   Feature: 8D (Audio Mute)

FedeDP, 2021-02-13:
Thank you! The output is fine!
Well, if Backlight works fine, Dimmer should work too as they make the same calls to Clightd (indeed Dimmer asks Backlight to call Clightd and set 20% backlight level).
Can you test with:
`busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight SetAll "d(bdu)s" 0.2 1 0.05 30 ""`
This should smooth to 20%, and it is exactly the same call made by Dimmer.

FedeDP, 2021-02-13:
Mmh i tried with your config, here's the output:

> (D)[23:18:30]{display.c:34}     Entering dimmed state...
(D)[23:18:30]{backlight.c:156}  Received msg: 1
(D)[23:18:30]{screen.c:71}      Received msg: 1
(D)[23:18:30]{utils.c:56}       Pausing SCREEN
(D)[23:18:30]{interface.c:429}  Received msg: 1
(D)[23:18:30]{interface.c:449}  Emitting 'DisplayState' property
(D)[23:18:30]{backlight.c:156}  Received msg: 1
(D)[23:18:30]{utils.c:56}       Pausing BACKLIGHT
(D)[23:18:30]{interface.c:429}  Received msg: 1
(D)[23:18:30]{interface.c:449}  Emitting 'BlPct' property
(D)[23:18:30]{bus.c:51} Received msg: 0
(D)[23:18:30]{backlight.c:513}  Backlight 'amdgpu_bl0' level updated: 0.70.
(D)[23:18:30]{interface.c:429}  Received msg: 1
(D)[23:18:30]{interface.c:449}  Emitting 'BlPct' property
(D)[23:18:35]{bus.c:51} Received msg: 0
(D)[23:18:35]{backlight.c:513}  Backlight 'amdgpu_bl0' level updated: 0.65.
(D)[23:18:35]{interface.c:429}  Received msg: 1
(D)[23:18:35]{interface.c:449}  Emitting 'BlPct' property
(D)[23:18:40]{bus.c:51} Received msg: 0
(D)[23:18:40]{backlight.c:513}  Backlight 'amdgpu_bl0' level updated: 0.60.
(D)[23:18:40]{interface.c:429}  Received msg: 1
(D)[23:18:40]{interface.c:449}  Emitting 'BlPct' property
(D)[23:18:45]{bus.c:51} Received msg: 0
(D)[23:18:45]{backlight.c:513}  Backlight 'amdgpu_bl0' level updated: 0.55.
(D)[23:18:45]{interface.c:429}  Received msg: 1
(D)[23:18:45]{interface.c:449}  Emitting 'BlPct' property
(D)[23:18:49]{bus.c:51} Received msg: 0
(D)[23:18:49]{display.c:54}     Leaving dimmed state...

As you can see, by setting 5s for each step, it is super slow to go into dimmed state (20% backlight level).
Maybe this is why you are not seeing the transition?
Can you test with default value, ie: 30?

promeneur, 2021-02-14:
Yes if I am concentrate I see the transition with 30 ms and this is the first time.
I did not see the transition with 5000 ms.

But you forget that I opened this report because I was not seeing the 30 ms transition.

I think the problem is that the transition needs to be concentrate to see it.

Sometimes we are reading and thinking, and we don't want a black screen and sometime getting a black screen is not a  problem. I think the brightness of a step of the transition must be very different from the previous one. Transition must be an alert to the user. The user must have the time to react. 5s is a minimum to react.

We must have a compromise between saving energy and usability.

Some smartphone as Samsung one's gets this feature. If you look at the screen then no dimming. But what I suggest is easier.

promeneur, 2021-02-14:
An easy solution. Add a progressive bar showing the transition and let 5s to react before the black screen.

FedeDP, 2021-02-14:
I think you are mixing 2 concepts here:
* Dimmer -> automatic lowering of screen backlight to avoid wasting energy when unused 
* Dpms -> blanking screen when unused for much more time
By default Clight has dimming kick in after {45,20}seconds, while Dpms kicks in after {15,5}minutes.

> I think the problem is that the transition needs to be concentrate to see it.

This is normal: if you set a dimmering that lasts 5seconds for each step, it means that in best case scenario (largest step needed), it will last -> [(1.0 - 0.2) / 0.05] * 5 (ie: 0.05 step every 5 seconds going from 100% to 20% backlight), that is 90seconds; obviously you won't see be able to notice as it is far to slow.
Normally, dimming should last a couple of seconds max to give user the time to revert the process if she/he is indeed in front of screen.
Maybe, instead of 30ms, you can set eg: 100ms; that should be enough.

> An easy solution. Add a progressive bar showing the transition and let 5s to react before the black screen.

Clight has no GUI :) 

Btw i think initially there was surely a bug that got fixed somewhere during all these tests, but was "covered" by the 5000ms setting!

closed 2021-02-16 by FedeDP