FedeDP/Clight #192: [BUG]

nikp123, 2021-02-08:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.2-2

**Used distribution:**
> ArchLinux

**Describe the bug**
When calibrating clight (using -w), the perceived brightness level doesn't change significantly even though there is less light in the room.

**Expected behavior**
I expect the perceived brightness level to change if I go to a room with less light.

**To Reproduce**
1. Start clight with the ```-w``` command line flag
2. Do measurements in one room 
3. Change the light level
4. Do measurements again with different lighting

**Comments**
I believe this behavior is due to auto gain or auto white balance being on, from what I know how cameras and their drivers work.
Even though I have no experience myself with camera hardware under Linux, I found some resources that might be useful in your journey: https://www.kernel.org/doc/html/v4.19/media/kapi/v4l2-controls.html

If you find a way to disable auto gain or auto white balance in clight, it would be much appreciated. Thanks!

Please attach a Clight verbose log too!
```
nikp123@HP-Laptop /home/nikp123 $ clight --verbose -w
Config file /home/nikp123/.config/clight.conf not found.
Inhibit module is not needed. Disabling.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
|          
*---------->BR
Welcome to Clight wizard. Press ctrl-c to quit at any moment.
Waiting for sensor...
Set timeout of 0s 1ns on fd 18.
Sensor '/dev/video0' is now available.
Resumed as a sensor is now available.
Pausing BACKLIGHT
Sensor available. Start? [Y/n]: > Backlight level updated: 1.00.
y

Move to nightlight-like light.
Are you ready? [Y/n]
y
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.884812.
Ambient brightness is currently 0.885; redo capture? [y/N]: > Backlight level updated: 0.90.
Backlight level updated: 0.81.
Backlight level updated: 0.73.
Backlight level updated: 0.65.
Backlight level updated: 0.59.
Backlight level updated: 0.53.
Backlight level updated: 0.47.
Backlight level updated: 0.43.
Backlight level updated: 0.38.
Backlight level updated: 0.35.
Backlight level updated: 0.38.
y
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.884098.
Ambient brightness is currently 0.884; redo capture? [y/N]: > n
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.380

Move to a dark room.
Are you ready? [Y/n]
y
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.566695.
Ambient brightness is currently 0.567; redo capture? [y/N]: > Backlight level updated: 0.34.
Backlight level updated: 0.31.
Backlight level updated: 0.28.
Backlight level updated: 0.25.
Backlight level updated: 0.22.
Backlight level updated: 0.20.
Backlight level updated: 0.18.
Backlight level updated: 0.16.
Backlight level updated: 0.15.
Backlight level updated: 0.13.
Backlight level updated: 0.12.
Backlight level updated: 0.10.
Backlight level updated: 0.09.
Backlight level updated: 0.08.
Backlight level updated: 0.08.
Backlight level updated: 0.07.
Backlight level updated: 0.07.
Backlight level updated: 0.08.
Backlight level updated: 0.07.
Backlight level updated: 0.08.
y
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.820657.
Ambient brightness is currently 0.821; redo capture? [y/N]: > y
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.818186.
Ambient brightness is currently 0.818; redo capture? [y/N]: > y
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.894072.
Ambient brightness is currently 0.894; redo capture? [y/N]: > y
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.807863.
Ambient brightness is currently 0.808; redo capture? [y/N]: > 
```

FedeDP, 2021-02-08:
Hi! Thanks for opening this bug request!
As @NICHOLAS85 suggested, Clight does set camera default values (https://github.com/FedeDP/Clightd/blob/master/src/modules/sensors/camera.c#L209) initially, and user can override them.
I did not force any option because not all cameras support them and it was quite a mess to handle everything correctly.

Look here for more informations: https://github.com/FedeDP/Clightd/wiki/Sensors#camera

PPS: i will give wizard mode some love before releasing next version because it is not quiet enough right now :D

nikp123, 2021-02-08:
Sorry I wasn't aware that it was already a part of the codebase. Maybe add that in the default config file so it's easier to find?

FedeDP, 2021-02-08:
Uh should already be there under sensor settings, see https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L108 :)

FedeDP, 2021-02-08:
Ahhh i see, i may add a proper example, you are right!

FedeDP, 2021-02-10:
Sensor settings string does now expose a proper example: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L109

closed 2021-02-10 by FedeDP