FedeDP/Clight #193: Give Clight a hand!

FedeDP, 2021-02-08:
Hi everyone!
As Clight is quickly growing in usage and community, and there are quite some people here that hangout between issues, i was wondering whether anyone would like to step up and help me provide a smoother experience with a better documentation.

Unfortunately i have a limited time and i'd like to only focus on development;
I think the main Clight weak point right now is its wiki pages / documentation.
Despite my efforts (and believe me, it takes quite some time to write wiki pages, maintain them up to date with any new features etc etc), documentation falls behind and often issues are opened just to ask questions that normally should have been answered in the doc.

Eg: i think Custom Modules and DBus API are very cool features, but i guess nobody or so uses them. I am not good at advertising Clight features!

I think that a FAQ wiki page could help, but i am opened to a much wider contribution on any wiki page (or any other ideas is super accepted).

Is anybody willing to help? 
Thank you very much to everyone involved :)

promeneur, 2021-02-09:
I can help for what I know about Clight.

- openSuse installation
- some console commands

FedeDP, 2021-02-10:
@promeneur thank you very much! I can give you access as soon as you wish!

nullobsi, 2021-02-16:
this sounds like something I may be able to help with! i think writing a more simplified overview of the DBus api would be nice, and a FAQ page with questions like "how do I change brightness," "pausing clight," etc would be a nice addition :smile:

FedeDP, 2021-02-16:
Yep, i think the FAQ page can be a great and useful addition! I will add both as contributor!
Thank you very much to @promeneur  and @nullobsi !!

promeneur, 2021-02-27:
Hello 

I don't know anything about editing a wiki and this wiki in particular.
I don't want to take time to learn a new tool.
Furthermore, I can't learn by modifying the clight wiki. I could destroy it.

If you want I can send to you text files which you will include them in the wiki.

promeneur, 2021-02-27:
With LibreOffice, I can produce a file with wiki format.

FedeDP, 2021-02-27:
@promeneur there is no problem, really. Feel free to contribute the way you like; btw honestly github wiki pages are quite simple to use; it is like writing in your text editor! I see that @nullobsi already created a FAQ page (thank you!): https://github.com/FedeDP/Clight/wiki/FAQ

nullobsi, 2021-02-27:
github wiki uses markdown, which is used in comments and almost anywhere you see rich text!

FedeDP, 2021-02-27:
> github wiki uses markdown, which is used in comments and almost anywhere you see rich text!

Yep exactly, and it is pretty simple actually, at least for basic usage!
@nullobsi mind to add a small index at the top of the FAQ page with direct links to each "question -> answer"? 
I think it is better for quick glances :)

FedeDP, 2021-02-28:
I updated the faq page; i think it is really usefult! And can only grow up from now on :)

promeneur, 2021-02-28:
Here is a howto about opensuse installation of Clight. Feel free to tune it.

[opensuse_Installation.txt](https://github.com/FedeDP/Clight/files/6056607/opensuse_Installation.txt)

FedeDP, 2021-02-28:
> Here is a howto about opensuse installation of Clight. Feel free to tune it.

https://github.com/FedeDP/Clight/wiki/Build#installation---opensuse

closed 2021-08-21 by FedeDP