FedeDP/Clight #195: [BUG]  Unable to change brightness via dbus API

mnussbaum, 2021-02-11:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**

Latest version from clight-git and clightd-git AUR packages, installed as of 2/11/21

**Used distribution:**

Arch / clight-git

**Describe the bug**

The `IncBl` and `DecBl` dbus commands seem to have no impact. 

**Expected behavior**

I expected the dbus commands to dim or brighten the screen.

**To Reproduce**

Running the following many times doesn't dim the screen at all:

```
busctl --user call org.clight.clight /org/clight/clight org.clight.clight DecBl "d" 0.9
```

I've confirmed the latest clightd is running and isn't logging any errors, and I'm manually running `clight --verbose` and not seeing any errors logged. Here's the `clight --verbose` output:

```
$ clight --verbose
Config file /home/mnussbaum/.config/clight.conf not found.
Clightd found, version: 5.2.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **
|     *
|    *
|   *
|  *
|
| *
|*
|
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|
|
|       ***
|     **
|    *
|   *
|  *
|
|**
|
*---------->BR
Keyboard backlight calibration supported.
Disarmed timerfd on fd 38.
AC cable disconnected.
Laptop lid opened.
Emitting 'AcState' property
Disarmed timerfd on fd 41.
Emitting 'LidState' property
Error loading from cache file.
```

I'm happy to help debug this any way I can! Thanks!

FedeDP, 2021-02-11:
Hi! Thanks for taking the time to open this bug report!
It seems Clight is doing nothing (from your log file).
Can you share same log file after eg: 4-5 mins of Clight running?

FedeDP, 2021-02-11:
GIven your log, i'd say that DAYTIME module is not starting (ie: geoclue cannot provide a location for you, maybe?)

mnussbaum, 2021-02-11:
Ah thanks for the pointer to the geoclue issue! I don't have geoclue running, but mistakenly thought I had `latitude` and `longitude` set in the config instead.

I made the mistake of modifying `/etc/default/clight.conf` instead of creating a per-user config in my home dir. When I recently uninstalled and reinstalled the package it overwote my lat/long settings in the default config file and clight was indeed blocking looking for geoclue.

Thanks so much for your help and sorry to bother you with a user error!

closed 2021-02-11 by mnussbaum

FedeDP, 2021-02-11:
Ahaha classic :smile:
Nice to heat you solved the issue!