FedeDP/Clight #196: ambient gamma adjusting keeps oscillating up and down

promeneur, 2021-02-12:
clight, clightd master branch 12-2-2021

Instead of adjusting gamma to ambient brightness clight keeps oscillating up and down. I must switch to no mabient gamma.

[clight.log](https://github.com/FedeDP/Clight/files/5972621/clight.log)

FedeDP, 2021-02-12:
Hi! Thanks for opening this issue.
I cannot reproduce it here! Can you share a verbose log that starts with ambient_gamma enabled?
That one is not verbose and it has
> * Ambient gamma:		Disabled

as starting conf (i assume you changed it at runtime with clight-gui).

promeneur, 2021-02-12:
>> it has ambient gamma disable

No, the log is the record of:

start with no ambient gamma
switch to ambient gamma
you see 2 groups of oscillations
switch to no ambient gamma

NICHOLAS85, 2021-02-12:
Do you have multiple monitors?

NICHOLAS85, 2021-02-12:
Additionally, according to your log `Ambient Gamma` is disabled. Make sure it's enabled in your config and that clight-gui isn't turning it off(if you use the gui). If it's not your config or clight-gui then there might be another bug present

promeneur, 2021-02-12:
One monitor

promeneur, 2021-02-12:
i use clight-gui to switch on/off abient gamma

FedeDP, 2021-02-12:
I need a verbose log :) thank you!

NICHOLAS85, 2021-02-12:
What DE do you use?

promeneur, 2021-02-13:
I use KDE.

promeneur, 2021-02-13:
[clight.log](https://github.com/FedeDP/Clight/files/5975675/clight.log)

[clight.conf.txt](https://github.com/FedeDP/Clight/files/5975677/clight.conf.txt)

The ambient brightness is almost constant.

The temperature keeps oscillating between 4100 K and about 6000 K

FedeDP, 2021-02-13:
From your log, it seems it is working fine. 
You are seeing the issue because of #190: dimmer is not actually dimming your screen, thus you see a normal backlight with a 4100 temperature.
Instead, you should see a backlight level of 20% with 4100 gamma temp.
We can safely close this one, and focus on fixing other issue!

closed 2021-02-13 by FedeDP