FedeDP/Clight #197: [BUG] clight reports monitor backlight change too often

nullobsi, 2021-02-16:
**Clight version the issue has been seen with:**
> current master

**Used distribution:**
> arch linux

**Describe the bug**
after checking htop only to find waybar using up an entire cpu core, i have traced the issue down to clight. it keeps reporting the brightness of my external monitor over the dbus BlPct property... i have an internal/laptop monitor and an external monitor. the logs are spammed with

```
Backlight 'LF80D0028500' level updated: 1.00.
Received msg: 1
Emitting 'BlPct' property
Received msg: 0
Backlight 'LF80D0028500' level updated: 1.00.
Received msg: 1
Emitting 'BlPct' property
Received msg: 0
```
and so forth. LF80 is my external monitor. this doesn't happen when I just have my internal monitor connected-- it is connected via HDMI usb-c adapter. since the brightness control doesn't entirely work over the hdmi (it used to- but only to dim), i wonder if there is a way to blacklist monitors entirely?

**Expected behavior**
clight should only emit Changed on the BlPct property when the brightness actually changes.

**To Reproduce**
unfortunately, i'm not sure it is easily reproducible- but i'd be happy to do any debugging!

FedeDP, 2021-02-16:
Hi! It seems like you got hit by #106 ! Are these log lines continuously appearing?
Can you share your clight log?
If this actually is #106 bug, i can use the log to understand and hopefully fix it!
It seems like the bug is that clight continuously receives signals from Clightd. That is weird because Clightd is not using 100% of cpu, so it seems like a client-side bug.

FedeDP, 2021-02-16:
>  If this actually is #106 bug, i can use the log to understand and hopefully fix it!

I recently added the "Received msg: " debug line to help me catch this bug ;)

nullobsi, 2021-02-16:
hmm, i don't think it's that because it isn't clight that's using up CPU-its just sending a bunch of dbus messages
it went like this:
clightd -> clight -> clight-gui -> waybar AppIndicator

and waybar was using an entire core processing the spamny request to update the tray icon

nullobsi, 2021-02-16:
And yes, these log lines are repeating; I can send you a full log in a bit

FedeDP, 2021-02-16:
> And yes, these log lines are repeating; I can send you a full log in a bit

Thank you!
> and waybar was using an entire core processing the spamny request to update the tray icon

I see but Clight should not send continuously the update! It should only send it when a capture is taken and backligth is calibrated; 
it will send a single update for "smooth" targets (ie: target backlight and smooth params) + another update for each step until target backlight is reached, with new backlight.
In case smoothing is disabled, it will just send a single update with set  value.

nullobsi, 2021-02-16:
https://pastebin.com/UePBTJKK

This is my clight log; it's not capturing, it's just updating BlPct property.... here's my config too:
```
verbose = false;
resumedelay = 0;
backlight : 
{
  no_smooth_transition = false;
  trans_step = 0.05;
  trans_timeout = 30;
  no_auto_calibration = false;
  pause_on_lid_closed = true;
  capture_on_lid_opened = false;
  screen_sysname = "intel_backlight";
  shutter_threshold = 0.0;
  ac_timeouts = [ 10, 20, 10 ];
  batt_timeouts = [ 1200, 5400, 600 ];
};
sensor : 
{
  captures = [ 5, 5 ];
  devname = "";
  settings = "";
  ac_regression_points = [ 0.0, 0.15, 0.29, 0.45, 0.61, 0.74, 0.81, 0.88, 0.93, 0.97, 1.0 ];
  batt_regression_points = [ 0.0, 0.15, 0.23, 0.36, 0.52, 0.59, 0.65, 0.71, 0.75, 0.78, 0.8 ];
};
keyboard : 
{
  disabled = false;
  dim = false;
  ambient_br_thresh = 1.0;
};
gamma : 
{
  no_smooth_transition = true;
  trans_step = 50;
  trans_timeout = 300;
  long_transition = false;
  ambient_gamma = false;
  temp = [ 6500, 4000 ];
};
daytime : 
{
  event_duration = 1800;
  sunrise = "";
  sunset = "";
};
dimmer : 
{
  no_smooth_transition = [ false, false ];
  trans_steps = [ 0.05, 0.05 ];
  trans_timeouts = [ 30, 30 ];
  dimmed_pct = 0.2;
  timeouts = [ 45, 30 ];
};
dpms : 
{
  timeouts = [ 300, 300 ];
};
screen : 
{
  num_samples = 10;
  contrib = 0.1;
  timeouts = [ -1, -1 ];
};
inhibit : 
{
  inhibit_docked = false;
  inhibit_pm = true;
};
```
and, just now it started freezing when i open it:
```
Clightd found, version: 5.2.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **
|     *
|    *
|   *
|  *
|
| *
|*
|
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|
|
|       ***
|     **
|    *
|   *
|  *
|
|**
|
*---------->BR
init_kbd_backlight(): No such file or directory
Keyboard backlight calibration unsupported.
Disarmed timerfd on fd 38.
Received msg: 1
Received msg: 0
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
AC cable connected.
Received msg: 1
Laptop lid opened.
Received msg: 1
Emitting 'AcState' property
Disarmed timerfd on fd 41.
Received msg: 1
Emitting 'LidState' property
Received msg: 1
32.67 -116.98 loaded from cache file.

^[[A^[[B^C^C^C^C^C^C^CLoc distance: 32.67,-116.98 -> 91.00,181.00 : 3993.51 km.
New location received: 32.67, -116.98.
Received 2. Leaving.
Latest location stored in cache file!
Set timeout of 0s 1ns on fd 42.
Received msg: 1
Received msg: 1
Received msg: 1
Received msg: 1
Emitting 'Location' property
Received msg: 1
Received msg: 1
```

FedeDP, 2021-02-16:
Can you share your log file? So that i have timestamps :) Thanks!

nullobsi, 2021-02-17:
https://pastebin.com/3SRMpJBf

Alright, I've got the log.

FedeDP, 2021-02-17:
Thank you very much!
The issue is clear: your external monitor, as you said, does not properly support ddcci; so Clightd "smooths" its backlight until it reaches Clight requested one; but on your external monitor it never reaches that backlight level thus continuously looping (if you see, you have a 30ms smooth timeout for backlight and you have 33 logs for each second ;) ) and sending dbus signals.

Can you share `sudo ddcutil capabilities` output?

nullobsi, 2021-02-17:
```
Model: ACER
MCCS version: 2.0
Commands:
   Op Code: 01 (VCP Request)
   Op Code: 02 (VCP Response)
   Op Code: 03 (VCP Set)
   Op Code: 07 (Timing Request)
   Op Code: 0C (Save Settings)
   Op Code: E3 (Capabilities Reply)
   Op Code: F3 (Capabilities Request)
VCP Features:
   Feature: 02 (New control value)
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 08 (Restore color defaults)
   Feature: 0B (Color temperature increment)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values:
         05: 6500 K
         08: 9300 K
         0b: User 1
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 52 (Active control)
   Feature: 60 (Input Source)
      Values:
         01: VGA-1
         03: DVI-1
         04: DVI-2
   Feature: 6C (Video black level: Red)
   Feature: 6E (Video black level: Green)
   Feature: 70 (Video black level: Blue)
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B2 (Flat panel sub-pixel layout)
   Feature: B6 (Display technology type)
   Feature: C6 (Application enable key)
   Feature: C8 (Display controller type)
   Feature: C9 (Display firmware level)
   Feature: CC (OSD Language)
      Values:
         01: Chinese (traditional, Hantai)
         02: English
         03: French
         04: German
         05: Italian
         06: Japanese
         09: Russian
         0a: Spanish
         0d: Chinese (simplified / Kantai)
         14: Dutch
         16: Finnish
   Feature: D6 (Power mode)
      Values:
         01: DPM: On,  DPMS: Off
         05: Write only value to turn off display
   Feature: DF (VCP Version)
```
This is the output.

FedeDP, 2021-02-17:
Thank you!
Hopefully last request: can you paste here `journalctl -u clightd.service  -b -f` output (i just need to know if there is any "failed to set backlight for ..." log!)
You must be on clightd latest -git version (4d ago). Thank you very much!
If the log line is present, then at least ddcutil is telling that it cannot set backlight on the monitor. Fingercrossed!

nullobsi, 2021-02-17:
I get a lot of spam of "display not found" (probably every 30ms like you observed)

And yes, I do get "Failed to set backlight"

```
 2月 17 09:34:46 NULL-HUB clightd[1446]:  to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
 2月 17 09:34:46 NULL-HUB clightd[1446]: [default]|BACKLIGHT|: failed to set backlight for LF80D0028500
```
It's a little hard to copy, but there's also 
```
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:44 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
 2月 17 09:34:45 NULL-HUB clightd[1446]: Display not found
```

FedeDP, 2021-02-17:
So, if you try a `sudo ddcutil setvcp 10 80`, your external monitor won't change backlight, right?

nullobsi, 2021-02-17:
It actually does, which is weird.

FedeDP, 2021-02-18:
Ok! I cannot understand why it is happening, but it seems like ddcutil fails to open display (and it prints "Display not found"!) then Clightd obviously fails to set backlight (it does not even try obviously).
I am investigating more, may be i do something wrong!

FedeDP, 2021-02-19:
In 5f1c3dae0428691b4b85fc488d03df2857b0d41f i improved external monitor handling code. 
Can you test if it now works for you?

NICHOLAS85, 2021-02-19:
Just ran into this issue without having an external monitor connected.
[clight.log](https://github.com/FedeDP/Clight/files/6011640/clight2.txt)

```
# ddcutil capabilities
Display not found

#journalctl -u clightd.service -b -f
-- Journal begins at Sun 2020-12-20 14:55:13 EST. --
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|IDLE|: Starting Client 1
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|IDLE|: Starting Client 0
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|BACKLIGHT|: Target pct: 0.80
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|BACKLIGHT|: intel_backlight reached target backlight: 0.80.
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|IDLE|: Client 0 -> Idle: 1
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|BACKLIGHT|: Target pct: 0.06
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|BACKLIGHT|: intel_backlight reached target backlight: 0.06.
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|IDLE|: Leaving idle state.
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|BACKLIGHT|: Target pct: 0.80
Feb 18 15:23:23 Nicks-Laptop clightd[9666]: [default]|IDLE|: Stopping Client 1
```

nullobsi, 2021-02-19:
hmm, it appears the latest master has fixed the issue!

nullobsi, 2021-02-19:
i'll keep using it for a bit, and i'll close this issue if it doesn't pop up again :+1: thank you for your hard work!

FedeDP, 2021-02-19:
> hmm, it appears the latest master of clightd has fixed the issue!

Does monitor backlight work now?

@NICHOLAS85 you got hit by #106 !! And that's a great news! And the error is different from @nullobsi  one as you can see from your output; ok now i must understand why INTERFACE module goes mad :)

FedeDP, 2021-02-19:
@NICHOLAS85 
It seems like sd_bus_process() continuously loops for "Inhibit" dbus monitor. 
In Clight master i added a new debug print; it will be really helpful if you could upgrade to it!

FedeDP, 2021-02-24:
@nullobsi  did backlight change now?
btw i just committed a couple of new fixes on Clightd for this!

nullobsi, 2021-02-26:
yup, due to hotplugging sometimes the monitor does not show up
but it appears to be working fine when it does! and there is no more wild cpu usage

thank you :)

closed 2021-02-26 by nullobsi