FedeDP/Clight #198: [FEATURE REQ] Apply gamma changes to different VTs

Barbarossa93, 2021-02-16:
**Is your feature request related to a problem? Please describe.**
I've recently been using light-locker as a screen locking tool and the way it works is that it switches VTs when its locking the screen. If gamma is active on the original VT (the one I logged into, which I think is vt7 with lightdm), it isn't applied to the locking VT (which I believe is VT8 by default with light-locker). I've found it jarring when I'm locking/unlocking the screen with the differences in gamma

**Describe the solution you'd like**
Ideally, clight would be able to apply gamma to other VTs besides the logged in one

Frankly, I'm not sure if this is possible, but I wanted to ask anyway. I've been surprised before with the bumblebee fix :)

FedeDP, 2021-02-16:
Hi! Thanks for opening this feature request!
Well, i guess you should discover on which X screen does light-locker start, and using same bumblebee fix on that new X screen, i guess :)
Ie: try to update synctemp_bumblebee module doubling the system() calls, eg:
```
snprintf(cmd, sizeof(cmd), "busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set \"ssi(buu)\" \"%s\" \"%s\" %d %d %d %d", 
                BUMBLEBEE_X_DISPLAY, xauth, up->new, up->smooth, up->step, up->timeout);
system(cmd);

snprintf(cmd, sizeof(cmd), "busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set \"ssi(buu)\" \"%s\" \"%s\" %d %d %d %d", 
                LIGHTLOCKER_X_DISPLAY, xauth, up->new, up->smooth, up->step, up->timeout);
system(cmd);
```

Barbarossa93, 2021-02-16:
Well I'll be darned, apparently light-locker is on an X display! Looks like its on :1 by default. I think the thing I'm unsure of is that the :1 display doesn't start until the lock initiates. Will the block you mentioned apply for X displays that have yet to be started?

FedeDP, 2021-02-16:
You are right! Does light-locker always restart itself? Ie: i think it runs in background with the ":1" display always open... you can give this a try with a quick:
`busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set "ssi(buu)" ":1" $XAUTHORITY 4000 0 0 0`
(please replace $XAUTHORITY with your user env $XAUTHORITY).
Check if it does not return any error, it means it is working (well you can try to lock screen and see!); if it instead returns any error it means light-locker spawns the X display every time, and closes it.

Barbarossa93, 2021-02-16:
Unfortunately I get an error message :( I don't see any options to allow light-locker to run in the background. Since it's part of lightdm however, I know I can execute commands after the X display is started or before the greeter is launched by setting the command in `/etc/lightdm/lightdm.conf`. Is there a simple command to ping clight to check and trigger gamma changes as needed?

Barbarossa93, 2021-02-16:
Indeed, placing that test command you gave me into a script and putting it at `display-setup-script` in `lightdm.conf` worked! All that I need is a command that checks whether it is appropriate to enable/disable gamma and then triggers the appropriate response

Barbarossa93, 2021-02-16:
Unfortunately, my script didn't seem to work:
```
#!/usr/bin/env bash

isItNightOrDay=$(busctl --user get-property org.clight.clight /org/clight/clight org.clight.clight DayTime | awk '{print $2}')

if [[ "$isItNightOrDay" == "1" ]]; then
    busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set "ssi(buu)" ":1" $(echo "$XAUTHORITY") 4000 0 0 0 || true
fi

```

I'm not sure how to query whether its day or night from outside a user

FedeDP, 2021-02-17:
Yep, the most simple way is just to get gamma temp from Clightd for the other X display and apply it (pseudocode):
```
temp = busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Get "ss" ":0" $XAUTHORITY
busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set "ssi(buu)" ":1" $XAUTHORITY temp 0 0 0
```

Barbarossa93, 2021-02-17:
Perfect, thanks :)

closed 2021-02-17 by Barbarossa93