FedeDP/Clight #199: gamma setting not immediately executed

promeneur, 2021-02-16:
opensuse 15.2
clight , clightd master branch

ambient gamma

i have this script
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#Set gamma to max gamma
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 6500

/usr/bin/vlc

#Restore gamma to min gamma 
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 3500
***************************************************************************************************************************

i must wait for 10 mn before temperature is set to 6500 K or 3500 K

FedeDP, 2021-02-16:
Hi! Thank you for discovering this bug! Yup, it is fixed with new clight git master!
Thanks!

promeneur, 2021-02-17:
success

closed 2021-02-17 by promeneur