FedeDP/Clight #200: failed to set gamma temperature

promeneur, 2021-02-21:
openSuse 15.2
clight 20/02/2021, clightd  13/02/2021

A new phenomenon.

This afternoon adjusting gamma failed to set gamma temperature.

[clight.log](https://github.com/FedeDP/Clight/files/6018796/clight.log)

promeneur, 2021-02-22:
I was  playing with an adroid-x86 virtuabox machine.

I was playing with android app using webcam.

Sharing a webcam between host and guest is a static sharing [*]. When you start an android-x86 VM then the VM takes the webcam and releases the webcam when shutting down the VM.

[*] we can't install guest additions then there is only static sharing. In the contrary with a dynamical sharing the webcam is captured only if an app uses it.

FedeDP, 2021-02-22:
It seems like Clightd was failing to set gamma temp. Can you share Clightd log from journalctl?
(journalctl -u clightd -b)

promeneur, 2021-02-23:
It's a random phenomenon. I must wait for a new occurrence.

promeneur, 2021-02-24:
I get it !

[clightd.log](https://github.com/FedeDP/Clight/files/6037936/clightd.log)

Also, I did not get the screen switched off.

FedeDP, 2021-02-24:
Great!
It seems like your X DISPLAY is getting some issues:
```
Invalid MIT-MAGIC-COOKIE-1 key
No protocol specified
Error: Couldn't open :0.
```
I found eg: https://unix.stackexchange.com/questions/199891/invalid-mit-magic-cookie-1-key-when-trying-to-run-program-remotely 

It is surely not related to Clight (or Clightd), but is something that changes your environment!
Do you by chance change the environment in which Clightd runs? Eg: unmount your home partition maybe? (that contains your Xauthority file).
See also: https://bbs.archlinux.org/viewtopic.php?id=65991

promeneur, 2021-02-24:
No I don't play with that kind of settings. It's too techie for me.

But I know that SUSE changed the way to manage this setting to enforce security for about 3 months.

See my bug report to suse about some app getting some trouble about using display since the change.

The important thing is:

x@grognon:~> env | grep -e DISPLAY -e AUTHORITY
DISPLAY=:0
XAUTHORITY=/run/user/1000/xauth_anrAKo

It seems that at every pc start or kde session start the filename xauth_<suffix> gets a random <suffix>

FedeDP, 2021-02-24:
That is not an issue because Clight uses XAUTHORITY env variable!
Maybe something is messing with your Xauthority file?

> But I know that SUSE changed the way to manage this setting to enforce security for about 3 months.

Mind to share a link?

promeneur, 2021-02-24:
No I assume that because three months ago I did nor get any trouble with powertop and luckybackup using display:0

And the answer of "env | grep -e DISPLAY -e AUTHORITY" was "XAUTHORITY=~/.Xauthority"

i fix my previuos comment : It seems that at every pc start or kde session start the filename xauth_suffix gets a random suffix.

FedeDP, 2021-02-24:
> No I assume that because three months ago I did nor get any trouble with powertop and luckybackup using display:0

Ah, i see.
What login manager do you use? Sddm? 
Anyway, that is surely not causing the issue; maybe at some point the file becomes non-readable? Or it gets corrupted?

promeneur, 2021-02-24:
yes with kde this is sddm.

FedeDP, 2021-02-25:
I don't have any clue then (i thought that maybe some login manager was doing weird things with your X env, but sddm works fine).
This seems pretty interesting too: https://unix.stackexchange.com/questions/219684/invalid-mit-magic-cookie-1-error-whenever-computer-falls-asleep

FedeDP, 2021-03-11:
I cannot give you much more support, sorry. It is unrelated to Clight too!
If you can find a fix, feel free to mention me here: i am quite curious!
Thank you very much!

closed 2021-03-11 by FedeDP