FedeDP/Clight #202: [BUG] sigsegv on dbus introspection

nullobsi, 2021-02-26:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> as of this writing, latest master.

**Used distribution:**
> Arch Linux

**Describe the bug**
it seems using the autocomplete of fish shell on busctl causes clight to coredump with a SIGSEGV.

**Expected behavior**
clight should not segfault

**To Reproduce**
Fish shell has wonderful autocomplete support for busctl, making it really easy to call methods. However, it seems using the autocomplete on /org/clight/clight/Conf/Backlight causes clight to sigsegv when trying to introspect the interfaces. Typing `busctl --user call org.clight.clight /org/clight/clight/Conf/Backlight` and then pressing tab to autofill the interface causes clight to crash. The backtrace comes out to:
```
#0  0x00007faa9571f537 in __strlen_avx2 () from /usr/lib/libc.so.6
#1  0x00007faa957ae2f2 in ?? () from /usr/lib/libsystemd.so.0
#2  0x00007faa957ba3ab in ?? () from /usr/lib/libsystemd.so.0
#3  0x00007faa957ba56e in ?? () from /usr/lib/libsystemd.so.0
#4  0x00007faa957bc559 in ?? () from /usr/lib/libsystemd.so.0
#5  0x00007faa957d2a64 in ?? () from /usr/lib/libsystemd.so.0
#6  0x0000562fac6996bb in ?? ()
#7  0x00007faa95cb84f4 in ?? () from /usr/lib/libmodule.so.5
#8  0x00007faa95cb7bd3 in ?? () from /usr/lib/libmodule.so.5
#9  0x00007faa95cb7e85 in modules_ctx_loop_events () from /usr/lib/libmodule.so.5
#10 0x0000562fac6941ff in main ()
```
which seems very strange. The exact command fish runs is: `busctl --user introspect --list org.clight.clight /org/clight/clight/Conf/Backlight`

Clight verbose log:
```
Clight
* Software version:		4.4
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Fri Feb 26 14:10:26 2021

Starting options:

### GENERIC ###
* Verbose (debug):		Disabled
* ResumeDelay:		0

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth steps:		0.05
* Smooth timeout:		30
* Daily timeouts:		AC 10	BATT 1200
* Nightly timeout:		AC 20	BATT 5400
* Event timeouts:		AC 10	BATT 600
* Backlight path:		intel_backlight
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Enabled
* Capture on lid opened:		Disabled

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		Unset
* Settings:		Unset

### KEYBOARD ###
* Dim:		Disabled
* Threshold:		1.00
* Timeouts:		AC 15	BATT 5

### GAMMA ###
* Smooth trans:		Disabled
* Smooth steps:		50
* Smooth timeout:		300
* Daily screen temp:		6500
* Nightly screen temp:		4000
* Long transition:		Disabled
* Ambient gamma:		Disabled

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800

### DIMMER ###
* Smooth trans:		ENTER: Enabled, EXIT: Enabled
* Smooth steps:		ENTER: 0.05, EXIT: 0.05
* Smooth timeout:		ENTER: 30, EXIT: 30
* Timeouts:		AC 45	BATT 30
* Backlight pct:		0.20

### DPMS ###
* Timeouts:		AC 300	BATT 300

### SCREEN ###
* Timeouts:		AC -1	BATT -1
* Contrib:		0.10
* Samples:		10

### INHIBIT ###
* Docked:		Disabled
* PowerManagement:		Enabled

(I)[14:10:26]{main.c:127}	Clightd found, version: 5.3.
(I)[14:10:26]{backlight.c:415}	AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
(I)[14:10:26]{backlight.c:415}	BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
|          
*---------->BR
(I)[14:10:26]{keyboard.c:129}	Keyboard backlight calibration unsupported.
(I)[14:10:26]{upower.c:54}	AC cable connected.
(I)[14:10:26]{upower.c:65}	Laptop lid opened.
(I)[14:10:26]{location.c:62}	New location received: ####
(I)[14:10:26]{daytime.c:175}	Next alarm due to: Fri Feb 26 17:12:00 2021
(I)[14:10:26]{gamma.c:181}	6500 gamma temp set.
(I)[14:10:27]{backlight.c:328}	Ambient brightness: 0.143 -> Backlight pct: 0.231.
(W)[14:10:31]{main.c:109}	Received sigsegv signal. Aborting.
```

nullobsi, 2021-02-26:
doing some debugging, the error happens on sd_bus_process under modules/bus.c
[image]

FedeDP, 2021-02-26:
Hi! 
That is super weird! Does it happen with bash or zsh too?
Segfaulting on sd_bus_process given correct bus is being used (ie: pointer is not corrupted) is up to systemd; but i don't think an autocompletion script should matter...
I can tell you that with bash it works fine!
Did you restart clight after upgrading (or better still restart your session/pc)?

nullobsi, 2021-02-26:
It is very easily reproducible, even on reboot. Bash causes it to crash as well.

nullobsi, 2021-02-26:
using dbus-monitor it appears it is the call to org.freedesktop.DBus.Properties GetAll on the /org/clight/clight/Conf/Backlight object that causes the crash

FedeDP, 2021-02-26:
Other object paths don't have the problem?

nullobsi, 2021-02-26:
i will test out other object paths later, but I can say that passing other interface names to GetAll (like "org.clight.clight") do not cause a crash, just "org.clight.clight.Conf.Backlight". Perhaps it is the same for others

FedeDP, 2021-02-26:
Can you try to comment out L115 of interface.c and retry?
`SD_BUS_WRITABLE_PROPERTY("BacklightSyspath", "s", NULL, NULL, offsetof(bl_conf_t, screen_path), 0),`
Just a test!

FedeDP, 2021-02-26:
Btw it works fine on my other laptop too here!

nullobsi, 2021-02-28:
commenting that line fixes the issue

FedeDP, 2021-02-28:
Great! I've had issues with string properties in the past.
Out of curiosity, systemd version?
It seems like a bug in it (here works fine on both v247 and v245).

FedeDP, 2021-02-28:
Hi! New master should spot a fix for the issue; i cannot test it as i cannot reproduce it; can you test it?
Moreover, with your vesion, i guess that `/org/clight/clight/Conf/Sensor` should cause a crash too; i fixed that too (can you test that indeed it crash for you, before updating?)

Thank you very much, if the fix is fine i will release a bugfix 4.4 version :)

nullobsi, 2021-03-02:
it appears your fix worked! testing with the older version, it appears there is no crash on the Sensor path, though... not sure what's up with that

closed 2021-03-02 by nullobsi