FedeDP/Clight #203: [BUG] when there is an audio streaming clicght switches off the screen

promeneur, 2021-03-04:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> clight 4.3 , clightd 5.2

**Used distribution:**
> openSuse 15.2

**Describe the bug**
clight does not takes care of audio streaming. After a delay screen is switched off thus also the inbuilt speakers !

**Expected behavior**

clight does not switch off the screen when there is an audio streaming.

nullobsi, 2021-03-04:
Hi,

I think this feels a little out of scope from Clight. It is hard to check such a setting- if you're using Pulse, should we inhibit when anything is playing? How should we tell when an output is internal speakers? etc.

You could use clight-gui or scripts manually to inhibit DPMS when you want, or with a more complicated solution use MPRIS to check if there is a "media player" running and inhibit. That is the point of the DBus interface, for automation. :) You can try disabling DPMS, and instead lowering your brightness to a really low number to keep the display on.

promeneur, 2021-03-06:
This is the same problem with the same feature of KDE in energy saving settings.
I sent a bug report to KDE.

FedeDP, 2021-03-06:
Hi! Indeed it seems like your music player does not send org.freedesktop.Screensaver.Inhibit request.
That's why both clight and kde are having the same issue! 
Out of curiosity, which music player are you using?

promeneur, 2021-03-06:
I use Firefox.

https://www.arteradio.com/emission/fenetre_sur_cour

click on "écouter"

promeneur, 2021-03-06:
I experimented the same problem with qtwebkit in akregator.

FedeDP, 2021-03-06:
It works fine here:

```
New ScreenSaver inhibition held by 'firefox': 'audio-playing'. Cookie: 1780419494.
ScreenSaver inhibition grabbed.
Inhibitions ctr: 1
Pausing DPMS
Emitting 'Inhibited' property
Pausing DIMMER
```

Can you share a verbose log?
Firefox version?

FedeDP, 2021-03-06:
> clight does not takes care of audio streaming with for example the inbuilt explorer webkit in Akregator. After a delay monitor is switched off thus also the inbuilt speakers !

I am guessing that either KDE or some dpms setting (eg: `xset` called by some script) is overpassing Clight (ie: it is not Clight switching off your screen, but something else).

promeneur, 2021-03-06:
>> I am guessing that either KDE or some dpms setting (eg: xset called by some script) is overpassing Clight (ie: it is not Clight 
>> switching off your screen, but something else).

I disabled kde "switch off screen".

>> Firefox version?

86.0

FedeDP, 2021-03-06:
Care to share a verbose log?

promeneur, 2021-03-06:
i have just experimented. This time no problem. The monitor is not switched off after 5 mn ! It's a random phenomenon.

[clight.log](https://github.com/FedeDP/Clight/files/6095215/clight.log)

FedeDP, 2021-03-06:
It seems like you are using 
> (D)[12:32:02]{interface.c:629}	New ScreenSaver inhibition held by '/usr/bin/google-chrome-stable': 'Video Wake Lock'. Cookie: 603570492.

google chrome, not firefox ?

Maybe sometimes the browser does not send the Inhibit request?

promeneur, 2021-03-06:
Strangely in log there is no occurrence of Firefox while listening an audio streaming.

The occurrences of Chrome are due to using permanently website as Instagram, Facebook, Twitter etc. with Chrome. Randomly the current item may content a video.

promeneur, 2021-03-06:
I think that Firefox gets a problem.

FedeDP, 2021-03-06:
> Strangely in log there is no occurrence of Firefox while listening an audio streaming.

Weird, i got firefox 86 and it works fine! @nullobsi care to test too?
May be some misconfiguration, i don't really know.
Does firefox inhibit screensaver when watching a video?

promeneur, 2021-03-06:
Firefox+video viewing no problem.

(D)[12:54:50]{screen.c:134}	Average screen-emitted brightness: 0.037216.
(D)[12:54:50]{timer.c:32}	Set timeout of 30s 0ns on fd 43.
(D)[12:54:50]{interface.c:455}	Emitting 'ScreenComp' property
(D)[12:55:02]{interface.c:629}	New ScreenSaver inhibition held by 'firefox': 'video-playing'. Cookie: 168002245.
(D)[12:55:02]{inhibit.c:61}	ScreenSaver inhibition grabbed.
(D)[12:55:02]{inhibit.c:63}	Inhibitions ctr: 1
(D)[12:55:02]{utils.c:56}	Pausing DPMS
(D)[12:55:02]{interface.c:455}	Emitting 'Inhibited' property
(D)[12:55:02]{utils.c:56}	Pausing DIMMER
(D)[12:55:20]{screen.c:134}	Average screen-emitted brightness: 0.036863.
(D)[12:55:20]{timer.c:32}	Set timeout of 30s 0ns on fd 43.

FedeDP, 2021-03-07:
That is weird; should work fine; 
Can you check output of `busctl --user monitor org.freedesktop.ScreenSaver` while starting/pausing audio play?

promeneur, 2021-03-08:
today it works. Monitor is not switched off !

It seems a random phenomenon.

 busctl --user monitor org.freedesktop.ScreenSaver
Monitoring bus message stream.
‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=7  Timestamp="Mon 2021-03-08 13:09:45 UTC"
  Sender=:1.152  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.152
  MESSAGE "ss" {
          STRING "firefox";
          STRING "audio-playing";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=111  Timestamp="Mon 2021-03-08 13:09:45 UTC"
  Sender=:1.12  Destination=org.kde.Solid.PowerManagement.PolicyAgent  Path=/org/kde/Solid/PowerManagement/PolicyAgent  Interface=org.kde.Solid.PowerManagement.PolicyAgent  Member=AddInhibition
  UniqueName=:1.12
  MESSAGE "uss" {
          UINT32 4;
          STRING "firefox";
          STRING "audio-playing";
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=193  ReplyCookie=111  Timestamp="Mon 2021-03-08 13:09:45 UTC"
  Sender=:1.16  Destination=:1.12
  UniqueName=:1.16
  MESSAGE "u" {
          UINT32 3;
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=112  ReplyCookie=7  Timestamp="Mon 2021-03-08 13:09:45 UTC"
  Sender=:1.12  Destination=:1.152
  UniqueName=:1.12
  MESSAGE "u" {
          UINT32 18065;
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=113  Timestamp="Mon 2021-03-08 13:09:50 UTC"
  Sender=:1.12  Destination=org.kde.Solid.PowerManagement.PolicyAgent  Path=/org/kde/Solid/PowerManagement/PolicyAgent  Interface=org.kde.Solid.PowerManagement.PolicyAgent  Member=HasInhibition
  UniqueName=:1.12
  MESSAGE "u" {
          UINT32 5;
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=197  ReplyCookie=113  Timestamp="Mon 2021-03-08 13:09:50 UTC"
  Sender=:1.16  Destination=:1.12
  UniqueName=:1.16
  MESSAGE "b" {
          BOOLEAN true;
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=8  Timestamp="Mon 2021-03-08 13:11:28 UTC"
  Sender=:1.152  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=UnInhibit
  UniqueName=:1.152
  MESSAGE "u" {
          UINT32 18065;
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=114  Timestamp="Mon 2021-03-08 13:11:28 UTC"
  Sender=:1.12  Destination=org.kde.Solid.PowerManagement.PolicyAgent  Path=/org/kde/Solid/PowerManagement/PolicyAgent  Interface=org.kde.Solid.PowerManagement.PolicyAgent  Member=ReleaseInhibition
  UniqueName=:1.12
  MESSAGE "u" {
          UINT32 3;
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=115  ReplyCookie=8  Timestamp="Mon 2021-03-08 13:11:28 UTC"
  Sender=:1.12  Destination=:1.152
  UniqueName=:1.12
  MESSAGE "" {
  };
[... 80 lines cut ...]
‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=214  ReplyCookie=120  Timestamp="Mon 2021-03-08 13:12:10 UTC"
  Sender=:1.16  Destination=:1.12
  UniqueName=:1.16
  MESSAGE "" {
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=121  ReplyCookie=10  Timestamp="Mon 2021-03-08 13:12:10 UTC"
  Sender=:1.12  Destination=:1.152
  UniqueName=:1.12
  MESSAGE "" {
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1  Priority=0 Cookie=122  Timestamp="Mon 2021-03-08 13:12:10 UTC"
  Sender=:1.12  Destination=org.kde.Solid.PowerManagement.PolicyAgent  Path=/org/kde/Solid/PowerManagement/PolicyAgent  Interface=org.kde.Solid.PowerManagement.PolicyAgent  Member=HasInhibition
  UniqueName=:1.12
  MESSAGE "u" {
          UINT32 5;
  };

‣ Type=method_return  Endian=l  Flags=1  Version=1  Priority=0 Cookie=215  ReplyCookie=122  Timestamp="Mon 2021-03-08 13:12:10 UTC"
  Sender=:1.16  Destination=:1.12
  UniqueName=:1.16
  MESSAGE "b" {
          BOOLEAN false;
  };

FedeDP, 2021-03-11:
Ok! I need same command's output when screensaver inhibition is not working! Thanks!

FedeDP, 2021-07-05:
Closing for now; feel free to reopen this!

closed 2021-07-05 by FedeDP