FedeDP/Clight #204: [BUG]

hunterzero99, 2021-03-10:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Has been happening for both clight-git with clightd-git from AUR.

**Used distribution:**
> Arch Linux 

**Describe the bug**
clight --wizard is outputting all zeros 

**Expected behavior**
regression points should not be all zeros 

**To Reproduce**
run clight --wizard

Please attach a Clight verbose log too!

Welcome to Clight wizard. Press ctrl-c to quit at any moment.
Waiting for sensor...
Sensor available. Start? [Y/n]: > y

Move to nightlight-like light.
Are you ready? [Y/n]

Ambient brightness is currently 0.044; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.000

Move to a dark room.
Are you ready? [Y/n]

Ambient brightness is currently 0.075; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.100

Move to normal-light room.
Are you ready? [Y/n]

Ambient brightness is currently 0.485; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.500

Move to a bright room.
Are you ready? [Y/n]

Ambient brightness is currently 0.562; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.750

Move to daylight-like light.
Are you ready? [Y/n]

Ambient brightness is currently 1.000; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 1.000

Computing new backlight curve...
Computing new regression points...
[ 0.000,  0.000,  0.000,  0.000,  0.000,  0.000,  0.000,  0.000,  0.000,  0.000,  0.000  ]
BL
^
|          
|          
|          
|          
|          
|          
|          
|          
|          
|          
***********>BR
Don't forget to set these points in clight conf file!
Bye!

FedeDP, 2021-03-10:
Hi! Thanks for opening this issue!
I just pushed a small fix on clight master, can you update clight and retry?
Moreover, while retrying, care to also pass `--verbose` flag? 
In case it is not working still, can you paste here output with verbose flag? 
Thank you!

hunterzero99, 2021-03-10:
All fixed!  Wizard output what seems like (at least mostly) valid results:

[ 0.000,  0.289,  0.532,  0.728,  0.877,  0.978,  1.000,  1.000,  1.000,  0.913,  0.778  ]

Verbose log output is below:
```
Welcome to Clight wizard. Press ctrl-c to quit at any moment.
Waiting for sensor...
Set timeout of 0s 1ns on fd 18.
Sensor '/dev/bus/usb/005/002' is now available.
Resumed as a sensor is now available.
Pausing BACKLIGHT
Backlight 'J0DKG7120D3L' level updated: 0.00.
Sensor available. Start? [Y/n]: > 

Move to nightlight-like light.
Are you ready? [Y/n]

Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 0.044680.
Ambient brightness is currently 0.045; redo capture? [y/N]: > y
Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 0.286840.
Ambient brightness is currently 0.287; redo capture? [y/N]: > y
Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 0.044000.
Ambient brightness is currently 0.044; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.000

Move to a dark room.
Are you ready? [Y/n]

Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 0.312760.
Ambient brightness is currently 0.313; redo capture? [y/N]: > y
Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 0.095920.
Ambient brightness is currently 0.096; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.100

Move to normal-light room.
Are you ready? [Y/n]

Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 0.205880.
Ambient brightness is currently 0.206; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.350

Move to a bright room.
Are you ready? [Y/n]

Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 0.660680.
Ambient brightness is currently 0.661; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.600

Move to daylight-like light.
Are you ready? [Y/n]

Captured [5/5] from '/dev/bus/usb/005/002'. Ambient brightness: 1.000000.
Ambient brightness is currently 1.000; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > 
Backlight level is: 0.900

Computing new backlight curve...
0.044 -> 0.000
0.096 -> 0.100
0.206 -> 0.350
0.661 -> 0.600
1.000 -> 0.900
New curve: y = -0.001340 + 1.255462x + -0.377419x^2
Computing new regression points...
[ 0.000,  0.289,  0.532,  0.728,  0.877,  0.978,  1.000,  1.000,  1.000,  0.913,  0.778  ]
BL
^
|    ****  
|   *    * 
|         *
|  *       
|          
| *        
|          
|*         
|          
|          
*---------->BR
Don't forget to set these points in clight conf file!
Bye!
```

FedeDP, 2021-03-10:
Mmh i think there is still something wrong with the regression points compute! I will check again!

FedeDP, 2021-03-10:
Ok, i found the second stupid bug :D 
Can you check again? Thank you!

hunterzero99, 2021-03-10:
Looks good now, well done!

[ 0.000,  0.126,  0.295,  0.445,  0.578,  0.693,  0.790,  0.868,  0.929,  0.971,  0.996  ]

commit 05e27e76 (2021-03-10): Fixed another small stupid bug in expand_regr_points. Fixes #204.