FedeDP/Clight #209: Support for external display and external cam?

samuaz, 2021-05-06:
Hello, thanks for the software is working pretty good with my oled alienware with kernel 5.12 under archilunux.

Want to know if it is possible to support an external display?

My setup with an external display connected to my laptop, and use my laptop display as secundary display.

but also i have external camera in top of the external display.

So would be nice to can also control the external display brightness? for now im using a  DDC app under linux to change the external display brightness from linux

thanks.

FedeDP, 2021-05-09:
> Hello, thanks for the software is working pretty good with my oled alienware with kernel 5.12 under archilunux.

Great! :) You're welcome!

> i checking that the increase of the brightness for the external display is working but not the decreasing

Wow, that is really weird; Clightd uses ddcutil library to manage external displays brightness; it should obviously work in both ways (ie: increasing and decreasing);
can you try in terminal:
`sudo ddcutil setvcp 10 50` (it should set your external monitor backlight to 50%)
and then `sudo ddcutil setvcp 10 100` (to set it to 100%)
If they work, clightd works too!

> My setup with an external display connected to my laptop, and use my laptop display as secondary display. But also I have external camera in top of the external display. Both cameras laptop and external has an infrared sensor
Another questions I'm checking that in low light conditions ant night in my house the ambient sensor is reporting 100% but i think that level is so high for the actual light condition is there anyway to calibrate the ambient sensor?

I don't know about infrared sensor; maybe it is breaking Clight algorithm to compute ambient brightness (ie: kicking in on low-light conditions)?
Other than that, yours seems a pretty standard use case (mine too!).

Can you try 
`busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 ""`
in terminal, in various different ambient light conditions?

samuaz, 2021-05-11:
Thanks for quickly response, both ddcutil commands working perfectly

and about 

`busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 ""
`
this is the output at night with a room lights on:
`sad "/dev/video0" 5 0.990141 0.991419 0.991639 0.991218 0.991224
`
and this is the output at night with room lights off:

`sad "/dev/video0" 5 0.105882 0.121569 0.12549 0.129412 0.14902
`

FedeDP, 2021-05-14:
Wow, they both seems ok!
Can you share a clight verbose log? Thank you very much!

samuaz, 2021-07-03:
Hi sorry delay, i can confirm that the exterrnal display was working on ubuntu based distro popOs perfectly. now i move to archlinux base endeavour os and i have a different issue just clight is not working i try to force the capture but nothing happen and the sensors just say 0 this is the log:

`
samuaz@M15r3 ~]$ clight
Clightd found, version: 5.4.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
|          
*---------->BR
Keyboard backlight calibration unsupported.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
AC cable connected.
Laptop lid opened.
Error loading from cache file.

`

~/.cache/clight is empty

clighd status

`● clightd.service - Bus service to manage various screen related properties (ga>
     Loaded: loaded (/etc/systemd/system/clightd.service; enabled; vendor prese>
     Active: active (running) since Fri 2021-07-02 22:48:11 -03; 26min ago
   Main PID: 28412 (clightd)
      Tasks: 3 (limit: 38179)
     Memory: 1.2M
        CPU: 13.084s
     CGroup: /system.slice/clightd.service
             └─28412 /usr/libexec/clightd

jul 02 22:48:11 M15r3 systemd[1]: Starting Bus service to manage various screen>
jul 02 22:48:11 M15r3 systemd[1]: Started Bus service to manage various screen >
`

FedeDP, 2021-07-05:
Hi! It seems like Clight is awaiting on a user position by Geoclue; is it installed?
Can you check https://github.com/FedeDP/Clight/wiki/FAQ#geoclue-asks-for-permission or https://github.com/FedeDP/Clight/wiki/FAQ#clight-takes-up-to-20s-to-start ?

FedeDP, 2021-08-21:
Closing; feel free to reopen the issue if you are still facing it!
(btw geoclue code was recently revisited with better error handling and making it async).

closed 2021-08-21 by FedeDP