FedeDP/Clight #210: Cannot set gamma on Wayland

logix2, 2021-05-14:
Using Ubuntu 21.04 with Wayland (Intel graphics), on an Asus Zenbook, Clight installed from git (using the script from the wiki) cannot set the gamma. 

```
$ clight --verbose

Clightd found, version: 5.4.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
|          
*---------->BR
Keyboard backlight calibration supported.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
Disarmed timerfd on fd 40.
AC cable disconnected.
Laptop lid opened.
Emitting 'AcState' property
Disarmed timerfd on fd 43.
Emitting 'LidState' property
REDACTED loaded from cache file.
Loc distance:REDACTED -> REDACTED : 3087.66 km.
New location received: REDACTED.
Set timeout of 0s 1ns on fd 44.
Emitting 'Location' property
Next alarm due to: Fri May 14 20:01:00 2021
Set timeout of 25550s 0ns on fd 44.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
Set timeout of 0s 1ns on fd 45.
Sensor '/dev/video0' is now available.
Resumed as a sensor is now available.
Emitting 'NextEvent' property
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.989.
Ambient brightness: 0.989 -> Backlight pct: 0.798.
Set timeout of 1200s 0ns on fd 45.
Emitting 'AmbientBr' property
set_temp(): No plugin available for your configuration.
Failed to set gamma temperature.
Emitting 'BlPct' property
Emitting 'KbdPct' property
REDACTED received from Geoclue2.
Backlight 'intel_backlight' level updated: 0.80.
Emitting 'BlPct' property
Loc distance: REDACTED -> REDACTED : 0.01 km.
Failed to validate location request.
```

And the systemd log:
```
May 14 12:54:10 logix-laptop systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
May 14 12:54:44 logix-laptop clightd[71500]: open: No such file or directory
```

Let me know if more information is needed.

FedeDP, 2021-05-14:
Hi! Thanks for opening this issue!

> set_temp(): No plugin available for your configuration.

Unfortunately, there is no standard way on wayland to manage screen temperature. If you're using ubuntu, i guess you're using gnome.
Clightd implements wlroots gamma protocol for wayland support: https://github.com/FedeDP/Clight/wiki/Modules#wayland-support (that redirects to https://github.com/FedeDP/Clightd/wiki/Gamma#Wl).
It is only available on wlroots based compositor, not much i can do, sorry!
Hopefully, someday a standardized protocol will appear...
For now, you can just let Gnome manage screen temperature instead of Clight, disabling clight GAMMA feature.

logix2, 2021-05-14:
Hi! Got it, thank you for the answer!

closed 2021-05-14 by logix2

FedeDP, 2021-05-14:
You're welcome :)