FedeDP/Clight #211: [BUG] clight adjusts brightness once only

Hampycalc, 2021-05-23:
**Clight version the issue has been seen with:**
> 5.3

**Used distribution:**
> Arch AUR

**Describe the bug**
Running `clight  --verbose` will perform a brightness adjustment once and only once, immediately when run. The verbose output is:
```
Config file /home/tom/.config/clight.conf not found.
Clightd found, version: 5.3.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
|          
*---------->BR
Keyboard backlight calibration supported.
Disarmed timerfd on fd 38.
upower_check(): No route to host
UPower not present. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Emitting 'AcState' property
Set timeout of 0s 1ns on fd 41.
Emitting 'LidState' property
Set timeout of 30s 0ns on fd 41.
Error loading from cache file.
geoclue_get_client(): The name org.freedesktop.GeoClue2 was not provided by any .service files
Geoclue2 appears to be unsupported.
Failed to init.
No location provider found; fallback to DAY daytime.
Emitting 'Location' property
Emitting 'DayTime' property
Killing GAMMA as no location provider is available.
Set timeout of 0s 1ns on fd 15.
Sensor '/dev/iio:device0' is now available.
Resumed as a sensor is now available.
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.044000.
Ambient brightness: 0.044 -> Backlight pct: 0.058.
Set timeout of 600s 0ns on fd 15.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.24.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.14.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.09.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.06.
Emitting 'BlPct' property
Set timeout of 30s 0ns on fd 41.
Set timeout of 30s 0ns on fd 41.
Set timeout of 30s 0ns on fd 41.
Set timeout of 30s 0ns on fd 41.
Set timeout of 30s 0ns on fd 41.
Set timeout of 30s 0ns on fd 41.
Set timeout of 30s 0ns on fd 41
```

**Expected behavior**
I would expect the screenbrightness to continue to be adjusted over time

**To Reproduce**
Run `clight --verbose`

FedeDP, 2021-05-23:
Hi! Thanks for opening this bug report!
I see:
> Ambient brightness: 0.044 -> Backlight pct: 0.058.
Set timeout of 600s 0ns on fd 15.

Clight does not continuously monitor ambient brightness; it does only a "capture + backlight calibration" every N minutes (where N varies between ac/battery).
Are you sure Clight is not kicking in every once in a while?
You can find its timeouts for backlight calibration in conf file, eg: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L68

Hampycalc, 2021-05-23:
It is indeed working correctly. My apologies - I did not find that setting. I have now set the timeout to 10 seconds as I have an inbuilt ambient light sensor on my laptop and everything is working great.
Many thanks for the awesome software.

closed 2021-05-23 by Hampycalc

FedeDP, 2021-05-23:
You're welcome! Thanks for your kind words :)