FedeDP/Clight #212: [BUG] Failed to retrieve AC state; fallback to ON_AC and OPEN lid.

patri9ck, 2021-05-31:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.5.r9.g93407c8-1

**Used distribution:**
> Arch Linux (AUR package)

**Description**
First: I do not know whether the issue is called because of the message in the title. I do not even know whether this message is just a warning message or an actual error.

When I start Clight using `clight --ambient-gamma`, this is the output:
```
Clightd found, version: 5.4.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
|          
*---------->BR
init_kbd_backlight(): No such file or directory
Keyboard backlight calibration unsupported.
Disarmed timerfd on fd 38.
upower_check(): Permission denied
UPower not present. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Emitting 'AcState' property
Set timeout of 0s 1ns on fd 41.
Emitting 'LidState' property
Set timeout of 30s 0ns on fd 41.
Error loading from cache file.
```
For some reason, Clight outputs another version than `pacman -Qi clight`.

Nothing happens. I am using an external USB camera which works with every other application. I feel like Clight does not detect my camera but I don't know how to check it.
**Expected behavior**
My color temperature should change after I start Clight.

**To Reproduce**
See above.

FedeDP, 2021-05-31:
Hi! Thanks for openining this PR!

The "failed to retrieve AC state" is a warning level log (if you check clight log file, each line is prepended with a "(W)" or "(E)" etc etc.
It just means you haven't got UPower installed thus clight is not able to determine if you're on AC or on BATTERY, thus it fallbacks to AC and open lid.

> (For some reason, Clight outputs another version than pacman -Qi clight.)

Clight outputs **Clightd** version (note the final 'd').

> Nothing happens then.

It seems like Clight is awaiting on getting locaiton info from Geoclue. Have you got geoclue installed?
You can try to set a fixed location in clight conf file!

patri9ck, 2021-05-31:
Yes, I have Geoclue installed and it also seems to be working. I am testing it with `/usr/lib/geoclue-2.0/demos/agent` and `/usr/lib/geoclue-2.0/demos/where-am-i`. But why does Clight even need Geoclue in the first place. Shouldn't it set the color temperature etc. using my webcam?

FedeDP, 2021-05-31:
Clight uses Geoclue to compute Sunrise/Sunset times for your position! It is needed to have different timeouts between day and night for backlight calibration, and to properly set screen temperature at day/night.
About your issue: https://github.com/FedeDP/Clight/wiki/FAQ#clight-takes-up-to-20s-to-start

patri9ck, 2021-05-31:
Thanks, I will take a look at it.

closed 2021-05-31 by patri9ck

FedeDP, 2021-05-31:
You're welcome! Thanks for taking the time to try Clight :)