FedeDP/Clight #213: [FEATURE REQ] Be able to toggle Clight or reset the changes made by it

patri9ck, 2021-05-31:
Currently I'm toggling Clight by killing the process and restarting it. But when I kill it, the changed made by Clight stay. Therefore it would be nice if Clight reset the changes by itself when it's killed or if it provided a command to reset it. Alternatively, Clight could have a command like `clight --toggle` which then does everything for you.

FedeDP, 2021-06-01:
Hi! Thanks for opening this feature request!
What would you like to be restored? eg: screen temperature and backlight level? Only screen temp?

Note however, that you can disable each feature from clight conf file; in case you only want to trigger backlight calibration manually, you should set "no_auto_calibration = true" in conf file: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L85; then, if you only wish to let clight calibrate the backlight and nothing more, you should also disable gamma, dpms, dimmer, screen and keyboard modules.

patri9ck, 2021-06-01:
Thanks for the answer. I would like to have really everything restored (I guess that would be backlight/gamma and screen temperature? Did I miss something?). But that is just my personal preference. Therefore I think other people would like to restore only a few things so a command line or config option would probably be very helpful.

FedeDP, 2021-06-08:
Uh, gamma is screen temperature (yeah, i should really update its name...).
I think only backlight and gamma should be restored; i think i can add a "restore = true" option in clight conf file (by default disabled).

patri9ck, 2021-06-08:
Yes, that would be very nice.

closed 2021-07-14 by FedeDP

FedeDP, 2021-07-15:
You're welcome!

commit ee220be3 (2021-07-14): Added options to restore screen backlight and temperature upon clight exit. Fixes #213.