FedeDP/Clight #214: [FEATURE REQ] Tell Clight to use only a certain area of the camera

patri9ck, 2021-06-01:
When the lights on my room are turned off and I open a window with dark colors (like my terminal, which has a grey background), my monitor does not light up my room as much as when I open a window with bright colors (like Google Chrome, when I am on a website with a white background). Therefore Clight changes my brightness significantly when I change my window. It detects a brightness of 0.1 when I open up my terminal and a brightness of 0.9 when I open Google Chrome on a website with a white background.

One solution could be that the user is able to specify a certain area on the camera which Clight should use to calibrate all the things. For example only a few pixels on the upper right corner which are not affected by the monitor.

FedeDP, 2021-06-08:
Hi! This is what SCREEN module is for: it compensates for screen-emitted brightness; it is highly simple but should work fine, at least it tries :) 

Anyway, the suggestion is good! I can add a camera sensor specific parameter to let users specify the area used to compute ambient backlight, eg: something like: `x: 0.4-0.6,y: 0.2,0.3` to only use X between 40% and 60% of image length, and Y between 20% and 30% (starting from top left), or something like that.
Thanks for your input!

patri9ck, 2021-06-08:
Thank you! I will take a look at the module.

FedeDP, 2021-07-05:
Is SCREEN module working fine?

patri9ck, 2021-07-15:
It's kinda working but it's to be honest not that satisfying.

FedeDP, 2021-07-15:
I will try to add the new feature in the next clightd release cycle!

FedeDP, 2021-07-17:
Move to Clightd.