FedeDP/Clight #215: [BUG] Clight can't interact with devices while Clightd works as expected

PJungkamp, 2021-06-09:
**Clight version the issue has been seen with:**
> Clight: 4.6 
> Clightd: 

**Used distribution:**
> Arch Linux (AUR: clight-git & clightd-git)

**Describe the bug**
I've never used clight before and cant get it working.

clightd dbus calls work as expected:
```
% busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor IsAvailable "s" ""
sb "/dev/iio:device1" true 
```
```
% busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 ""
sad "/dev/iio:device1" 5 0.235928 0.235928 0.235928 0.235928 0.235928
```
```
[clight.log](https://github.com/FedeDP/Clight/files/6625768/clight.log)
% busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight GetAll "s" ""
a(sd) 1 "intel_backlight" 0.3
```
```
% busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight Set "d(bdu)s" 0.6 true 0.02 50 ""
b true
```

But clight can't set backlight brightness. This does nothing:
```
% busctl --user call org.clight.clight /org/clight/clight org.clight.clight IncBl "d" 0.4
```

clight-gui can set the brightness in the backlight tab, but shows ALS light level always as 0.

the tray context menu of clight-gui always shows brightness and ambient light as 0 and the increase or decrease brightness menu entries don't do anything.

I suspect the problem lies within clight since the clightd dbus api seems and thus filed the issue here.

**Expected behavior**
clight sets brightness based on ALS.

**To Reproduce**
1. I start clight
2. it doesn't seem to do anything

Verbose log: [clight.log](https://github.com/FedeDP/Clight/files/6625778/clight.log)

nullobsi, 2021-06-09:
For clarification, clight-gui uses ClightD directly on the Brightness tab while using the Clight reported value for the tray icon

FedeDP, 2021-06-09:
Uh, didnt see the verbose log attached. Sorry am from mobile :)
I will give it a look asap!

FedeDP, 2021-06-09:
@PJungkamp it seems like you were in dimmed state: 
> (D)[19:13:06]{display.c:34}	Entering dimmed state...
(D)[19:13:06]{display.c:40}	A lower than dimmer_pct backlight level is already set. Avoid changing it.

In dimmed state, backlight requests are ignored by clight, see: https://github.com/FedeDP/Clight/blob/master/src/modules/backlight.c#L281.

Moreover, it seems like Clight is waiting on user position from geoclue.

PJungkamp, 2021-06-10:
I added the clightc section to the geoclue config (as noted FAQ) and it fixed it!

I hadn't recognized that I've installed geoclue (it was installed as a dependency of xdg-dektop-portal for webrtc screen sharing on sway).

Thank you!

closed 2021-06-10 by PJungkamp