FedeDP/Clight #216: [BUG]  Error loading from cache file

Hampycalc, 2021-06-15:
**Clight version the issue has been seen with:**
4.5

**Used distribution:**
arch

**Describe the bug**
clight fails to load with the log:
```Clightd found, version: 5.3.
AC curve: y = 0.004196 + 0.180627x + -0.008054x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
*          
+---------->BR
BATT curve: y = 0.018392 + 0.142830x + -0.006410x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
*          
+---------->BR
Keyboard backlight calibration supported.
Disarmed timerfd on fd 38.
upower_check(): No route to host
UPower not present. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Emitting 'AcState' property
Set timeout of 0s 1ns on fd 39.
Emitting 'LidState' property
Set timeout of 30s 0ns on fd 39.
Error loading from cache file.
```

I cannot find any cache file to remove/purge. Reinstalling that package does not fix the problem

**Expected behavior**
clight should run without error

Hampycalc, 2021-06-16:
An update:
If, when this error occurs (the executable hangs, rather than exits) I run CTRL+C and re-execute "clight", something it will run correctly. It will always display "Error loading from cache file." and also states "failed to init" but will run regardless.
It runs correctly approximately 1 in 5 attempt on my machine.
For reference, this is all prior to loading any sort of DE or WM.

Hampycalc, 2021-06-16:
Also - I'm running clightd v5.3

Hampycalc, 2021-06-16:
I "fixed" the problem...
It seems from the source code that this error occurs when trying to read/write the longitude and latitude settings. I did not have longitude and latitude set in my config. I entered both of these and the problem went away. Possibly, this is a coincidence....

FedeDP, 2021-06-20:
Hi!
Sorry for being late and thanks for opening the bug report!
The "Error loading from cache file" is indeed just a warning: clight stores current user position in a cache file when leaving; that way it can quickly load latest user position on restart/boot.
It seems that, in your case, Clight fails to find its cache file; moreover, geoclue is slow to give a position; thus Clight "stalls" waiting on a position.

Most probably you are being hit by: https://github.com/FedeDP/Clight/wiki/FAQ#clight-keeps-running-even-after-my-session-was-killed or https://github.com/FedeDP/Clight/wiki/FAQ#clight-takes-up-to-20s-to-start

closed 2021-08-21 by FedeDP