FedeDP/Clight #217: USB Camera not detected

yochananmarqos, 2021-07-02:
**Clight version the issue has been seen with:**
`clight-git` 4.5.r9.g93407c8-1
`clightd-git` 5.3.r8.gb9d81a3-1

**Used distribution:**
Manjaro GNOME

**Describe the bug**

Clight works fine with my built-in laptop webcam, but now I have two monitors connected and keep the laptop lid closed. I thought I would use my old Logitech QuickCam Communicate STX for Clight, but I can't seem to get it to work. It's an older one I received for free around 2005. It works with Guvcview but is not detected by Cheese.

I tried setting `devname` to `"video2";` and `"/dev/video2";`, but it didn't make a difference.

```
❯ v4l2-ctl --device /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
	[1]: 'YUYV' (YUYV 4:2:2)v4l2-ctl -d /dev/video2 --list-ctrls
```

```
❯ v4l2-ctl -d /dev/video2 --all
Driver Info:
	Driver name      : gspca_zc3xx
	Card type        : USB Camera (046d:08d7)
	Bus info         : usb-0000:02:00.0-1
	Driver version   : 5.12.14
	Capabilities     : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (gspca_zc3xx: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'JPEG' (JFIF JPEG)
	Field             : None
	Bytes per Line    : 640
	Size Image        : 115790
	Colorspace        : JPEG
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
JPEG compression:
	Quality: 75
	Markers: 0x00000018
		Define Huffman Tables
		Define Quantization Tables
Streaming Parameters Video Capture:
	Frames per second: invalid (0/0)
	Read buffers     : 2

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
                          gamma 0x00980910 (int)    : min=1 max=6 step=1 default=4 value=4 flags=slider
                       exposure 0x00980911 (int)    : min=781 max=18750 step=1 default=2343 value=2343 flags=inactive, volatile
                 gain_automatic 0x00980912 (bool)   : default=1 value=1 flags=update
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=0 value=2
				0: Disabled
				1: 50 Hz
				2: 60 Hz
                      sharpness 0x0098091b (int)    : min=0 max=3 step=1 default=2 value=2 flags=slider

JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=50 max=87 step=1 default=75 value=75
```

clight.log:
```
(W)[09:20:28]{config.c:375}	Config file /home/yochanan/.config/clight.conf not found.

Clight
* Software version:		4.6
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Fri Jul  2 09:20:28 2021

Starting options:

### GENERIC ###
* Verbose (debug):		Enabled
* ResumeDelay:		0

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth steps:		0.05
* Smooth timeout:		30
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Backlight path:		Unset
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled
* Capture on lid opened:		Disabled

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		/dev/video2
* Settings:		Unset

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800

### DPMS ###
* Timeouts:		AC 900	BATT 300

### SCREEN ###
* Timeouts:		AC 30	BATT -1
* Contrib:		0.10
* Samples:		10

### INHIBIT ###
* Docked:		Disabled
* PowerManagement:		Disabled
* Backlight:		Disabled

(I)[09:20:28]{main.c:127}	Clightd found, version: 5.4.
(I)[09:20:28]{backlight.c:514}	AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
(I)[09:20:28]{backlight.c:514}	BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
|          
|**        
|          
*---------->BR
(W)[09:20:28]{interface.c:403}	Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[09:20:28]{interface.c:404}	Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
(D)[09:20:28]{timer.c:34}	Disarmed timerfd on fd 36.
(I)[09:20:28]{upower.c:54}	AC cable connected.
(D)[09:20:28]{interface.c:456}	Emitting 'AcState' property
(D)[09:20:28]{timer.c:32}	Set timeout of 0s 1ns on fd 37.
(D)[09:20:28]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:20:28]{location.c:95}	40.43 -105.09 loaded from cache file.
(D)[09:20:28]{my_math.c:248}	Loc distance: 40.43,-105.09 -> 91.00,181.00 : 3444.24 km.
(I)[09:20:28]{location.c:62}	New location received: 40.43, -105.09.
(D)[09:20:28]{timer.c:32}	Set timeout of 0s 1ns on fd 38.
(D)[09:20:28]{interface.c:456}	Emitting 'Location' property
(I)[09:20:28]{daytime.c:175}	Next alarm due to: Fri Jul  2 20:02:00 2021
(D)[09:20:28]{timer.c:32}	Set timeout of 38492s 0ns on fd 38.
(D)[09:20:28]{interface.c:456}	Emitting 'Sunrise' property
(D)[09:20:28]{interface.c:456}	Emitting 'Sunset' property
(D)[09:20:28]{interface.c:456}	Emitting 'DayTime' property
(D)[09:20:28]{interface.c:456}	Emitting 'NextEvent' property
(D)[09:20:58]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:21:28]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:21:58]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:22:28]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:22:58]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
```

FedeDP, 2021-07-05:
Hi! Thanks for opening the bug report;
Clight should obviously be able to handle external cameras as well as internal ones, even hotplugging them.
`v4l2-ctl` output looks fine for both usb cameras!

> ❯ busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor Capture "sis" "" 5 ""
sad "/dev/video2" 5 0.919776 0.920563 0.921465 0.921133 0.920143

seems fine!
Can you try with:

`busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor Capture "sis" "/dev/video2" 5 ""`
?
In your case, that should be the exact same dbus call that Clight does.

yochananmarqos, 2021-07-05:
Same result with both cameras:

```
❯ busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor Capture "sis" "/dev/video2" 5 ""
sad "/dev/video2" 5 0.176471 0.145098 0.172549 0.172549 0.176471
```

Fresh log trying both cameras. Neither camera took a light sample and the backlight did not change:

```
(W)[09:11:22]{interface.c:403}	Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[09:11:22]{interface.c:404}	Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
(D)[09:11:22]{timer.c:34}	Disarmed timerfd on fd 36.
(I)[09:11:22]{upower.c:54}	AC cable connected.
(D)[09:11:22]{interface.c:456}	Emitting 'AcState' property
(D)[09:11:22]{timer.c:32}	Set timeout of 0s 1ns on fd 37.
(D)[09:11:22]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:11:22]{location.c:95}	40.43 -105.09 loaded from cache file.
(D)[09:11:22]{my_math.c:248}	Loc distance: 40.43,-105.09 -> 91.00,181.00 : 3444.24 km.
(I)[09:11:22]{location.c:62}	New location received: 40.43, -105.09.
(D)[09:11:22]{timer.c:32}	Set timeout of 0s 1ns on fd 38.
(D)[09:11:22]{interface.c:456}	Emitting 'Location' property
(I)[09:11:22]{daytime.c:175}	Next alarm due to: Mon Jul  5 20:02:00 2021
(D)[09:11:22]{timer.c:32}	Set timeout of 39038s 0ns on fd 38.
(D)[09:11:22]{interface.c:456}	Emitting 'Sunrise' property
(D)[09:11:22]{interface.c:456}	Emitting 'Sunset' property
(D)[09:11:22]{interface.c:456}	Emitting 'DayTime' property
(D)[09:11:22]{interface.c:456}	Emitting 'NextEvent' property
(D)[09:11:52]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:12:22]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:12:52]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:13:22]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:13:52]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:14:22]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:14:52]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:15:22]{screen.c:147}	Start compensating for screen-emitted brightness.
(D)[09:15:22]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
```

I noticed some Libmodule messages in the service log with either camera:

```
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
     Loaded: loaded (/etc/systemd/system/clightd.service; disabled; vendor preset: disabled)
     Active: active (running) since Mon 2021-07-05 09:11:18 MDT; 1min 30s ago
   Main PID: 336712 (clightd)
      Tasks: 3 (limit: 38335)
     Memory: 2.0M
        CPU: 349ms
     CGroup: /system.slice/clightd.service
             └─336712 /usr/lib/clightd/clightd

Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
Jul 05 09:12:48 serval clightd[336712]: Libmodule @ stack_peek:112| (stack_length(s) > 0) condition failed.
```

FedeDP, 2021-07-05:
> ❯ busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor Capture "sis" "/dev/video2" 5 ""
sad "/dev/video2" 5 0.176471 0.145098 0.172549 0.172549 0.176471

So, video2 is working fine with Clightd!

> Fresh log trying both cameras. Neither camera took a light sample and the backlight did not change:

Is BACKLIGHT module enabled? I mean, i'd expect a log like this:

```
(I)[17:21:28]{keyboard.c:129}   Keyboard backlight calibration supported.
(W)[17:21:28]{interface.c:403}  Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[17:21:28]{interface.c:404}  Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
(D)[17:21:28]{timer.c:34}       Disarmed timerfd on fd 44.
(I)[17:21:28]{upower.c:54}      AC cable disconnected.
(I)[17:21:28]{upower.c:65}      Laptop lid opened.
(D)[17:21:28]{interface.c:456}  Emitting 'AcState' property
(D)[17:21:28]{timer.c:34}       Disarmed timerfd on fd 47.
(D)[17:21:28]{interface.c:456}  Emitting 'LidState' property
(D)[17:21:28]{location.c:95}    45.65 9.15 loaded from cache file.
(D)[17:21:28]{my_math.c:248}    Loc distance: 45.65,9.15 -> 91.00,181.00 : 2995.88 km.
(I)[17:21:28]{location.c:62}    New location received: 45.65, 9.15.
(D)[17:21:28]{timer.c:32}       Set timeout of 0s 1ns on fd 48.
(D)[17:21:28]{interface.c:456}  Emitting 'Location' property
(I)[17:21:28]{daytime.c:175}    Next alarm due to: Mon Jul  5 20:43:00 2021
(D)[17:21:28]{timer.c:32}       Set timeout of 12092s 0ns on fd 48.
(D)[17:21:28]{interface.c:456}  Emitting 'Sunrise' property
(D)[17:21:28]{interface.c:456}  Emitting 'Sunset' property
(I)[17:21:28]{test.c:17}        We're now during the day!
(D)[17:21:28]{interface.c:456}  Emitting 'DayTime' property
(I)[17:21:28]{gamma.c:186}      Normal transition to 6500 gamma temp.
(D)[17:21:28]{interface.c:456}  Emitting 'NextEvent' property
(D)[17:21:28]{timer.c:32}       Set timeout of 0s 1ns on fd 53.
(D)[17:21:28]{backlight.c:314}  Sensor '/dev/video0' is now available.
(D)[17:21:28]{backlight.c:601}  Resumed as a sensor is now available.
(D)[17:21:28]{interface.c:456}  Emitting 'Temp' property
(D)[17:21:28]{interface.c:456}  Emitting 'Temp' property
(D)[17:21:28]{interface.c:456}  Emitting 'SensorAvail' property
(D)[17:21:30]{backlight.c:328}  Captured [5/5] from '/dev/video0'. Ambient brightness: 0.992.
(I)[17:21:30]{backlight.c:364}  Ambient brightness: 0.992 -> Backlight pct: 0.799.
```

> I noticed some Libmodule messages in the service log with either camera:

Those are normal logs :)

yochananmarqos, 2021-07-05:
Yes, the backlight module is enabled.

After unplugging the Communicate STX and plugging the C270 back in, the results changed:

```
❯ busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor Capture "sis" "" 5 ""
sad "/dev/video0" 5 0.913339 0.913191 0.913457 0.915879 0.912542
❯ busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf SensorName "s" "/dev/video2"
Failed to set property SensorName on interface org.clight.clight.Conf: Unknown interface org.clight.clight.Conf or property SensorName.
```

FedeDP, 2021-07-05:
Uh, i now see what's happening! Clight is autopausing BACKLIGHT module because laptop lid is closed i guess!

THere was a bug in the lid state initialization; can you try upgrading clight?

yochananmarqos, 2021-07-05:
That seems to have done the trick. However, the light sensor sucks on both cameras. It set the brightness to 25% in a room with daylight. I adjusted the shades and turned the ceiling light on and it went up to 35%. Ugh.

```
(W)[09:32:24]{config.c:375}	Config file /home/yochanan/.config/clight.conf not found.

Clight
* Software version:		4.6
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Mon Jul  5 09:32:24 2021

Starting options:

### GENERIC ###
* Verbose (debug):		Enabled
* ResumeDelay:		0

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth steps:		0.05
* Smooth timeout:		30
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Backlight path:		Unset
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled
* Capture on lid opened:		Disabled

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		video2
* Settings:		Unset

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800

### DPMS ###
* Timeouts:		AC 900	BATT 300

### SCREEN ###
* Timeouts:		AC 30	BATT -1
* Contrib:		0.10
* Samples:		10

### INHIBIT ###
* Docked:		Disabled
* PowerManagement:		Disabled
* Backlight:		Disabled

(I)[09:32:24]{main.c:127}	Clightd found, version: 5.4.
(I)[09:32:24]{backlight.c:514}	AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
(I)[09:32:24]{backlight.c:514}	BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
[... 108 lines cut ...]
(D)[09:36:25]{screen.c:147}	Start compensating for screen-emitted brightness.
(D)[09:36:25]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:36:55]{screen.c:134}	Average screen-emitted brightness: 0.044392.
(D)[09:36:55]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:36:55]{interface.c:456}	Emitting 'ScreenComp' property
```

Swapped from the C270 to the Communicate:
```
(D)[09:37:09]{backlight.c:314}	Sensor '/dev/video2' is now available.
(D)[09:37:09]{backlight.c:314}	Sensor '/dev/video2' is now available.
(D)[09:37:25]{screen.c:134}	Average screen-emitted brightness: 0.047569.
(D)[09:37:25]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[09:37:25]{interface.c:456}	Emitting 'ScreenComp' property
(D)[09:37:39]{backlight.c:328}	Captured [5/5] from '/dev/video2'. Ambient brightness: 0.225.
(I)[09:37:39]{backlight.c:362}	Ambient brightness: 0.225 (-0.048 screen compensation) -> Backlight pct: 0.287.
(D)[09:37:39]{interface.c:456}	Emitting 'AmbientBr' property
(D)[09:37:39]{interface.c:456}	Emitting 'BlPct' property
(D)[09:37:39]{backlight.c:618}	Backlight '1CR1060B60' level updated: 0.30.
(D)[09:37:39]{interface.c:456}	Emitting 'BlPct' property
(D)[09:37:39]{backlight.c:618}	Backlight 'nvidia_0' level updated: 0.28.
(D)[09:37:40]{backlight.c:618}	Backlight 'nvidia_0' level updated: 0.28.
(D)[09:37:40]{backlight.c:618}	Backlight '1CR1060B60' level updated: 0.29.
(D)[09:37:40]{interface.c:456}	Emitting 'BlPct' property
```

FedeDP, 2021-07-05:
> That seems to have done the trick.

That's great!

> It set the brightness to 25% in a room with daylight.

You can try playing around with sensor settings: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L118

yochananmarqos, 2021-07-05:
That's a bit over my head. I really didn't understand that bit in the wiki.

FedeDP, 2021-07-05:
No problem, it is quite hard to explain!
You can customize webcam parameters with that string from clight; in the 
> 10094849=1,10094850=166

example, you are telling Clightd to set exposure to manual and use a 166 absolute exposure.
Values are taken from https://github.com/torvalds/linux/blob/master/include/uapi/linux/v4l2-controls.h.

In your case, most probably exposure is what you are looking for, thus you can try with:
`10094849=1,10094850=166` 
and change the `166` value until you are satisfied.

To quickly check, you can use Clightd:
`busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=166"`

Some example output from my laptop:
```
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=166"sad "/dev/video0" 5 0.987421 0.988139 0.988406 0.988011 0.987929
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 ""
sad "/dev/video0" 5 0.991378 0.990845 0.990624 0.991038 0.991274
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=120"
sad "/dev/video0" 5 0.954657 0.951261 0.944336 0.946891 0.945686
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=10"
sad "/dev/video0" 5 0.952471 0.983956 0.196078 0.207843 0.207843
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=100"
sad "/dev/video0" 5 0.880083 0.880307 0.880073 0.884266 0.880661
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=90"
sad "/dev/video0" 5 0.865359 0.863399 0.862332 0.865229 0.855576
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=70"
sad "/dev/video0" 5 0.863757 0.982622 0.754902 0.794925 0.789542
[federico][DPFXPS][~]
$ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=50"
sad "/dev/video0" 5 0.669039 0.658461 0.663302 0.658131 0.658602
```

yochananmarqos, 2021-07-05:
That failed for some reason:

```
❯  busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "" 5 "10094849=1,10094850=186"
Call failed: No such device
```

Specifying `video2` worked, but it doesn't seem to make a difference. Is a higher number a longer exposure?
```
❯ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "/dev/video2" 5 "10094849=1,10094850=186"
sad "/dev/video2" 5 0.180392 0.184314 0.176471 0.176471 0.172549
❯ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "/dev/video2" 5 "10094849=1,10094850=206"
sad "/dev/video2" 5 0.180392 0.172549 0.172549 0.180392 0.180392
❯ busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "sis" "/dev/video2" 5 "10094849=1,10094850=106"
sad "/dev/video2" 5 0.184314 0.184314 0.172549 0.176471 0.184314
```

FedeDP, 2021-07-05:
> Specifying video2 worked,

Weird, maybe without specifying "video2", it tries to use video0 but video0 has no power because laptop lid is closed? 
What's the output of `ls -la /dev/video*` ?

>  Is a higher number a longer exposure?

Yep!
Taken from v4l2-ctl output you posted above:
```
               exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
				1: Manual Mode
				3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=166 value=667 flags=inactive
```

yochananmarqos, 2021-07-05:
```
❯ ls -la /dev/video*
crw-rw----+ 1 root video 81, 0 Jul  5 10:41 /dev/video0
crw-rw----+ 1 root video 81, 2 Jul  4 14:56 /dev/video2
crw-rw----+ 1 root video 81, 3 Jul  4 14:56 /dev/video3
```

Ah, no wonder. My Communicate STX is connected now and it suddenly decided it's now `video0`. Not sure why my laptop webam (Chicony) shows up as two devices. It does in Cheese and Guvcview, too.

My guess is when the laptop lid is closed and I plug in a USB camera, it defaults to `video0`. So in theory I shouldn't have to change the `devname` value.

```
❯ v4l2-ctl -d /dev/video0 --all
Driver Info:
	Driver name      : gspca_zc3xx
	Card type        : USB Camera (046d:08d7)
	Bus info         : usb-0000:02:00.0-1
	Driver version   : 5.12.14
	Capabilities     : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (gspca_zc3xx: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'JPEG' (JFIF JPEG)
	Field             : None
	Bytes per Line    : 640
	Size Image        : 115790
	Colorspace        : JPEG
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
JPEG compression:
	Quality: 75
	Markers: 0x00000018
		Define Huffman Tables
		Define Quantization Tables
Streaming Parameters Video Capture:
	Frames per second: invalid (0/0)
	Read buffers     : 2

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
                          gamma 0x00980910 (int)    : min=1 max=6 step=1 default=4 value=4 flags=slider
                       exposure 0x00980911 (int)    : min=781 max=18750 step=1 default=2343 value=2343 flags=inactive, volatile
                 gain_automatic 0x00980912 (bool)   : default=1 value=1 flags=update
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=0 value=2
				0: Disabled
				1: 50 Hz
				2: 60 Hz
                      sharpness 0x0098091b (int)    : min=0 max=3 step=1 default=2 value=2 flags=slider

JPEG Compression Controls

            compression_quality 0x009d0903 (int)    : min=50 max=87 step=1 default=75 value=75
```

```
[... 82 lines cut ...]
(I)[14:23:01]{interface.c:404}	Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
(D)[14:23:01]{timer.c:34}	Disarmed timerfd on fd 36.
(I)[14:23:01]{upower.c:54}	AC cable connected.
(I)[14:23:01]{upower.c:65}	Laptop lid closed.
(D)[14:23:01]{interface.c:456}	Emitting 'AcState' property
(D)[14:23:01]{timer.c:32}	Set timeout of 0s 1ns on fd 37.
(D)[14:23:01]{interface.c:456}	Emitting 'LidState' property
(D)[14:23:01]{timer.c:32}	Set timeout of 30s 0ns on fd 37.
(D)[14:23:01]{location.c:95}	40.43 -105.09 loaded from cache file.
(D)[14:23:01]{my_math.c:248}	Loc distance: 40.43,-105.09 -> 91.00,181.00 : 3444.24 km.
(I)[14:23:01]{location.c:62}	New location received: 40.43, -105.09.
(D)[14:23:01]{timer.c:32}	Set timeout of 0s 1ns on fd 38.
(D)[14:23:01]{interface.c:456}	Emitting 'Location' property
(I)[14:23:01]{daytime.c:175}	Next alarm due to: Mon Jul  5 20:02:00 2021
(D)[14:23:01]{timer.c:32}	Set timeout of 20339s 0ns on fd 38.
(D)[14:23:01]{interface.c:456}	Emitting 'Sunrise' property
(D)[14:23:01]{interface.c:456}	Emitting 'Sunset' property
(D)[14:23:01]{interface.c:456}	Emitting 'DayTime' property
(D)[14:23:01]{timer.c:32}	Set timeout of 0s 1ns on fd 39.
(D)[14:23:01]{backlight.c:604}	Paused as no sensor is available.
(D)[14:23:01]{utils.c:56}	Pausing BACKLIGHT
(D)[14:23:01]{interface.c:456}	Emitting 'NextEvent' property
(D)[14:23:01]{interface.c:456}	Emitting 'SensorAvail' property
(D)[14:23:01]{utils.c:56}	Pausing SCREEN
```

FedeDP, 2021-07-06:
Clight uses the first device Clightd finds, using udev.
Maybe kernel is somewhat "messing" with your webcam devices?
Or maybe you've got some usb autosuspend aggressive settings? Just a guess...

Reading through your `v4l2-ctl` outputs above, it seems none of your usb cameras supports `exposure_absolute` streaming parameter, that's why you weren't noticing any difference during your tests!

yochananmarqos, 2021-07-06:
> Maybe kernel is somewhat "messing" with your webcam devices?

Hmm, not sure about what the kernel would be doing. I'm using 5.12.14-2-MANJARO. 

> Or maybe you've got some usb autosuspend aggressive settings?

I have TLP installed with the default USB settings. My laptop is a desktop replacement (it has a desktop processor! 🤤), so it's always on AC power. The USB autosuspend shouldn't be coming into play.

```
❯ tlp-stat -u | grep uvcvideo
Bus 005 Device 003 ID 04f2:b685 control = auto, autosuspend_delay_ms = 2000 -- Chicony Electronics Co., Ltd Chicony USB2.0 Camera (uvcvideo)
Bus 001 Device 010 ID 046d:0825 control = auto, autosuspend_delay_ms = 2000 -- Logitech, Inc. Webcam C270 (uvcvideo, snd-usb-audio)
```

closed 2021-07-06 by yochananmarqos