FedeDP/Clight #218: How to prevent clight from setting the brightness of my external monitor?

coldspark29, 2021-07-30:
I don't want clight to set my external monitors brightness. Especially when the laptop lid is closed, it gets very dark. How can I stop it?

[image]

FedeDP, 2021-07-30:
Hi!
> I don't want clight to set my external monitors brightness. 

Never? If you never want Clight to adjust your external monitor backlight, you should rebuild clightd without ddcutil support!
If you only want to disable screen backlight calibration while laptop lid is closed, there is an option in Clight conf file: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L93.

> I think it would be great if there would be a box to tick besides the monitors, to enable/disable brightness control.

Note that clight-gui and clight comes from different developers ;) 
By the way, there is still no way to prevent a screen from being calibrated.

coldspark29, 2021-07-31:
No, the brightness of my monitor is okay, It doesn't need to be adjusted. I already found the switch to pause while the lid is closed. It certainly helps quite a bit, but it is currently daylight and clight has dimmed my monitor to 65%. It is just too dark.

FedeDP, 2021-08-02:
> No, the brightness of my monitor is okay, It doesn't need to be adjusted. 

In this case, you just need to rebuild clightd without ddc support: https://github.com/FedeDP/Clightd/wiki/Build#build-time-switches (ENABLE_DDC=0)
Which distro are you on? I can guide you!

> I think it would also be great if clight could use some sort of machine learning to process the values the user sets manually. 

Clight has a stupidly simple wizard mode that should help you create the correct backlight curve for your sensor, look at `clight --wizard`.

coldspark29, 2021-08-02:
I am not going into rebuilding clight. It is just too much trouble.

I have already tweaked the curve for my sensor. It is really good for my laptop screen now, but for the external monitor, I would need different values. Do you think per-monitor settings would be doable?

FedeDP, 2021-08-02:
Yep, latest Clight added support for per-monitor settings indeed!
Have a look at monitor_override in clight conf: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L149
It has detailed description on how to use it!

coldspark29, 2021-08-02:
Okay thanks :)

closed 2021-08-02 by coldspark29