FedeDP/Clight #219: [FEATURE REQ] Provide way to set keyboard brightness levels

coldspark29, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
My keyboard has two brightness levels and clight can only set it to brightest.

**Describe the solution you'd like**
I would like to be able to set the brightness level and two thresholds for them.

FedeDP, 2021-08-02:
Hi! Clight should set keyboard backlight level based on current screen backlight, ie: the lower the screen backlight, the higher the keyboard backlight.
Thus, with a backlight around 50%, you should have keyboard backlight around 50% of your max, ie: level 1.

coldspark29, 2021-08-02:
This is exactly the wrong way round in my opinion. When the ambiant light is brighter, the keyboard light should be brighter and it should get darker if the ambient light gets darker, so it doesn't blind me.

FedeDP, 2021-08-03:
Hi! in latest clight commit i implemented curve support for keyboard backlight too (just like clight already does for screen backlight).
This enables full customization of keyboard backlight.

Please note that these new config options are still not exposed through DBus API.

FedeDP, 2021-08-05:
The new config options are now fully exposed by DBus API (and documented in Clight wiki pages).

coldspark29, 2021-08-05:
Cool! Will have a look tomorrow :)

nullobsi, 2021-08-09:
I've added the config options to Clight GUI :)

coldspark29, 2021-08-09:
Great! Now only the monitor override is missing in clight-gui and everything is perfect. At least for me :P

FedeDP, 2021-08-15:
Can this be closed? :)
Btw i implemented dbus api for monitor override in cd174174d0d67b0816fcc1429d5e8d2be1374fc6.
I will now tell clight-gui developer that the new api is available!

coldspark29, 2021-08-15:
@FedeDP I honestly haven't tried, because the AUR script is not able to download the new version from Github. But I guess it can be closed if everything works as you say.

closed 2021-08-15 by coldspark29

nullobsi, 2021-08-15:
clight-git will always use the latest version of clight from GitHub :)

FedeDP, 2021-08-15:
Uh that's weird! Are you using clight-git?
Thanks for the feedback btw!

coldspark29, 2021-08-15:
@nullobsi 

```
/usr/bin/bash: warning: setlocale: LC_ALL: cannot change locale (c)
==> Making package: clightd-git 5.2.r2.gadb439b-1 (Sun Aug 15 19:40:44 2021)
==> Retrieving sources...
==> ERROR: /home/jan/.cache/yay/clightd-git/clightd is not a clone of https://github.com/FedeDP/Clightd.git
    Aborting...
error downloading sources: clightd-git
```

coldspark29, 2021-08-15:
@FedeDP Just realized you are the AUR maintainer as well. I already complained there :)

coldspark29, 2021-08-15:
@nullobsi Btw clight-gui is always deleting my monitor overrides and inserting an empty one. Should I make an issue?

FedeDP, 2021-08-15:
I am no more the aur maintainer :)
You should clean yay cache (rm the path it tells you) and you should be fine!

coldspark29, 2021-08-15:
I already tried ``yay -Ycc``, but it still happens

coldspark29, 2021-08-15:
Ah yeah, delete the folder you meant. Did that and it compiles now.

coldspark29, 2021-08-15:
@FedeDP  So how do I configure this? I can't find the Wiki page. Just paste regression points as with the monitor?

coldspark29, 2021-08-15:
So everything works really well now. Thank you both for developing this. When you make the right settings for your monitor and sensor, it works really well. 

@FedeDP There is one issue though with the keyboard brightness. It is not possible to set the brightness brighter for higher ambient light, because you will always have a continuous graph. In order to set the brightness to 50% from 0-10% ambient brightness and to 100% from 10-20% ambient brightness, it woul require a discontinuous graph. Else it would be at 50%, then at 100% and then at 50% again when it is getting dark.

[image]

nullobsi, 2021-08-15:
this is because clight runs the points through a curve regression, so it tries to fit a polynomial curve onto the points (you can see it in the logs, at least for the screen brightness)

nullobsi, 2021-08-15:
clight-gui just shows the points directly as a "line chart," I think I was meaning to get around to getting the graphs to show the actual regression but I couldn't find a suitable library or something :sweat:

coldspark29, 2021-08-15:
It shouldn't be too hard with the polynomial curve fitting. Just don't be afraid of the math :) You can play around with WolframAlpha or [plots](https://github.com/alexhuntley/Plots)