FedeDP/Clight #220: no more adjusting

promeneur, 2021-08-13:
opensuse 15.2
clight 4.5
clightd 5.3

clight and clightd are running.

but there is no color or light adjusting.

I wonder if this is due to the update to kde gear 21.08.0.

here is the log of clight.

[clight.log](https://github.com/FedeDP/Clightd/files/6981775/clight.log)

FedeDP, 2021-08-13:
Moved to Clight repo.  
It seems like Clight is awaiting on geoclue to give it a position as there is no cached position.  
Have a look at https://github.com/FedeDP/Clight/wiki/FAQ#clight-is-not-doing-anything

promeneur, 2021-08-13:
in the journal of geoclue service i found this error:

gclue_locator_set_time_threshold: assertion 'GCLUE_IS_LOCATOR (locator)' failed

 i restart the service then no problem.

thanks

closed 2021-08-13 by promeneur