FedeDP/Clight #221: [BUG] org.clight.clight.Conf.Kbd AcPoints/BatPoints not working

nullobsi, 2021-08-15:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> latest git rev (clight-git 4.6.r10.gcd17417-1)

**Used distribution:**
> arch linux

**Describe the bug**
it appears the Dbus interface for keyboard points isn't working; the call returns good, but after checking the set value, it doesn't change from the default

**Expected behavior**
when setting the DBus properties, they should be set, and the same value should be returned after

**To Reproduce**
trying a command like `busctl --user set-property org.clight.clight /org/clight/clight/Conf/Kbd org.clight.clight.Conf.Kbd BattPoints ad 11 0.5 0.45 0.4 0.35 0.3 0.25 0.2 0.15 0.1 0.5 0` proceeds fine, but `busctl --user get-property org.clight.clight /org/clight/clight/Conf/Kbd org.clight.clight.Conf.Kbd BattPoints` returns the original value, same for AC

i'm unsure if this is because of
```
(D)[11:48:40]{bus.c:188}	init_kbd_backlight(): No such file or directory
(I)[11:48:40]{keyboard.c:145}	Keyboard backlight calibration unsupported.
```

but here is the rest of the log
[clight.log](https://github.com/FedeDP/Clight/files/6988692/clight.log)

FedeDP, 2021-08-15:
Yes, if keyboard backlight is unsupported, keyboard module is killed, thus nobody will actually answer the request generated from the DBus interface.
I already want to kind of refactor the DBus Api to expose various /Conf/$module from within the module code, so that it is only exposed if the module is actually running.

FedeDP, 2021-08-15:
Thanks for opening the bug request! I will make sure to work on this!

commit 8fa523ee (2021-08-21): Moved all Conf/ dbus api inside each module, so that if module gets deregistered, its dbus api dies with it. Fixes #221
closed 2021-08-21 by FedeDP