FedeDP/Clight #224: [BUG] sigsegv when running `qdbus org.kde.KWin /Effects activeEffects`

NICHOLAS85, 2021-09-01:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Latest git (8047e69d85ec28e2c41c322b089506fa354cac5e)

**Used distribution:**
> Bedrock Linux 0.7.21beta2 Poki, Debian Testing stratum

**Describe the bug**
A clear and concise description of what the bug is.
Occasionally Clight would abort while I was using my computer, I found that a script I run was causing Clight to crash. I was able to narrow it down to the `qdbus org.kde.KWin /Effects activeEffects` command. This is a KDE specific command. This bug would not always happen if Clight was already running but is easy to reproduce when Clight was recently launched.

**Expected behavior**
Clight not to crash

**To Reproduce**
Launch Clight and quickly after run `qdbus org.kde.KWin /Effects activeEffects`

Logs:
[clightlog.txt](https://github.com/FedeDP/Clight/files/7093749/clightlog.txt)
[strace.txt](https://github.com/FedeDP/Clight/files/7093750/strace.txt)

NICHOLAS85, 2021-09-01:
Sure, to build in debug mode do I just change `-DCMAKE_BUILD_TYPE="Release"` to `-DCMAKE_BUILD_TYPE="Debug"` in the build script?

NICHOLAS85, 2021-09-01:
Heres the output at crash:
```
Program received signal SIGSEGV, Segmentation fault.
__strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:101
101     ../sysdeps/x86_64/multiarch/strcmp-avx2.S: No such file or directory.
```

FedeDP, 2021-09-01:
Ok can you execute "bt" in the gdb shell after the crash? And paste here the output! Thank you!

NICHOLAS85, 2021-09-01:
```
(gdb) bt
#0  __strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:101
#1  0x0000555555567dd2 in inhibit_parse_msg (m=0x5555556426b0)
    at /home/nicholas/github/built/Clight/src/modules/interface.c:330
#2  0x0000555555567820 in receive (msg=0x7fffffffb2b0, userdata=0x0)
    at /home/nicholas/github/built/Clight/src/modules/interface.c:211
#3  0x00007ffff7f936f3 in run_pubsub_cb () from /lib/libmodule.so.5
#4  0x00007ffff7f92e71 in recv_events () from /lib/libmodule.so.5
#5  0x00007ffff7f930dd in modules_ctx_loop_events () from /lib/libmodule.so.5
#6  0x000055555555f8fc in main (argc=2, argv=0x7fffffffb448) at /home/nicholas/github/built/Clight/src/main.c:45
```

FedeDP, 2021-09-01:
It seems Clight dies here:
```
      const char *interface = sd_bus_message_get_interface(m);
        
        for (int i = 1; i <= 2; i++) {
            if (!strcmp(interface, sc_interface)
```
It seems that "interface" is NULL there. I did not know that a message could be emitted with no interface (?).
I pushed a commit, can you try updating clight and see if crash persists?

NICHOLAS85, 2021-09-01:
After updating, it looks like the crash no longer persists!

FedeDP, 2021-09-01:
Thank you very much for the debug session!

closed 2021-09-01 by FedeDP

NICHOLAS85, 2021-09-01:
of course and thanks for the quick fix